tsduck/tsduck #1331: [Q] How to use tsfclean utility?

aliandwa, 2023-10-24:
Please show an example of a command line for clearing a single .ts file. The User Guide is uninformative and causes difficulties for me.

AlfonZ42, 2023-10-25:
For example `tsfclean -o fileout.ts filein.ts`.

lelegard, 2023-10-25:
Could you please describe how uninformative the user's guide is on tsfclean?  What should be added to make it more informative?

The command syntax itself is quite simple, there is only one option, the output file, as illustrated by @AlfonZ42.

What the command does is copied below from the user's guide:

> In its most general form, an MPEG transport stream file is just a set of TS packets. It can be a capture of a live stream, in which case the file starts and ends at arbitrary points in an endless transmission.
>
> However, when a media player manipulates stored contents, it expects the files to start with the actual beginning of an audio/video content. When the file format is an MPEG transport stream, the player expects some characteristics such as immediate identification of the services and PID’s, initial intra video frame, etc. Not matching these characteristics does not prevent the content from being rendered by the player but glitches are usually present at startup.
>
> The tsfclean command cleans up a TS file to make it more consistent for media players and other similar tools. The following transformations are applied:
> - The output file starts with the PAT, the CAT (if present on input), the SDT (if present on input) and the PMT’s of all services. Thus, the player is aware of the exact structure of the TS before processing the first audio / video data.
> - EIT present/following for actual existing services are kept. All other EIT’s are removed.
> - All other PSI/SI (including NIT, BAT, TDT and other broadcast-related tables), all null packets and all orphan PID’s are deleted.
> - In each video PID, all packets preceding the first intra-frame are deleted. If no intra-frame can be found (unknown video codec or scrambled contents), all packets preceding the first complete PES packet are deleted.
> - In each audio, subtitles or data component of the services, all packets preceding the first complete PES packet or section are deleted.
>
> If the input file contains several versions of a table (PAT, CAT, SDT or PMT’s), all successive versions are merged into one single version of the table. Some players are known to read the first table of each kind only and are not able to handle table updates as a TV receiver would do. Consequently, if a service or a component of a service appears and disappears several times, it becomes in fact declared from the start to the end of the output file. If incompatible non-cumulative changes are introduced in a table update, an error is reported.

What should we add?

aliandwa, 2023-10-25:
Thanks for the quick replies! I'm not a programmer and the command line syntax is confusing for me. So one or two examples for each description would be very helpful for understanding.

lelegard, 2023-10-25:
This is not programming, just shell usage.

The syntax of all TSDuck commands are quite similar to most shell commands. They use the same rules. TSDuck is not a GUI application where specific tasks are pre-programmed and accessible from a few mouse clicks. TSDuck is a generic framework which allows an infinity of combinations, just like ffmpeg. This is a lab tool for DTV engineers.

I do not recommend TSDuck to users who are not familiar with shell scripting.