tsduck/tsduck #1332: [FEAT] Add '+' to parent make rule to allow parallel build job

Dando-Real-ITA, 2023-10-25:
#### Is your feature request related to a problem?
When building from the git source with `make -j$(nproc)`, I get this warning: `make[2]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.` 
Normally when doing an incremental build after a git pull, by splitting the build and install phase I am able to use make in parallel. However when new objects are added the build fail with an error like:
`make[2]: *** No rule to make target '/opt/ffmpeg/src/tsduck/src/libtsduck/base/types/tsTypedEnumeration.h', needed by '/opt/ffmpeg/src/tsduck/bin/release-x86_64-gpu01/objs-libtsduck/tsArgsWithPlugins.o'.  Stop.`
Then I am only able to build using `make -j$(nproc) -B` and that's where the warnings kick in and the build is very slow

#### Proposed solution:
Add the + to parent make where requested

#### Alternatives:
Maybe there is an alternative in the command? If known please share

#### Additional information:
I think the easiest way to replicate is to build from git using `make -j$(nproc) -B`

Dando-Real-ITA, 2023-10-27:
Thank you for the suggestions,
I added to MAKEFLAGS, GNUMAKEFLAGS, and `alias make` to use always `-j$(nproc)`, and also added a `make clean` always before the build step, that should work.

lelegard, 2023-10-27:
> I added to MAKEFLAGS, GNUMAKEFLAGS, and alias make to use always -j$(nproc)

OK, but that is not the real solution, even if it works. 

Every day, I use `make -j10` (for 8 cores) on Linux, on macOS, and many VM's, without any of these environment variables, and it works without the error messages you got. If you got those errors, this must be because something is already defined in your environment or configuration. The real solution is finding these existing external causes.

closed 2024-03-31 by github-actions[bot]