tsduck/tsduck #1335: [BUG] Plugin tables add/remove/ same service with , PMT is not analyzed

enricovittorini, 2023-10-30:
#### Bug description:
When a service with PMT version x is removed form  TS, and then added again with the same PMT version x, table plugin does not see the PMT of the service

#### How to reproduce:
Add a service and take not of the version of the PMT (version: x)
Remove the service
Add again the same service with the same pmt version  (version: x).

#### Expected behavior:
When service is added again in the TS, the plugin should see the PMT even if with the same version as before.

#### Environment:
 - OS: Windows 
 - OS version: 10
 - TSDuck full version: 3.36-3353
 - Installation type: official binary

lelegard, 2023-10-30:
Hi @enricovittorini 

This is not a bug. This is the expected behaviour. By default, the plugin `tables` displays new tables. Per MPEG specification, a section with the same tid/tidext/version on the same PID is read only once. 

If a service is suspended (removed then added again with the same PID's) and the PMT did not change, there is no new PMT, this is the same PMT as before. The fact that it was suspended for a while does not change the expected behaviour of a section demux.

enricovittorini, 2023-10-30:
@lelegard , understood the logic of the plugin, but would be great to improve the it.

When a service is removed, it does not exists anymore. there is no memory of the history.
When it is added again (note, it could be a new service with exact the same structure as previous one), I think the PMT should be seen.

enricovittorini, 2023-10-30:
Ok thanks. 

I could workaround I think with a restart of the tables plugin with there is still the bug #996 pending.

AlfonZ42, 2023-10-30:
My understanding is that when a table changes, its version should be increased (and _tables_ plugin detects this change). So when a service is removed, _PAT_/_SDT_/etc. are changed, when it's added back, they are changed again.

Is it possible to make use of these changes (even though the service's _PMT_ does not change) in your scenario?

enricovittorini, 2023-10-30:
Thanks @AlfonZ42 for your answer. 

When you remove the service it does not exists anymore, nor its PMT and there is not memory of it anywhere.

When you add it again, you start from scratch. 

it is a brand new service. It can happen that the PMT is the same and the service that was removed before.

I could implement something if I was able to restart the tables plugin with json-line output probably, but cannot test it as per the mentioned bug.

lelegard, 2023-10-30:
> there is not memory of it anywhere.

Sorry, but this is your assumption only, not otherwise specified.

> When you add it again, you start from scratch. it is a brand new service.

Your assumption only, again. It is also valid to say that the same unchanged service is back.

In theory, if your start a really new service, with same PMT PID and same service id but a new service (can we still say it is a new one?), then the broadcaster has to remember that this PMT PID was used before for a "different" service of same service id. And the broadcaster has to remember the last version of that PMT. And the broadcaster has to use a new version for the "new" PMT (on same PID with same service id).

Too complicated. In practice, in case of reconfiguration of a TS, I have always seen that new services are assigned new service ids and new PID's.

reachmahesh80, 2023-11-02:
If a service is added or removed - then the version number has to change. Thats the only way to notify the receiver on change in the tables.

closed 2024-04-07 by github-actions[bot]