tsduck/tsduck #1336: [Q] Signal Stats from DVB

SonnyWalkman, 2023-10-31:
Hello @lelegard,

Using DVB plugin..

I'm using the` -I dvb [options] ` to set and receive dvb-s and dvb-t signals.  I'm using a Multi system card which works well.  
Is there a mechanism of reporting signal stats (signal lock, signal strength BER etc) through DVB plugin?

I know there is femon however, I'm asking if there is any appetite to add status reporting to the plugin and send to file or a UDP channel in json ?
I've scanned the forum and found a few requests. Is there any way to report these than having to hack with `femon` on another solution for windows? This would of some benifit?

AlfonZ42, 2023-10-31:
I think `tslsdvb -v` command lists some such parameters (on Linux only).

SonnyWalkman, 2023-10-31:
Only shows what’s available on the machine.

Sent from Gmail Mobile iPhone 13

lelegard, 2023-10-31:
> Only shows what’s available on the machine.

But how could we get more ? Something that is not available on the machine ? 😄

SonnyWalkman, 2023-10-31:
Hi Thierry,  
Is there appetite to have front stats build into the dvb plugin with options  (idea)
`-I dvb -f xxxxxxxx  -m QPSK ...  --stats [sub options]` udp or to file?

TSduck being platform aware:-

1. Linux, femon  in the DVB5 API.
2. iOS  femon 
3. Windows not sure how DirectX handles stats or BDA from headend cards? 

[WINDOWS]?I can't seem to get any clear understanding on DirectX however, there is likely signal reporting included in the API since I'm using a Windows USB DVB-T receiver, and it reports signal stats with the application is comes with.

 I understand from a few questions and your comment Thierry there aren't any standards on signal statistics for providing these signal levels, SNR, BER etc left for the driver developer to determine. For me using one manufacture TBS technologies makes it easy to determine what is delivered in stats since they have one driver to fit all their PCIe cards.   

I'm implementing receiving both DVB-S and DVB-T with 4 frontends to mimick each dvb stream similar to a set top box.
Having signal stats is a must have and knowing Video and Audio PID's. The later  could be retrieved from other TSduck such as analyse etc.

Happy to discuss further and hear your recommendations?

AlfonZ42, 2023-11-01:
Here is an example output of `tslsdvb -v` on my Windows machine to illustrate the differences in values:
```
0: "AVerMedia 716x BDA DVBT Tuner" (DVB-T, DVB-T2)
     Signal locked:    yes
     Signal strength:  0.1 dB

1: "WinFast DTV Dongle Dual Filter" (DVB-T, DVB-T2)
     Signal locked:    yes
     Signal strength:  -40 dB

3: "AVerMedia BDA DVBT Tuner" (DVB-T, DVB-T2)
     Signal locked:    yes
     Signal strength:  0.092 dB
```
They all are tuned to the same channel, using exactly the same signal.
Are you able to deduce that all saved test streams were 100% without errors?

Playing around with `tslsdvb -e`, I got the following that might be relevant:
```
SIGNAL_STRENGTH (SignalStats) : get
SIGNAL_QUALITY (SignalStats) : get
SIGNAL_PRESENT (SignalStats) : get
SIGNAL_LOCKED (SignalStats) : get
SAMPLE_TIME (SignalStats) : get
RF_TUNER_FREQUENCY (FrequencyFilter) : get set
RF_TUNER_BANDWIDTH (FrequencyFilter) : get set
RF_TUNER_FREQUENCY_MULTIPLIER (FrequencyFilter) : get set
```
which seems to suggest more values might be listed, but for reasons outside my expertise are not.

On the other hand, I understand that supporting these features on poorly standardised Windows has very low priority.

lelegard, 2023-11-01:
Hi @SonnyWalkman, @AlfonZ42 

Let's be clear here. TSDuck is a transport stream processing framework. As such, it reads/writes TS from various sources, DVB, network, etc. Reading UDP or HTTP TS from the network does not mean that TSDuck shall include network monitoring tools. The same applies for DVB sources. I can admit that zillions of network monitoring tools already exists while almost none on DVB (at least as free open source software). Nonetheless, TSDuck is not designed to become a DVB _signal_ monitoring tool.

Moreover, TSDuck aims to remain portable. Supporting DVB is a breach in this contract because it is not supported on macOS or BSD, only Linux and Windows. I do not want to go further into fragmentation.

As you noted, even inside one operating system (Windows), the standard API is both limited in features and overly complicated in structure. In short, this is a piece of crap. Additionally, the signal monitoring data do not even use the same units between drivers, as you demonstrated. A signal strength cannot be 0.1 dB and -40 dB at the same time. And which unit is used where is not even documented.

Many tuners have capabilities which go beyond Windows BDA API. As a consequence, their proprietary drivers include additional API's which are proprietary and undocumented. They can be used only by the proprietary software which comes with the tuner. These API's cannot be used by generic tools such as TSDuck.

As a consequence, TSDuck shall remain what it is, a portable flexible generic transport stream manipulation framework. Nothing more. It does not monitor my Ethernet network, my Wifi network, my DVB network, my fridge or my coffee machine.

DVB monitoring shall be done by appropriate tools. Most of them will be proprietary and attached to specific hardware, either cheap consumer tuners or professional monitoring tools. That is a fact.

So, the answer is a final no. Sorry.

SonnyWalkman, 2023-11-02:
@lelegard 
I'm not expecting TSDuck to diverge into a fully-fledged processing system.  I think your've steered into the weeds here.
I'm fully aware and understand TSDucks purpose as a tool kit. 

I also acknowledge in person for the time and effort which has gone into developing such. It is well written which passion and devotion and I'd continue to support with donations since I am using it for community purposes.

I develop in Go and could assist however C++ is not my preferred language of choice. 

> So, the answer is a final no. Sorry

@AlfonZ42 your comments have provided me a path on how to achieve stats using `tslsdvb -v` Unfortunately involves another command line exec to extract the stats report from the tool without needing to CGO directly to the TSDuck library. Possible however.

In essence, obviously TSDuck **does** under the hood provide front end signal statistics? 

Not sure what all this is about? I didn't indicate TSDuck to monitor networks? Totally confused. 😕😕😕

> Reading UDP or HTTP TS from the network does not mean that TSDuck shall include network monitoring tools. The same applies for DVB sources. I can admit that zillions of network monitoring tools already exist while almost none on DVB (at least as free open source software). Nonetheless, TSDuck is not designed to become a DVB signal monitoring tool.

I was asking if there could be way to cater in polling sig stats with in dvb plugin which had already past the parameters to the adaptor and frontend in the first instance? Why do this again through `tslsdvb -v`  and having to regex out the response?

Thanks very much @AlfonZ42 for your effort in your reply. This can give me some hope to a temporary solution.

Now knowing my humble request is dead in the water.

 I'll close these comments.

closed 2023-11-02 by SonnyWalkman