tsduck/tsduck #1337: [Q] How to get scte35 splice info sections in the callback

reachmahesh80, 2023-10-31:
How to get scte35 splice info sections in the callback of a new plugin or in an application?
For ex: tsp -I file xac  -P splicemonitor -d -O drop 
Above command provides me information about spliceInsert command. What I need is to use this inside a c++ program and extract all of the packets in the scte 35 sections.

I am using the sample plugin : tsduck/sample/sample-plugin/tsplugin_sample.cpp
And the below in test_sample.sh:
tsp -I file Channel5_xac.ts -P splicemonitor -d -O drop -P sample --count

On execution of test_sample.sh - I see packet count of : 557753 - which seems wrong. 
Also I am printing the pid info from the below callback - its not printing the pid of the scte35 packets.  

ts::ProcessorPlugin::Status ts::SamplePlugin::processPacket(TSPacket& pkt, TSPacketMetadata& pktData)
{
    if (doCount) {
        counter++;
    }
    printf("pid = 0x%x \n", pkt.getPID());
    return TSP_OK;
}
Any help in using the tsduck for this purpose will be highly appreciated.

lelegard, 2023-10-31:
Hi @reachmahesh80 

If you write a C++ application, then you are in the ideal situation where you can do everything you like.

The first question is: Do you want to integrate this processing inside some larger existing C++ application? Or do you simply want some specific processing in a custom plugin in a `tsp` command?

In the former case, start from `sample/sample-app-custom`. It provides an application which does the same thing as `tsp` but inside an application thread. A user-defined C++ class is used as plugin.

In the latter case, start from `sample/sample-plugin`, as you did.

In both cases, you end you with a C++ class which implements a custom plugin. In you case, I would recommend to mimic the structure of `splicemonitor` in `src/tsplugins`: use a section demux, pass all packets to it and get the demuxed tables in a handler. Either specify the SCTE35 PID on the command line or parse the PAT and PMT's to detect SCTE35 PID's.

In a word, your plugin should be basically a copy of `splicemonitor` except that you do something else when you get the SCTE35 tables.

reachmahesh80, 2023-10-31:
Thanks @lelegard  - appreciated your quick feedback.
I am fine with processing in a custom plugin. Sure I will explore using (or copying) splicemonitor. 
I will not know the scte35 pid - so I will have to obtain it dynamically. Should I parse the PAT/PMT tables to get the SCTE35 pid or Is there a function already developed that I can use it ? Ideally I would like to get the (full) SpliceInfoSection and corresponding pid including the service id under which scte35 section is present.

**Example:**
tsp -I file xac -P splicemonitor -d -O drop 
* splice PID 0x03C4 (964), event 0x3AB4D5E5 (984929765) out, occurrence #1
* SCTE 35 Splice Information, TID 252 (0xFC), PID 964 (0x03C4)
  Short section, total size: 50 bytes
  - Section 0:
    Protocol version: 0
    Encryption: none
    PTS adjustment: 0x000000000
    CW index: 0x00 (0), tier: 0xFFF (4095)
    Command type: 0x05 (SpliceInsert), size: 20 bytes
    Splice event id: 0x3AB4D5E5 (984929765), cancel: 0
    Out of network: yes, program splice: yes, duration set: yes, immediate: no
    Time PTS: 0x0B01CEA22 (2,954,684,962, 32,829,832 ms)
    Duration PTS: 0x0002932E0 (2,700,000, 30,000 ms), auto return: yes
    Unique program id: 0x0000 (0), avail: 0x01 (1), avails expected: 1
    - Descriptor 0: SCTE 35 Avail (0x00, 0), 8 bytes
      Identifier: 0x43554549 ("CUEI")
      Provider id: 0x3AB4D5E5
    CRC32: 0xEF956C20 (OK)

lelegard, 2023-10-31:
> I will not know the scte35 pid - so I will have to obtain it dynamically. Should I parse the PAT/PMT tables to get the SCTE35 pid or Is there a function already developed that I can use it ? Ideally [...]

You have to study how `splicemonitor` works. A `SignalizationDemux` is used to collect all PMT's. Collected PMT's are passed to the `handlePMT` virtual method of the plugin. This method inspects the components of the service, find the SCTE35 PID's and add them to a `SectionDemux`.

Whenever the `SectionDemux` finds a table in the PID's (a splice information table in that case), it passes it to the `handleTable` virtual method of the plugin. The splice information table is then a C++ object of class `SpliceInformationTable` and you can explore what you want with it.

My suggestion:
- Copy `tsplugin_splicemonitor.cpp`, change the names of the file, the class and the plugin name.
- Make sure it works as `splicemonitor` does, under the new name.
- Get rid of all the monitoring and display stuff, just keep a message inside  `handleTable` to show that you get a valid splice information table.
- Then do what you want in  `handleTable` with that table.

reachmahesh80, 2023-10-31:
Thanks I'll try it out.

reachmahesh80, 2023-10-31:
Is there a way to extract splice info section (RAW data) from the below handleTable function() ? 
void ts::SpliceMonitorPlugin::handleTable(SectionDemux& demux, const BinaryTable& table) 

We see the SerialisedPayload function returns a PSI buffer object which has the SpliceInfo section buffer but this function is a protected member, Is there a way to invoke this from handleTable?

lelegard, 2023-10-31:
Have a look at the documentation of the `BinaryTable` class, either in the [header file](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/dtv/signalization/tsBinaryTable.h) or the [doxygen doc](https://tsduck.io/doxy/classts_1_1BinaryTable.html).

It contains a list of binary sections. And each binary section exposes its binary data.

reachmahesh80, 2023-11-01:
Thanks it works.

closed 2023-11-01 by reachmahesh80