tsduck/tsduck #1338: [BUG] tspcontrol restart: analyze plugin not restarted, but tsp is

enricovittorini, 2023-10-31:
#### Bug description:
When tspcontrol is used to restart the analyze plugin with the below command, it seems that the tsp is restarted and not the analyze plygin

#### How to reproduce:
run the following command:
`tsp -v --control-port 3001 -I ip 239.0.0.1:5500 -l 172.16.0.100 -P analyze -i 2 -c --json-line  -O drop`

get the plugin id
```
tspcontrol -t localhost:3001 list
 0: -I ip
 1: -P analyze
 2: -O drop
```

from the tsp output i get
```
* tsp: user interrupt, terminating...
* ip: leaving multicast group 239.0.0.1 from local address 172.16.0.100
```

#### Expected behavior:
something similar when the "tables" plugin is restarted. on tsp output you can see that the plugin is restarted.

```
control commands: received from 127.0.0.1:64334: restart -s 1
* tables: restarting due to remote tspcontrol
```

#### Environment:
 - OS: Windows 
 - OS version: 10
 - TSDuck full version: 3.36-3360
 - Installation type: official binary

lelegard, 2023-10-31:
Hi @enricovittorini 

I do not see the same thing. With the same command, I get this:
~~~
* analyze: { "pids": ... }
* control commands: received from 127.0.0.1:37867: list
* analyze: { "pids": ... }
~~~
And this:
~~~
* analyze: { "pids": ... }
* control commands: received from 127.0.0.1:46307: restart -s 1
* analyze: restarting due to remote tspcontrol
* analyze: { "pids": ... }
~~~

Tried on Windows and Ubuntu.

enricovittorini, 2023-10-31:
@lelegard  here it is the stderr output 
[output.txt](https://github.com/tsduck/tsduck/files/13220031/output.txt)

`tsp  --control-port 3001 -I ip 239.0.0.1:5500 -l 172.16.0.100 -P analyze -i 2 -c --json-line -O drop 2> output.err`

`tspcontrol -t localhost:3001 restart -s 1`

enricovittorini, 2023-10-31:
@lelegard  here the stderr output: 
[output.txt](https://github.com/tsduck/tsduck/files/13220251/output.txt)

`tsp -v  --control-port 3001 -I ip 239.0.0.1:5500 -l 172.16.0.100 -P analyze -i 2 -c --json-line -O drop 2> output.txt`

`tspcontrol -t localhost:3001 restart -s 1`

enricovittorini, 2023-11-01:
cannot replicate anymore

closed 2023-11-01 by enricovittorini