tsduck/tsduck #1340: [BUG] tables plugin restart, missing PMT

enricovittorini, 2023-11-03:
#### Bug description:
when plugin "tables" is restarted, the PMTs tables are not analyzed

#### How to reproduce:
`tsp --control-port 3001 -I ip 239.0.0.1:5555 -l 172.16.0.100 -P bitrate_monitor -p 1 -t 1 -P tables --psi-si --log-json-line -O drop`

then:
tspcontrol -t localhost:3001 restart -s 2

#### Expected behavior:
PMT tables are analyzed again as the plugin is restarted

#### Environment:
 - OS: Windows
 - OS version: 10
 - TSDuck full version: 3.36-3360
 - Installation type: official binary

lelegard, 2023-11-03:
Hi @enricovittorini 

This is a bug indeed. I fixed it.

The `tspcontrol` utility and the ability to restart plugins were introduced late, after most plugin were already developed. All plugins has to go through some review to evaluate and fix the impact restart. There are some remaining corner cases which are not properly reset when a plugin is restarted. Not an excuse, just an explanation...

enricovittorini, 2023-11-03:
Amazing! @lelegard thanks a lot as usual!

closed 2023-11-18 by lelegard