tsduck/tsduck #1342: [BUG] Service with PID from other Service

SuppaMan150, 2023-11-03:
#### Bug description:
I encountered a problem with TSDuck Analyze when analyzing TS files. Despite a service being of type 2 (digital radio), the list of PIDs includes the video PID of another channel present in the transponder. Also the total bitrate of the service is affected by the video track. After verification through television and another TS analysis software, it is confirmed that the service should contain only audio track.

#### How to reproduce:
Dump the file by:
~~~
.\tsp -I dvb --frequency 690000000 -P until --seconds 10 -O file "dump.ts"
~~~
Analyze the file by
~~~
.\tsp --default-pds eacem -I file "dump.ts" -P analyze -o "analysis.txt" -O drop
~~~

#### Expected behavior:
The service PID list shoud contain only the PMT and MPEG-2 Audio track.

#### Errors and logs:
Here is the section relative to the particular service.
~~~
|=============================================================================|
|  Service: 0x0068 (104), TS: 0x0200 (512), Original Netw: 0x001D (29)        |
|  Service name: inBlu2000, provider: Persidera                               |
|  Service type: 0x02 (Digital radio sound service)                           |
|  TS packets: 19,427, PID's: 3 (clear: 3, scrambled: 0)                      |
|  PMT PID: 0x0474 (1140), PCR PID: 0x044D (1101)                             |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Digital radio sound service .................. C    3,640,763 b/s  |
|  0x044D  AVC video (1920x1080, high profile, level 4.0  C+   3,499,832 b/s  |
|  0x0474  PMT .......................................... C        5,060 b/s  |
|  0x0476  MPEG-2 Audio (ita, Audio layer II, 128 kb/s,   C      135,870 b/s  |
|          (C=Clear, S=Scrambled, +=Shared)                                   |
|=============================================================================|
~~~

#### Environment:
 - OS: Windows
 - OS version: Windows 10
 - TSDuck full version: 3.35-3258
 - Installation type: Portable

First noticed the bug using version 3.30.2710 (installer) on Windows 8.1, confirmed on the last version with newer OS.

#### Attachments:
I attach the original dump.ts capture, the full analysis log, and a screenshot of the analysis of another program.

[analysis.txt](https://github.com/tsduck/tsduck/files/13255373/analysis.txt)
[analysis.png](https://github.com/tsduck/tsduck/assets/40044327/11a53dc4-62f5-486b-801e-68187c0fa225)
[dump.zip](https://github.com/tsduck/tsduck/files/13255374/dump.zip)

lelegard, 2023-11-04:
Hi @SuppaMan150 

This is not a bug in TSDuck. The video PID 0x044D is declared as PCR PID in the PMT of the radio service:

~~~
$ tstables dump.ts --tid 2 --tid-ext 0x0068

* PMT, TID 2 (0x02), PID 1140 (0x0474)
  Version: 1, sections: 1, total size: 27 bytes
  - Section 0:
    Program: 104 (0x0068), PCR PID: 1101 (0x044D)   <==================== here
    Elementary stream: type 0x04 (MPEG-2 Audio), PID: 1142 (0x0476)
    - Descriptor 0: ISO-639 Language (0x0A, 10), 4 bytes
      Language: ita, Type: 0x00 (undefined)
~~~

Consequently, the command `tsanalyze` includes that PID into the service.

We can argue whether or not this is legitimate. In practice, `tsanalyze` reports a high-level analysis. Thus, it includes all PID's which are somehow related to a service in that service. Then, if you notice something suspicious, dive into the details using `tstables`.

Other radio services in the TS use their own audio PID as PCR PID. However, their audio PID contains PCR and PTS and the audio PID alone can be used to synchronize the audio rendering.

In the case of service 0x0068, the audio PID does not contain any PCR, only PTS. This audio service is therefore unable to synchronize itself using its own PID's. It needs an external synchronization source, which is found here in a TV service named TV2000.

This is very unusual. First, the TV and radio services must be synchronously encoded to share a common PCR source. Second, the TV service TV2000 is signaled twice, as services 0x0064 and 0x0065. The two services have different ids, different PMT's but share the same components. There is something suspicious about these services.

Finally, let me share another error in the signalisation of the TS. In the NIT, they use something which looks like a logical channel number descriptor (a private descriptor which is defined by several operators) but it is not preceded by a private data specifier descriptor which establishes the private context. This is illegal in DVB terms. Probably, the receivers have some hard-coded interpretation of the logical channel number descriptor without caring about any private data specifier. This is not the first fime I see this: receivers being "bug compatible" with the broadcaster...

Demonstration:
~~~
$ tstables dump.ts  --pid 16

* NIT Actual, TID 64 (0x40), PID 16 (0x0010)
  Version: 3, sections: 1, total size: 184 bytes
  - Section 0:
    Network Id: 12289 (0x3001)
    Network information:
    - Descriptor 0: Network Name (0x40, 64), 10 bytes
      Name: "Persidera2"
    Transport Stream Id: 512 (0x0200), Original Network Id: 29 (0x001D)
    - Descriptor 0: Service List (0x41, 65), 57 bytes
      Service id: 13 (0x000D), Type: 0x02 (Digital radio sound service)
      Service id: 40 (0x0028), Type: 0x19 (Advanced codec HD digital television service)
      Service id: 55 (0x0037), Type: 0x16 (Advanced codec SD digital television service)
      Service id: 60 (0x003C), Type: 0x19 (Advanced codec HD digital television service)
      Service id: 65 (0x0041), Type: 0x19 (Advanced codec HD digital television service)
      Service id: 70 (0x0046), Type: 0x16 (Advanced codec SD digital television service)
      Service id: 71 (0x0047), Type: 0x16 (Advanced codec SD digital television service)
      Service id: 72 (0x0048), Type: 0x16 (Advanced codec SD digital television service)
      Service id: 80 (0x0050), Type: 0x16 (Advanced codec SD digital television service)
      Service id: 90 (0x005A), Type: 0x16 (Advanced codec SD digital television service)
      Service id: 100 (0x0064), Type: 0x19 (Advanced codec HD digital television service)
      Service id: 101 (0x0065), Type: 0x19 (Advanced codec HD digital television service)
      Service id: 104 (0x0068), Type: 0x02 (Digital radio sound service)
      Service id: 105 (0x0069), Type: 0x02 (Digital radio sound service)
      Service id: 110 (0x006E), Type: 0x19 (Advanced codec HD digital television service)
      Service id: 120 (0x0078), Type: 0x16 (Advanced codec SD digital television service)
      Service id: 146 (0x0092), Type: 0x16 (Advanced codec SD digital television service)
      Service id: 151 (0x0097), Type: 0x16 (Advanced codec SD digital television service)
      Service id: 628 (0x0274), Type: 0x01 (Digital television service)
    - Descriptor 1: unknown (0x83, 131), 76 bytes    <=========== here, unrecognized private descriptor 0x83
      0000:  00 78 FC A7 00 92 FC 2C 00 97 FC 29 00 28  .x.....,...).(
      000E:  FC 40 00 3C FC 46 00 0D FF 02 00 41 FE 3A  .@.<.F.....A.:
      001C:  00 37 FC 2F 00 46 FC 08 00 47 FC 6C 00 48  .7./.F...G.l.H
      002A:  FD FC 00 50 FC 1A 00 5A FC 32 02 74 FF 8C  ...P...Z.2.t..
      0038:  00 64 FC 1C 00 69 FE DD 00 6E FC 9E 00 65  .d...i...n...e
      0046:  FE 10 00 68 FE D8                          ...h..
    - Descriptor 2: Terrestrial Delivery System (0x5A, 90), 11 bytes
      Centre frequency: 690,000,000 Hz, Bandwidth: 8 MHz
      Priority: high, Time slicing: unused, MPE-FEC: unused
[... 43 lines cut ...]
      Service Id:    40 (0x0028), Visible: 1, Channel number:  64
      Service Id:    60 (0x003C), Visible: 1, Channel number:  70
      Service Id:    13 (0x000D), Visible: 1, Channel number: 770
      Service Id:    65 (0x0041), Visible: 1, Channel number: 570
      Service Id:    55 (0x0037), Visible: 1, Channel number:  47
      Service Id:    70 (0x0046), Visible: 1, Channel number:   8
      Service Id:    71 (0x0047), Visible: 1, Channel number: 108
[... cut ...]
      Service Id:   101 (0x0065), Visible: 1, Channel number: 528
      Service Id:   104 (0x0068), Visible: 1, Channel number: 728
    - Descriptor 2: Terrestrial Delivery System (0x5A, 90), 11 bytes
      Centre frequency: 690,000,000 Hz, Bandwidth: 8 MHz
      Priority: high, Time slicing: unused, MPE-FEC: unused
      Constellation pattern: 64-QAM
      Hierarchy: non-hierarchical, native interleaver
      Code rate: high prio: 5/6, low prio: 5/6
      Guard interval: 1/4
      OFDM transmission mode: 8k, other frequencies: yes
~~~

SuppaMan150, 2023-11-09:
Hi,
Thank you very much for the clarification. In fact, I noticed this "issue" when using the tsanalyze output in JSON, which aggregates all the PIDs related to an SID into an array. Relying on that data to demux the various SIDs from a .ts file, I found myself with a file related to the DAB Radio with PID 104 that contained the video of another SID (100 or 101).
Following your advice, I tried using the tstables tool, and it works great because now I can discard that PID (only for SID 104 but not for the other two), resulting in a correct demux.

The only comment I have is that, at least in the textual output of tsanalyze, including the PID in question along with stream data (as in this case, "AVC video (1920x1080, high profile, level 4.0 ecc...") seems misleading because it makes it appear that the SID actually has a video stream (also because the bitrate indicated at the end is not the real one).

I also want to thank you for the tip on using PDS eacem for extracting the LCN; it works great!

Two points regarding the "peculiarities" observed in this case:
- It might not be so unusual for SID 104 to sync with the PID 1101 of TV2000, as they are both channels from the same broadcaster.
- It's a very common practice here in Italy, especially for national broadcasters, to have multiple SIDs for the same channel with the same video and audio streams. This practice is used to have more than one Logical Channel Number (LCN) for the same channel. For example, for TV2000, the LCN numbering is set at 28, but tuning to 528 gets you the same channel. In the past, channels from 501 onwards were used for the HD counterpart of the SD channel, but now they are simply used as duplicates.

lelegard, 2023-11-09:
> The only comment I have is that, at least in the textual output of tsanalyze, including the PID in question along with stream data (as in this case, "AVC video (1920x1080, high profile, level 4.0 [...]

The PID carries AVC video and it is displayed as such, independently of the service context.

We come back to the discussion whether it is appropriate that `tsanalyze` reports the PCR PID of a service when it is not otherwise referenced in the service as a component. This is a design choice. If there is only one good reason, let's try this one: without this, you wouldn't have notice this unusual peculiarity of that service. Right? No other analysis tool reported that...

> It might not be so unusual for SID 104 to sync with the PID 1101 of TV2000, as they are both channels from the same broadcaster.

All right. However, even for a single broadcaster, coupling the encoding of two distinct services creates more problem that is solves. It is a really weird idea to not include PCR in the audio PID of a radio service. They encode a radio service as if it was an audio track of a television service.

> In the past, channels from 501 onwards were used for the HD counterpart of the SD channel, but now they are simply used as duplicates.

OK, so this is the legacy of removed "HS simulcast logical channel number" (which were a pain to correctly interpret, by the way).

closed 2024-04-14 by github-actions[bot]