tsduck/tsduck #1343: [Q] regarding remote control

jangrayogesh, 2023-11-09:
i am unable to establish the connection for the remote control like i am unable to send the commands to udp port 4444
can someone help me with that or you can provide me the code for tsswitch for 2-3 file and i can send the control using udp.

I will be thankful for that.

lelegard, 2023-11-09:
Hi @jangrayogesh 

Please provide more details. Which operating system? Which distro? Which version? Which TSDuck version?

jangrayogesh, 2023-11-10:
i have solved this but i want to ask one more question 
can i add ts files while executing the tsswitch command?

lelegard, 2023-11-10:
If the question is "can we add a new input plugin on the fly to tsswitch, without stopping and restarting it", the answer is no.

closed 2023-11-27 by jangrayogesh