tsduck/tsduck #1344: [FEAT] Pre-Release for debian

enricovittorini, 2023-11-09:
It would be great to have pre-release packages for debian

lelegard, 2023-11-09:
Hi @enricovittorini 

Agreed. However, the nightly builds are created in the GitHub CI/CD environment which does not provide Debian runners, only Ubuntu runners (and Windows and macOS). 

Since rebuilding .deb packages for TSDuck on Debian is as simple as `make installer`, there is no real need to look for a complicated external CD/CD setup for Debian.

enricovittorini, 2023-11-11:
@lelegard I was trying to build tsduck on debian:stable docker could not install prerequisites. To fix them I had to install the following packages:

- sudo
- lsb-release  
- locales 
- locales-all

lelegard, 2023-11-11:
Hi @enricovittorini 

Packages such as `sudo` and the like are basic. I assumed that they are installed everywhere.

For the rest, execute first the script `scripts/install-prerequisites.sh` [as documented here](https://tsduck.io/doxy/building.html).

enricovittorini, 2023-11-11:
@lelegard  on debian:stable docker form dockerhub it is not the case:

```
# sudo
/bin/sh: 1: sudo: not found
# 
```

After installing sudo only, pre-req script does not install anything:

```
root@9be689751d70:/tsduck/scripts# ./install-prerequisites.sh 
root@9be689751d70:/tsduck/scripts# 
```
after installing lsb-release, the script gets executed correcty.

`root@9be689751d70:/tsduck/scripts# ./install-prerequisites.sh 
Packages: git git-lfs g++ cmake flex bison dos2unix curl tar zip linux-libc-dev dpkg-dev python3 libedit-dev pcscd libpcsclite-dev libcurl4 libcurl4-openssl-dev libsrt-openssl-dev librist-dev libusb-1.0-0-dev default-jdk doxygen graphviz
Hit:1 http://deb.debian.org/debian stable InRelease
Hit:2 http://deb.debian.org/debian stable-updates InRelease
Hit:3 http://deb.debian.org/debian-security stable-security InRelease
Reading package lists... Done
Building dependency tree... Done`

Still not enough, without locales-all (indeed no need for "locale" package) there are warnings during the "make"

`root@9be689751d70:/tsduck# make
bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)`

Once isntalled locales-all, all is ok
`root@9be689751d70:/tsduck# make
  [DEP] base/xml/tsxmlElement.cpp
  [DEP] base/xml/tsxmlDocument.cpp
  [DEP] base/xml/tsxmlDeclaration.cpp
  [DEP] base/xml/tsxmlComment.cpp
  [DEP] base/xml/tsxmlAttribute.cpp`

lelegard, 2023-11-11:
Rationale:

- `sudo` : without it, you cannot install anything.
- `lsb-release` : required to identify the system and version, required to decide which packages to install with which command.
- locale during the build : previously, some people had issues with their locale which affected the sort order of some shell commands during the build, therefore the locale of the build process is forced to en_US.UTF8 to enforce a reproductible build.

I would say that a docker container is an execution environment for applications, not a developer's environment. Developers need many tools and a typical full operating system. This is why `scripts/install-prerequisites.sh` assumes a reasonable developer's environment and installs specific tools and libraries which are needed to build TSDuck.

Reading this script, you will see it supports a large number of operating systems and versions. It took years to get there. All addressed configurations were tested at some point, typically in a virtual machine. So, there is no room to add "non-standard" development environments.

enricovittorini, 2023-11-12:
@lelegard thanks.
My goal was to give a clue to those who will have the same issue in the future. Maybe it could be worth to add a note in "how to build" tsduck section.

lelegard, 2023-11-12:
> Maybe it could be worth to add a note in "how to build" tsduck section.

Precisely, there is a complete section about this in the [TSDuck programmer's guide](https://tsduck.io/doxy/): https://tsduck.io/doxy/building.html

This section has been existed for a long time and I often post a link to it, almost every time someone new needs to build TSDuck, for instance to take advantage of a new feature or bug  fix. I hardly see what to add.

ajkq, 2023-11-19:
> @lelegard I was trying to build tsduck on debian:stable docker could not install prerequisites. To fix them I had to install the following packages:
> 
> * sudo
> * lsb-release
> * locales
> * locales-all

You can omit the locales-all package, and instead do this:

```
RUN echo "en_US.UTF-8 UTF-8" >>/etc/locale.gen && \
locale-gen
```

closed 2024-04-21 by github-actions[bot]