tsduck/tsduck #1345: [Q] How to play a specific DVB-T stream via tsduck ?

stulluk, 2023-11-09:
Apologies for my newbie questions. 
I have a 80Gbyte sized TS file. This TS seems to be captured from here: :  https://www.digitalbitrate.com/dtv.php?mux=MUX3-SL&liste=1&live=13&lang=en
I am trying to find right cmdline arguments to play this stream with DTU-315.
I used following arguments:
```bash
stulluk /media/G550 $  tsp --debug -v -I file 722_Vectra_4h.ts -O dektec --pilots --modulation DVB-T --frequency 634000000 --bandwidth 8 --guard-interval 1/4 --convolutional-rate 5/6
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp --debug -v
* Debug: debug level set to 1
* Debug: file: debug level set to 1
* Debug: file: ====> tsp -I file 722_Vectra_4h.ts
* Debug: dektec: debug level set to 1
* Debug: dektec: ====> tsp -O dektec --pilots --modulation DVB-T --frequency 634000000 --bandwidth 8 --guard-interval 1/4 --convolutional-rate 5/6
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: file: detected TS file format TS
* Debug: file: initial buffer load: 44,620 packets, 8,388,560 bytes
* file: unknown initial input bitrate
* dektec: using Dektec device 0, output channel 0 (DTU-315 port 1)
* Debug: dektec: attaching to device DTU-315 serial 0x0000000012C69028
* Debug: dektec: attaching to port 1
* Debug: dektec: resetting channel, mode: 1
* Debug: dektec: setting TxMode, tx: 17, stuff: 0
* dektec: using DVB-T FEC 5/6, bandwidth 8-MHz, constellation 64-QAM, guard 1/4, transmission 8K
* Debug: dektec: DtapiModPars2TsRate(..., 9, 4, 4400180, -1, -1)
* dektec: setting output TS bitrate to 24,882,353 b/s
* Debug: dektec: SetModControl(9, 4, 4400180, -1)
* dektec: setting output carrier frequency to 634,000,000 Hz
* Debug: dektec: SetRfMode(0)
* dektec: output fifo size: 16,777,216 bytes, max: 16,777,216 bytes, typical: 16,777,216 bytes
* Debug: dektec: SetTsRateBps(24882353/1), ie. 24882353
* Debug: dektec: setting TxControl to 2
* dektec: initial output bitrate: 24,882,353 b/s
* dektec: Will preload FIFO before starting transmission. Preload FIFO size: 13,421,696 bytes.
* Debug: file: input thread started
* Debug: dektec: output thread started
* Debug: control commands: debug level set to 1
* Debug: control commands: debug level set to 1
* Debug: control commands: debug level set to 1
* Debug: control commands: debug level set to 1
* Debug: control commands: debug level set to 1
* Debug: control commands: debug level set to 1
* Debug: control commands: found 0 packet processor plugins
* Debug: file: input: got bitrate 50,686,263 b/s
* dektec: DTU-315 output FIFO load is 13,421,696 bytes, starting transmission
* Debug: file: input: got bitrate 50,686,261 b/s
* Debug: file: input: got bitrate 50,686,262 b/s
* Debug: file: input: got bitrate 50,686,262 b/s
* Debug: file: input: got bitrate 50,686,263 b/s
* Debug: file: input: got bitrate 50,686,265 b/s
^C* tsp: user interrupt, terminating...
* Debug: aborting all plugins...
* Debug: aborting plugin file
* Debug: aborting plugin dektec
* Debug: file: stopping the input plugin
* Debug: file: input thread aborted after 624,680 packets
* Debug: dektec: stopping the output plugin
* dektec: terminating DTU-315 output
* Debug: dektec: SetRfMode(1)
* Debug: dektec: detach channel, mode: 0
* Debug: dektec: detach device
* dektec: DTU-315 output terminated
* Debug: dektec: output thread aborted after 580,060 packets (580,060 output)
* Debug: Report logging thread terminated
stulluk /media/G550 $  tsbitrate 722_Vectra_4h.ts 
TS bitrate: 50,686,261 b/s (188-byte), 54,999,985 b/s (204-byte)
stulluk /media/G550 $  tst
tstabcomp   tstabdump   tstables    tstclnt     tsterinfo   tstestecmg  
stulluk /media/G550 $  tsterinfo --bitrate 50686261
  Nominal bitrate ........ 31,668,449.2 b/s
  Bitrate difference ..... 19,017,811.8 b/s
  Bandwidth ................. 8,000,000 Hz
  FEC (high priority) ............. 7/8 
  Constellation ................ 64-QAM 
  Guard interval ................. 1/32 
stulluk /media/G550 $     
stulluk /media/G550 $  tsterinfo --bitrate 50686261 --frequency 634000000
UHF channel: 41, offset: 0
  Nominal bitrate ........ 31,668,449.2 b/s
  Bitrate difference ..... 19,017,811.8 b/s
  Bandwidth ................. 8,000,000 Hz
  FEC (high priority) ............. 7/8 
  Constellation ................ 64-QAM 
  Guard interval ................. 1/32 
stulluk /media/G550 $  
stulluk /media/G550 $  tsp --version
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.36-3358
stulluk /media/G550 $  
```

And sadly, with these arguments, my box is stuttering terribly during playback. I tried `--stuffing` but it didn't helped. Removed `--pilots` , no luck either.

I also tried to use tsanalyse to see if there is a way to find modulation parameters from the original TS file, but I couldn't find.

Can you please guide me, how can I find the right parameters?

konrad-dabek, 2023-11-10:
Reading your file and logs - file seems to be not from DVB-T, but from DVB-C source (and file name suggest this - Vectra is a Polish cable operator).
DVB-T mux from digital bitrate would have 24,8 Mbit/s.
Logs from your attempt are showing that the file has in fact 50 Mbit/s. Stuttering is a result of an attempt to push 50 Mbit/s file through 24,8 Mbit/s bandwidth - at output you will massively lose packets.
I'm suspecting it's DVB-C/256QAM, 8 MHz bandwidth.

lelegard, 2023-11-10:
> streams from Polish DVB-T/T2 muxes would be interesting to put them into TSDuck Streams Repository? If yes, I'll provide them.

Yes, definitely. Thanks.

stulluk, 2023-11-10:
@konrad-dabek thank you so much ! You were correct, I just used following command :

```bash
stulluk /media/G550 $  tsp --debug -v -I file 722_Vectra_4h.ts -O dektec --modulation 256-QAM --frequency 634000000 --bandwidth 8 
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp --debug -v
* Debug: debug level set to 1
* Debug: file: debug level set to 1
* Debug: file: ====> tsp -I file 722_Vectra_4h.ts
* Debug: dektec: debug level set to 1
* Debug: dektec: ====> tsp -O dektec --modulation 256-QAM --frequency 634000000 --bandwidth 8
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: file: detected TS file format TS
* Debug: file: initial buffer load: 44,620 packets, 8,388,560 bytes
* file: unknown initial input bitrate
* dektec: using Dektec device 0, output channel 0 (DTU-315 port 1)
* Debug: dektec: attaching to device DTU-315 serial 0x0000000012C69028
* Debug: dektec: attaching to port 1
* Debug: dektec: resetting channel, mode: 1
* Debug: dektec: setting TxMode, tx: 17, stuff: 0
* Debug: dektec: SetModControl(8, 2, 0, 0)
* dektec: setting output carrier frequency to 634,000,000 Hz
* Debug: dektec: SetRfMode(0)
* dektec: output fifo size: 16,777,216 bytes, max: 16,777,216 bytes, typical: 16,777,216 bytes
* Warning: dektec: no input bitrate is available, use --bitrate in case of output error
* Debug: dektec: setting TxControl to 2
* dektec: initial output bitrate: 0 b/s
* dektec: Will preload FIFO before starting transmission. Preload FIFO size: 13,421,696 bytes.
* Debug: file: input thread started
* Debug: dektec: output thread started
* Debug: control commands: debug level set to 1
* Debug: control commands: debug level set to 1
* Debug: control commands: debug level set to 1
* Debug: control commands: debug level set to 1
* Debug: control commands: debug level set to 1
* Debug: control commands: debug level set to 1
* Debug: control commands: found 0 packet processor plugins
* Debug: file: input: got bitrate 50,686,263 b/s
* Debug: dektec: SetTsRateBps(50686263/1), ie. 50686263.17
* dektec: new output bitrate: 50,686,263 b/s
* dektec: DTU-315 output FIFO load is 13,421,696 bytes, starting transmission
* Debug: file: input: got bitrate 50,686,262 b/s
* Debug: dektec: SetTsRateBps(50686262/1), ie. 50686261.98
* dektec: new output bitrate: 50,686,262 b/s
* Debug: file: input: got bitrate 50,686,263 b/s
* Debug: dektec: SetTsRateBps(50686263/1), ie. 50686263.27
* dektec: new output bitrate: 50,686,263 b/s
* Debug: file: input: got bitrate 50,686,267 b/s
* Debug: dektec: SetTsRateBps(50686267/1), ie. 50686267.01
* dektec: new output bitrate: 50,686,267 b/s
* Debug: file: input: got bitrate 50,686,269 b/s
* Debug: dektec: SetTsRateBps(50686269/1), ie. 50686268.92
...
```
And it works like charm.

Sadly, I can't share this TS for two reasons:
1) This is non-public
2) it is 80Gbyte

However, if you can share some TS from MUX3 , I can try play it. It would be great.

konrad-dabek, 2023-11-10:
Happy to hear that you have found the right config.
For DVB-T/T2 Polish streams - I've sent them to Thierry. If files will be fine, they will show up soon in TSDuck Stream Repository, so you (and anybody else interested) will have a chance to check them out ;)

stulluk, 2023-11-10:
@konrad-dabek thank you. 
@lelegard  In the TSduck stream repository, some of the directories contain shell scripts which shows modulation parameters, they are quite useful to see playback parameters. I would be great to have this on all directories, since I couldn't find a way to find modulation params from the TS files.
Please correct me if I am wrong.
Additionally, I can also provide some "public" TS files, captured from Turksat & hotbird Satellites. If needed, just let me know and I can upload it somewhere.

Finally, closing this issue since it is solved.

lelegard, 2023-11-11:
Hi @stulluk 

> I would be great to have this on all directories, since I couldn't find a way to find modulation params from the TS files.

I added the capture scripts for the few streams I captured in France, terrestrial or satellite. The other files come from various contributors, including @konrad-dabek today. So, the presence of capture commands in shell scripts really depends on the contributor. 

However, the captures scripts were valid only the day of the capture. Every day, some service plan update occurs here and there, changing the modulation parameters or frequency. So, applying the same parameters later may not work anyway.

For DVB-T networks, use `tsscan` to scan your area. With DVB-T, there is no need to know the exact modulation parameters. Most tuners find them by themselves as long as the frequency is fine. This does not work with T2 with multiple PLP, though.

For satellite networks, have a look at https://www.lyngsat.com/ or similar sites. Some enthusiasts update the references of all satellites from virtually everywhere.

To translate this into TSDuck command line parameters, there is a PowerShell script named `get-tuning-parameters-from-lyngsat.ps1` in the [TSDuck sample area](https://github.com/tsduck/tsduck/tree/master/sample/lyngsat).

> Additionally, I can also provide some "public" TS files, captured from Turksat & hotbird Satellites. If needed, just let me know and I can upload it somewhere.

New streams are welcome. I have no public upload area (too dangerous for DoS). I recommend to use public transfer services such as https://wetransfer.com or equivalent.

stulluk, 2023-11-11:
@konrad-dabek  @lelegard  thank you so much for such detailed description.

I will try to provide similar TS files from Turksat & Hotbird, including 4K UHD FTA or Scrambled ones, adding capture & playback commands for reproducability.

stulluk, 2023-11-11:
@lelegard  is there a filesize limit or recommended size for TS files to be shared with you ?

lelegard, 2023-11-11:
> is there a filesize limit or recommended size for TS files to be shared with you ?

It depends on the bitrate of the TS. Between 1 and 2 minutes of contents is sufficient to get all relevant information. Of course, with high-bandwidth TS, 1 minute is probably enough to avoid huge files.

stulluk, 2024-05-10:
Dear @lelegard  @konrad-dabek , I would like to visit here after long time.

I am having a difficulty with dektec plugin.

Here is my environment: I have a DTU-315 and TBS5530. Both of them are connected to two separate USB ports of my PC. They are connected to each other with pro-quality cable. ( Using DVB-T Tuner of the TBS5530, not DVB-S )

And I am playing https://tsduck.io/streams/poland-dttv-warsaw/PL_DVB-T_2023-11-10_MUX3_522MHz.ts via DTU-315 with following command:

DVB-T Playback with DTU-315:

```bash
stulluk ~ $  tsp --debug -I file /media/WINDOWS/TS-FILES/PL_DVB-T_2023-11-10_MUX3_522MHz.ts --infinite -O dektec --modulation DVB-T --frequency 522000000 --bandwidth 8 --guard-interval 1/8 --transmission-mode 8K 
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp --debug
* Debug: debug level set to 1
* Debug: file: debug level set to 1
* Debug: file: ====> tsp -I file /media/WINDOWS/TS-FILES/PL_DVB-T_2023-11-10_MUX3_522MHz.ts --infinite
* Debug: dektec: debug level set to 1
* Debug: dektec: ====> tsp -O dektec --modulation DVB-T --frequency 522000000 --bandwidth 8 --guard-interval 1/8 --transmission-mode 8K
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: file: detected TS file format TS
* Debug: file: initial buffer load: 44,620 packets, 8,388,560 bytes
* file: initial input bitrate is 24,882,302 b/s
* dektec: using Dektec device 0, output channel 0 (DTU-315 port 1)
* Debug: dektec: attaching to device DTU-315 serial 0x0000000012C69028
* Debug: dektec: attaching to port 1
* Debug: dektec: resetting channel, mode: 1
* Debug: dektec: setting TxMode, tx: 17, stuff: 0
* dektec: using DVB-T FEC 3/4, bandwidth 8-MHz, constellation 64-QAM, guard 1/8, transmission 8K
* Debug: dektec: DtapiModPars2TsRate(..., 9, 2, 4399924, -1, -1)
* dektec: setting output TS bitrate to 24,882,353 b/s
* Debug: dektec: SetModControl(9, 2, 4399924, -1)
* dektec: setting output carrier frequency to 522,000,000 Hz
* Debug: dektec: SetRfMode(0)
* dektec: output fifo size: 16,777,216 bytes, max: 16,777,216 bytes, typical: 16,777,216 bytes
* Debug: dektec: SetTsRateBps(24882353/1), ie. 24882353
* Debug: dektec: setting TxControl to 2
* dektec: initial output bitrate: 24,882,353 b/s
* dektec: Will preload FIFO before starting transmission. Preload FIFO size: 13,421,696 bytes.
* Debug: control commands: debug level set to 1
* Debug: control commands: debug level set to 1
* Debug: control commands: debug level set to 1
* Debug: control commands: debug level set to 1
* Debug: control commands: debug level set to 1
* Debug: control commands: debug level set to 1
* Debug: file: input thread started
* Debug: control commands: found 0 packet processor plugins
* Debug: dektec: output thread started
* dektec: DTU-315 output FIFO load is 13,421,696 bytes, starting transmission
* Debug: file: input: got bitrate 24,882,298 b/s
* Debug: file: input: got bitrate 24,882,297 b/s
* Debug: file: input: got bitrate 24,882,296 b/s
* Debug: file: input: got bitrate 24,882,296 b/s
* Debug: file: input: got bitrate 24,882,296 b/s
^C* tsp: user interrupt, terminating...
* Debug: aborting all plugins...
* Debug: aborting plugin file
* Debug: aborting plugin dektec
* Debug: file: stopping the input plugin
* Debug: file: input thread aborted after 624,680 packets
* Debug: dektec: stopping the output plugin
* dektec: terminating DTU-315 output
* Debug: dektec: SetRfMode(1)
* Debug: dektec: detach channel, mode: 0
* Debug: dektec: detach device
* dektec: DTU-315 output terminated
* Debug: dektec: output thread aborted after 580,060 packets (580,060 output)
* Debug: Report logging thread terminated
stulluk ~ $  
```
[... 81 lines cut ...]
* Debug: control commands: found 0 packet processor plugins
* Debug: dvb: input: got bitrate 24,882,353 b/s
* Debug: dvb: input: got bitrate 24,882,353 b/s
* Debug: dvb: input: got bitrate 24,882,353 b/s
* Debug: dvb: input: got bitrate 24,882,353 b/s
^C* tsp: user interrupt, terminating...
* Debug: aborting all plugins...
* Debug: aborting plugin dvb
* Debug: aborting plugin file
* Debug: file: stopping the output plugin
* Debug: file: output thread aborted after 370,960 packets (370,960 output)
* Debug: dvb: stopping the input plugin
* Debug: dvb: input thread aborted after 371,960 packets
* Debug: Report logging thread terminated
stulluk ~ $  

```

So the bitrate from the recording side is correct. ( 24,882,353 b/s)

I also verified this by using another Windows PC, and playing this TS file via StreamXPress. Whole file played perfectly with MPV and VLC. StreamXpress also shows the bitrate as "24,882,353 b/s".

So the problem is in my DTU-315 playback side.

Can you please guide me, what am I missing in the playback command for this Poland 522Mhz stream ?

stulluk, 2024-05-10:
And a parallel question:

While playing 522Mhz via DTU-315 , on the other terminal, I use "tsscan" to find actual modulation parameters, but it reports guard-interval incorrectly and signal is not locked ( even when I forcefully pass GI to 1/8):

```bash
stulluk ~ $  tsscan --debug -d /dev/dvb/adapter0:0:0:0 --uhf-band --show-modulation 
tsscan: Debug: debug level set to 1
tsscan: Debug: ====> tsscan --debug -d /dev/dvb/adapter0:0:0:0 --uhf-band --show-modulation
tsscan: Debug: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/dvb/dvb0.frontend0/device/idVendor = "734c" (success)
tsscan: Debug: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/dvb/dvb0.frontend0/device/idProduct = "5530" (success)
tsscan: Debug: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/dvb/dvb0.frontend0/device/manufacturer = "TBS-Tech" (success)
tsscan: Debug: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/dvb/dvb0.frontend0/device/product = "TBS 5530" (success)
tsscan: Debug: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/dvb/dvb0.frontend0/device/version = "2.00" (success)
tsscan: Debug: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/dvb/dvb0.frontend0/device/serial = "" (failure)
scanning channel 21, 474,000,000 Hz
tsscan: Debug: trying offset 0
tsscan: Debug: using DVB-T as default delivery system
tsscan: Debug: adding standards DVB to none
tsscan: Debug: starting discarding frontend events
tsscan: Debug: finished discarding frontend events
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 1 DTVProperties:
tsscan: Debug: [0] cmd = 2 (DTV_CLEAR), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 11 DTVProperties:
tsscan: Debug: [0] cmd = 17 (DTV_DELIVERY_SYSTEM), data = 3 (0x00000003)
tsscan: Debug: [1] cmd = 3 (DTV_FREQUENCY), data = 474000000 (0x1C40AA80)
tsscan: Debug: [2] cmd = 4 (DTV_MODULATION), data = 3 (0x00000003)
tsscan: Debug: [3] cmd = 6 (DTV_INVERSION), data = 2 (0x00000002)
tsscan: Debug: [4] cmd = 5 (DTV_BANDWIDTH_HZ), data = 8000000 (0x007A1200)
tsscan: Debug: [5] cmd = 36 (DTV_CODE_RATE_HP), data = 9 (0x00000009)
tsscan: Debug: [6] cmd = 37 (DTV_CODE_RATE_LP), data = 9 (0x00000009)
tsscan: Debug: [7] cmd = 39 (DTV_TRANSMISSION_MODE), data = 1 (0x00000001)
tsscan: Debug: [8] cmd = 38 (DTV_GUARD_INTERVAL), data = 0 (0x00000000)
tsscan: Debug: [9] cmd = 40 (DTV_HIERARCHY), data = 0 (0x00000000)
tsscan: Debug: [10] cmd = 1 (DTV_TUNE), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: no input signal lock after 5,000 milliseconds
scanning channel 22, 482,000,000 Hz
tsscan: Debug: trying offset 0
tsscan: Debug: using DVB-T as default delivery system
tsscan: Debug: starting discarding frontend events
tsscan: Debug: one frontend event discarded
tsscan: Debug: one frontend event discarded
tsscan: Debug: finished discarding frontend events
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 1 DTVProperties:
tsscan: Debug: [0] cmd = 2 (DTV_CLEAR), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 11 DTVProperties:
tsscan: Debug: [0] cmd = 17 (DTV_DELIVERY_SYSTEM), data = 3 (0x00000003)
tsscan: Debug: [1] cmd = 3 (DTV_FREQUENCY), data = 482000000 (0x1CBABC80)
tsscan: Debug: [2] cmd = 4 (DTV_MODULATION), data = 3 (0x00000003)
tsscan: Debug: [3] cmd = 6 (DTV_INVERSION), data = 2 (0x00000002)
tsscan: Debug: [4] cmd = 5 (DTV_BANDWIDTH_HZ), data = 8000000 (0x007A1200)
tsscan: Debug: [5] cmd = 36 (DTV_CODE_RATE_HP), data = 9 (0x00000009)
tsscan: Debug: [6] cmd = 37 (DTV_CODE_RATE_LP), data = 9 (0x00000009)
tsscan: Debug: [7] cmd = 39 (DTV_TRANSMISSION_MODE), data = 1 (0x00000001)
tsscan: Debug: [8] cmd = 38 (DTV_GUARD_INTERVAL), data = 0 (0x00000000)
tsscan: Debug: [9] cmd = 40 (DTV_HIERARCHY), data = 0 (0x00000000)
tsscan: Debug: [10] cmd = 1 (DTV_TUNE), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: no input signal lock after 5,000 milliseconds
scanning channel 23, 490,000,000 Hz
tsscan: Debug: trying offset 0
tsscan: Debug: using DVB-T as default delivery system
tsscan: Debug: starting discarding frontend events
tsscan: Debug: one frontend event discarded
tsscan: Debug: one frontend event discarded
tsscan: Debug: finished discarding frontend events
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 1 DTVProperties:
[... 205 lines cut ...]
tsscan: Debug: starting discarding frontend events
tsscan: Debug: one frontend event discarded
tsscan: Debug: one frontend event discarded
tsscan: Debug: finished discarding frontend events
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 1 DTVProperties:
tsscan: Debug: [0] cmd = 2 (DTV_CLEAR), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 11 DTVProperties:
tsscan: Debug: [0] cmd = 17 (DTV_DELIVERY_SYSTEM), data = 3 (0x00000003)
tsscan: Debug: [1] cmd = 3 (DTV_FREQUENCY), data = 546000000 (0x208B4C80)
tsscan: Debug: [2] cmd = 4 (DTV_MODULATION), data = 3 (0x00000003)
tsscan: Debug: [3] cmd = 6 (DTV_INVERSION), data = 2 (0x00000002)
tsscan: Debug: [4] cmd = 5 (DTV_BANDWIDTH_HZ), data = 8000000 (0x007A1200)
tsscan: Debug: [5] cmd = 36 (DTV_CODE_RATE_HP), data = 9 (0x00000009)
tsscan: Debug: [6] cmd = 37 (DTV_CODE_RATE_LP), data = 9 (0x00000009)
tsscan: Debug: [7] cmd = 39 (DTV_TRANSMISSION_MODE), data = 1 (0x00000001)
tsscan: Debug: [8] cmd = 38 (DTV_GUARD_INTERVAL), data = 0 (0x00000000)
tsscan: Debug: [9] cmd = 40 (DTV_HIERARCHY), data = 0 (0x00000000)
tsscan: Debug: [10] cmd = 1 (DTV_TUNE), data = 4294967295 (0xFFFFFFFF)
^C
stulluk ~ $  
```

Can you please guide me for that , what is the correct way to find out modulation params in an unknown network with tsscan ?

lelegard, 2024-05-11:
Hi @stulluk 

First, please do not reopen old issues, closed six months ago. If you have a new question, like this one, please open a new issue. Reusing old issues for new questions (even if somewhat related to an old one) makes issue tracking and referencing difficult.

There are two questions here, 1) the differences in bitrate, 2) the reported modulation parameters.

### Bitrate

The nominal bitrate for the given modulation parameters is 24,882,352.94 b/s (let's say 24,882,353). In the first command, the input is a file. Assuming that it was captured from a real live stream with the same modulation parameters, its live bitrate was 24,882,353 b/s at the time it was captured.

When playing a TS file, tsp computes the bitrate from the PCR values and the number of packets between them. An average of all PID's with PCR's is used. And the measured bitrate is initially 24,882,302, then 24,882,296, and later oscillating +/- 1. The oscillation is due to the discrete nature of the measurement. We measure TS packets (1504 bits at a time), not bits. So, there is always some approximation. Any difference less than 1504 just means one packet. In your case, you have a deficit of 56 bits per second, equivalent to one packet missing every 26 seconds.

So, during the broadcast capture, there was an average of one missed packet every 26 seconds. Not that bad for a 2-minute capture.

On output, because of the deterministic relationship between modulation parameters and output bitrate, the Dektec modulator exactly broadcasts at 24,882,353 b/s and inserts null packets when the input is starving (in the case of a real-time input). Since the input is a file, the modulator is never starving (there is always some available data from the file) and the stream is played at a slightly faster rate than expected from the PCRs. But a difference of 56 b/s is only a variation of 0.0002 %. So, who cares?

### Reporting modulation parameters

This is entirely dependent on the tuner device, its driver, and the operating system.

From my experience, when you setup a tuner, providing the correct modulation parameters may matter or not. Typically, for DVB-S reception, if you provide one single incorrect modulation parameter, you receive nothing. With DVB-T, on the other hand, providing the right frequency is usually sufficient. If other parameters are incorrect, the tuner will automatically adapt.

Now, the question is: how can we get those correct modulation parameters that the tuner found by itself? And the answer is: it depends...

When I worked on terrestrial TV, between 2007 and 2011 (a bit old now), I noticed that reading the current modulation parameters from the tuner driver gave variable results:
- On Windows, the "current" values were never correct. It always returned the (possibly incorrect) input modulation parameters which were specified on the command line. So, this was typically a Windows DirectShow issue. And Windows is hopeless, as usual...
- On Linux, most of the time, the "current" values, as returned by the tuner driver, were correct. The hardware of the tuner was able to report them to the driver and the driver was able to pass them back to the application.
- However, with a limited number of tuners, the returned values were incorrect, even on Linux. This was linked to the tuner model or its driver. This means that, for a given tuner, the results were either always correct, or always incorrect.

This dates back from 15 years ago. In the meantime, I have noticed that more and more newer tuners were unable to report the correct current tuner parameters, even on Linux. It seems that new developers, either on the tuner hardware or the software drivers, are more and more lazy and don't care about reporting the actual modulation parameters.

In your case, `tsscan --show-modulation` means that `tsscan` queries the tuner each time a signal is found to get the actual tuning parameters. So, quite probably, the incorrect parameters you get are simply the result of the tuner or driver being unable to report them.

At the time I worked on the subject, because Windows was unable to report these parameters, I added the capability to "guess" what could be the possible DVB-T parameters for a given bitrate.

In you case:
~~~
$ tsterinfo --bitrate 24,882,353
  Nominal bitrate ....... 24,882,352.94 b/s
  Bitrate difference ............. 0.06 b/s
  Bandwidth ................. 8,000,000 Hz
  FEC (high priority) ............. 3/4 
  Constellation ................ 64-QAM 
  Guard interval .................. 1/8 

  Nominal bitrate ....... 24,882,352.94 b/s
  Bitrate difference ............. 0.06 b/s
  Bandwidth ................. 8,000,000 Hz
  FEC (high priority) ............. 5/6 
  Constellation ................ 64-QAM 
  Guard interval .................. 1/4 
~~~

You see that there is not always one single response. In that case, `tsterinfo` gives all possible combinations. In your case, the right answer is the first one.

stulluk, 2024-05-11:
Hi @lelegard 

Thank you very much for your detailed answer. I thought that this question was a followup of the previous one, because I still couldn't be able to play given TS streams via dektec plugin succesfully.

But it is OK, I will open a new ticket, and continue there, no problem.

closed 2024-05-11 by stulluk