tsduck/tsduck #1347: Is it possible to use C++11 to achieve this

bigbao9494, 2023-11-13:
Is it possible to use C++11 to achieve this？
thanks alot

lelegard, 2023-11-13:
?????????

bigbao9494, 2023-11-14:
Sorry for not clarifying. The current code implementation uses C++98. Is it possible to modify it to C++11

lelegard, 2023-11-14:
> The current code implementation uses C++98. Is it possible to modify it to C++11

The "current code" of what? This is the TSDuck repository. TSDuck now requires C++11. In the last years, several phases of refactoring were in progress to use C++11 patterns. The original threading classes are still there because the semantics of the C++11 `std::thread` is too different for direct rewrite. Maybe some day, but the amount of testing and the risk to introduce bugs are high.

bigbao9494, 2023-11-14:
I understand. Thank you very much

lelegard, 2023-11-14:
To be clear, TSDuck _is_ C++11. It uses a lot of meta programming, loops with implicit iterator, etc. It already uses a few C++17 features such as `std::optional`, using a temporary home-cooked implementation if you compile with C++11 or C++14.

bigbao9494, 2023-11-15:
Sorry, I don't quite understand the meaning of 'using a temporary home-cooked implementation if you compile with C++11 or C++14.' here. Can I just compile using C++11? Because some environments only have C++11.

lelegard, 2023-11-15:
Over time, the TSDuck code base is modernized. It started in 2005. Now, it only compiles with C++11, taking advantages of C++11 new features. One day, it will only compile with C++17, then C++20, etc. because new interesting features are standardized.

Previously, TSDuck had an internal class named `ts::Variable` which was basically the same thing as `std::optional` in C++17. I recently changed the code to use `std::optional`. It is not better than `ts::Variable`, it is just standard. However, as a temporary measure, we can still compile using C++11. During compilation, the C++ level is checked. If it is C++17 or higher, the standard `#include <optional>` is used. If it is C++11 or C++14, a custom class named `std::optional` is defined in TSDuck with the same interface and is essentially based on the former implementation of `ts::Variable`.

The day I decide to require C++17, I just drop the custom class named `std::optional`.

In the makefile, the version of the compiler is checked, gcc or clang. If the versions are recent enough, `-std=c++17` is used. Otherwise, `-std=c++11` is used.

bigbao9494, 2023-11-16:
Thank you very much, there is one more question:
1. Has the content in the base/system been replaced with C++11? Except for ts:: thread
2. So far, I can also force only C++11 to be used for compilation in my environment (if C++17 is supported)

lelegard, 2023-11-16:
> Has the content in the base/system been replaced with C++11? Except for ts:: thread

Not everything in base/system has C++11 equivalent, far from that. The strategy is very pragmatic. TSDuck needs some system services. If the current C++ standard library provides that service at the time it is needed by TSDuck, we use it. Otherwise, we develop it. Later, if new versions of the C++ standard provide an equivalent service, migration is considered.

> So far, I can also force only C++11 to be used for compilation in my environment (if C++17 is supported)

Use `make CXXFLAGS_STANDARD=-std=c++11`. Or 14, or 20, or whatever you want, as long as it is not less than 11. 

Currently, tests are performed in the CI/CD pipelines for C++11 and C++20. It you use a more recent standard and get some compilation issue, please report it here. The idea is to always support the most recent standard and drop old standards when it becomes a pain to support them.

bigbao9494, 2023-11-17:
thank you very much.

lelegard, 2023-11-18:
@bigbao9494, since you seem interested in C++11 features, I have a question for you.

As I mentioned, there is a long background refactoring process in progress to move some system-specific TSDuck classes to standard C++11 or even C++17 classes. Concerning multithreading, I have recently completed the migration of mutexes and condition variables to their counterparts in C++11. I am currently evaluating the migration of `ts::Thread` to `std::thread`. The large differences in semantics do not allow a direct migration. However, if the implementation of `ts::Thread` could be made portable, using `std::thread` internally, it would be an improvement.

However, I realize that `std::thread` is very limited, to a point where it becomes unusable in some situations.

Sometimes, you need to set attributes to a thread. Typically, only two properties are used in TSDuck: thread priority and thread stack size. There is no portable interface to set these properties in `std::thread`. There is a `native_thread()` method which can be used in non-portable code to set something on the thread. However, it is mostly useless. In practice, `std::thread` starts the thread in its constructor. So, you can only get `native_thread()` _after_ the thread started, when it is too late to set properties such as stack size.

I have read a lot of discussions on this, including on stackoverflow.com, a site I usually respect. There is a general consensus on "not possible". Some dogmatic people even dare to say that there should be no need to set a stack size, that this is a symptom of bad coding. These guys are obviously short-minded and lack of real-world coding experience. When you have to call a given library in a thread and your thread fails with a stack overflow, you just have to increase the stack size. Full stop. If `std::thread` uses the default size of 128 kB or 512 kB for each thread and you realize that the library you call needs 800 kB, no less, no more, you simply need to set a specific stack size.

So, @bigbao9494, do you know a way to specify a given stack size to a `std::thread`? Non-portable methods are acceptable as long as they exist on Windows, Linux, macOS and BSD.

If no such way exist, then the response to your original question _"is it possible to use C++11 to achieve this"_ is simply _"no it is not possible"_.

bigbao9494, 2023-11-18:
I'm glad to receive your careful answer. I understand what you mean, and it seems that the current design is the most reasonable. It's because my previous understanding of the code was insufficient and I thought the problem was too simple.

closed 2024-04-21 by github-actions[bot]