tsduck/tsduck #1348: [Q] Adding srt statistics pull request - notification mail error

enricovittorini, 2023-11-13:
I have added some line to tsSRTSocket.cpp file to report more statistics:

i.e:
`root.query(u"receive.interval", true).add(u"rate-mbps", stats.mbpsRecvRate);`

I submitted a pull request (#1346) but the notification mail reports that the compilation failed on some platform. basically all the stats that are "double" fails:

```
base/network/tsSRTSocket.cpp:1185:75: error: implicit conversion turns floating-point number into integer: 'double' to 'int64_t' (aka 'long') [-Werror,-Wfloat-conversion]
            root.query(u"receive.interval", true).add(u"rate-mbps", stats.mbpsRecvRate);
```

It compiled well on my debian docker with gcc version 12.2.0 (Debian 12.2.0-14)

lelegard, 2023-11-13:
Hi @enricovittorini 

GCC silently ignores downgrade of double floating point to integer while clang and msvc report a warning about the loss of precision. With TSDuck, all warnings are turned into fatal errors because many warnings are symptoms of bugs. So, all warnings shall be examined and _correctly_ fixed before the compilation passes. 

We also use most combinations of three different compilers, on three operating systems, with two different revisions of the C++ language to maximize to probability to detect potential problems in advance. 

In your case, the errors highlight the fact that the json implementation of TSDuck does not support floats yet. This is something I wanted to add a long time ago but forgot to do so. 

So, I will add json float support and merge your PR. I will handle this.

enricovittorini, 2023-11-13:
thanks @lelegard 
have a look at the message of my comit regarding the naming of the srt statistics in the json output. 
to me it would make more sense to keep the original srt name of the stats as the "key" in the json object.

lelegard, 2023-11-13:
Hi @enricovittorini 

PR #1346 has been merged.

> it would make more sense to keep the original srt name of the stats as the "key" in the json object.

Not really, for the following reasons:

1. This is a developer's perspective, not a user one.
2. For existing fields, it would break existing scripts using the JSON output.
3. The `SRT_TRACEBSTATS` structure is flat, with complex cryptic names. The JSON structure is hierarchical with send/receive and total/interval/instant levels.

closed 2024-04-14 by github-actions[bot]