tsduck/tsduck #1351: [Q] tsduck live probe

enricovittorini, 2023-11-13:
hello @lelegard, 
i would like to build a live probe with tsduck. I am not a developer but jsut trying to implement something with some coding. 
I would like to mention in my code that it is powered by tsduck. I am adding in the HTML footer a "credits" link that shows a modal with the links to tsduck website and license:

[image]

[image]

just wanted to check with you if this is fine for you.

thanks

lelegard, 2023-11-13:
I see why you needed more detailed SRT stats 😉

> just wanted to check with you if this is fine for you.

No problem at all. I carefully selected a very liberal license to allow the reuse of TSDuck in all circumstances. I hate restrictive licenses such as GPL which are compatible with virtually nothing except themselves. 

The BSD license is very simple. Just follow the two clauses, basically meaning you just have to mention the original TSDuck copyright text. As an example, the file `OTHERS.txt` in TSDuck contains the copyright of other libraries which are used by TSDuck. 

It looks like a nice tool. If it's useful to you, it may be useful to others (which is the story of TSDuck). I can only encourage you to publish it as open source.

EvariSTC, 2023-11-29:
@enricovittorini  please keep us updated

enricovittorini, 2023-11-29:
@EvariSTC I am progressing step by step... I am not a developer but trying to build something nice.

Things are progressing... 

[image]

closed 2024-04-28 by github-actions[bot]