tsduck/tsduck #1355: [BUG] Previous ECM is gone when scramble other service.

sichunqin, 2023-11-15:
#### Bug description:
[BUG] Previous ECM is gone when scramble other service.  

#### How to reproduce:
1. Scramble one service with fixed CW
2. Insert ECM for the service using inject plugin 
3. Scrambler another different service from the same TS file
4. Then the ECM for the first scrambled service is gone in the final TS file. 

I use the following script to perform the above actions: 

```
tsp -v -I file tnt-uhf25-506MHz-2019-01-22.ts -P scrambler 513 --dvb-csa2 --cw-file cw-dvb-csa2.txt -O file tnt1.ts.scr

tsanalyze --pid-list  tnt1.ts.scr
0 1 16 17 18 20 21 110 120 130 131 132 140 141 170 310 320 330 340 410 420 430 440 470 510 520 530 531 540 541 570 610 620 630 631 632 640 641 670 671 8191

tsp -I file tnt1.ts.scr  -P inject --binary tools/ecm_sec_csa2.bin --pid 790 --inter-packet 10 --stuffing -P pmt --add-ca-descriptor 57069/790 --service 513 -O file tnt1.ts.scr.ecm

tsanalyze --pid-list tnt1.ts.scr.ecm
0 1 16 17 18 20 21 110 120 130 131 132 140 141 170 310 320 330 340 410 420 430 440 470 510 520 530 531 540 541 570 610 620 630 631 632 640 641 670 671 790 8191

tsp -v -I file tnt1.ts.scr.ecm   -P scrambler 515 --dvb-csa2 --cw-file cw-dvb-csa2.txt  -O file tnt1.ts.scr.ecm.scr
tsanalyze --pid-list tnt1.ts.scr.ecm.scr
0 1 16 17 18 20 21 110 120 130 131 132 140 141 170 310 320 330 340 410 420 430 440 470 510 520 530 531 540 541 570 610 620 630 631 632 640 641 670 671 8191
```
   
For output of tsanalyze in last step, the ECM with PID 790 get lost from the last TS file. 

I am not sure if I miss something.  I don't why the original ECM get removed. 
Looking forward to your reply! Thanks

#### Expected behavior:

#### Errors and logs:

~~~
[be sure to surround verbatim log lines with "~~~" lines]
~~~

#### Environment:
 - OS: 
 - OS version: 
 - TSDuck full version: 
 - Installation type: 

#### Additional information:

lelegard, 2023-11-15:
Hi @sichunqin 

This is the classical problem of the lack of free space (stuffing) when trying to insert something, here the ECM.

Because of the file name, I assume here that the input stream you use for your tests was downloaded from the TSDuck stream repository, in the French terrestrial section.

First point: If you analyze that stream, you will see that there is no stuffing at all.

Note: All broadcast streams have some stuffing because the TS bitrate is fixed by the modulation parameters and the sum of all ES is never exactly that precise bitrate. However, the DVB-T tuner I used for capturing that stream has the very bad idea of dropping all null packets. It can make sense to watch TV on the computer. But it is very inconvenient for analysis. Anyway, the result is a TS file without any stuffing, no PID 8191 at all.

The second point to understand is how the `scrambler` plugin works when specifying a service (i.e. not a list of PID's on the command line). Scrambling a service means that no packet from that service shall remain in the clear. However, the precise list of PID's of the service is unknown until we get a PAT and then the PMT of the service to scramble. To avoid letting in the clear potential packets from the service to scramble, the `scrambler` plugin nullifies all TS packets before the PMT of the service. As a result, all packets at the beginning of the file are null packets (except global PID's such as PAT, CAT, etc. which cannot be part of the service to scramble).

Knowing these two points, what happens is the following scenario:

- File `tnt-uhf25-506MHz-2019-01-22.ts` contains no stuffing at all. If you had tried `tsanalyze --pid-list` on it, you would have seen that PID 8191 was _not_ present.
- After scrambling service 513, most packets at the beginning of `tnt1.ts.scr` are null packets. If you analyze this file, you will see that the first packet which is not null or a global PID is precisely the 513 PMT. This is the point where the `scrambler` plugin exactly knows what to scramble and what to pass. So, `tnt1.ts.scr` now has some stuffing, some free space, but only at the beginning of the file, before the first 513 PMT.
- In file `tnt1.ts.scr.ecm`, the ECM were inserted where there is some free space, meaning only at the beginning of the file. If you analyze the file, you will see that the ECM PID disappears after the first 513 PMT. Because there was no free space after that PMT. By the way, this also means that the ECM PID is unusable by receivers since the ECM PID disappears at the exact point where it becomes known as an ECM PID.
- During scrambling of the second service, the same initial nullification method applies. All packets before the PMT of service 515 are nullified. However, because the first scrambling operation erased everything, including the 515 PMT, before the 513 PMT, the first 515 PMT is now after the first 513 PMT. This means that the nullification process erases all ECM in PID 790 (because they are all before the 513 PMT).

Therefore, PID 790 was completely erased from file `tnt1.ts.scr.ecm.scr`.

Rule of thumb: When you insert something in a stream, ALWAYS make sure that you have enough free space (stuffing) in that stream.

To avoid this, make sure to have some free space in the initial input file, for instance using option `--add-input-stuffing` in the first `tsp` command.

For more details, please carefully read the documentation of `tsp` in the TSDuck user's guide.

sichunqin, 2023-11-17:
@lelegard Thanks a lot for your great explanation! It really help!

closed 2023-11-17 by sichunqin