tsduck/tsduck #1356: [FEAT] Reading of caption_service_descriptor from PMT

bbgdzxng1, 2023-11-16:
#### Is your feature request related to a problem?

I have been successfully using a TSDuck patch file to write caption_service_descriptor() data to the Program Mapping Table of an MPEG-TS file, in accordance with ATSC A/65.

``` xml
$ less "./patch.xml"

<tsduck>
  <!-- Match the PMT which has service_id="0x0001" -->
  <PMT service_id="0x0001">
    <!-- Match the component which has elementary_PID="0x0100" -->
    <component elementary_PID="0x0100">
      <!-- Add a caption_service_descriptor() using TSDuck's named key value pairs -->
      <caption_service_descriptor x-node="add">
        <service language="eng" digital_cc="false" line21_field="0" easy_reader="false"
          wide_aspect_ratio="false" />
      </caption_service_descriptor>
    </component>
  </PMT>
</tsduck>
```

This patch file works great on any TS file where the Video PID is `0x0100` and can be added to a 1 second TS file easily generated by FFmpeg with:

``` bash
$ ffmpeg -hide_banner \
  -f 'lavfi' -i smptebars=size='ntsc':rate='ntsc',setdar=ratio='(4/3)' \
  -f 'lavfi' -i sine=sample_rate=48000 \
  -map 0:v:0 \
  -codec:v 'mpeg2video' \
  -map 1:a:0 \
  -codec:a 'ac3' -ac:a 2 \
  -program st=0:st=1:title='myProgramTitle':program_num=0x0001 \
  -metadata:s:a:0 'language=eng' \
  -t '00:00:01.000'  \
  -f 'mpeg' "./test.ts"
```
and

``` bash
$ tsp \
  -I 'file' "./test1.ts" \
  -P 'pmt' --patch-xml "./patch.xml" \
  -P 'tables' --xml - \
  -O 'file' "./output.ts"
```

✅ This works great and the caption_service_descriptor in the PMT is successfully identified as being present by mediainfo, as well as https://media-analyzer.pro/app (a web-based app for analyzing TS files with WASM).  Great!

❌ However, although TSDuck can successfully **_write_** a `<caption_service_descriptor>` to a PMT using named key value pairs, when reading the resultant file, the `<caption_service_descriptor>` node is not supported when **_reading_** from a PMT.  When reading, the `<caption_service_descriptor>` node is not supported as a valid node within a PMT, it is considered only a `<generic_descriptor>`.

``` xml
$ tstables "./output.ts" --xml-output -

<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <PAT version="0" current="true" transport_stream_id="0x0001">
    <metadata PID="0"/>
    <service service_id="0x0001" program_map_PID="0x1000"/>
  </PAT>
  <PMT version="0" current="true" service_id="0x0001" PCR_PID="0x0100">
    <metadata PID="4,096"/>
    <component elementary_PID="0x0100" stream_type="0x02">
      <generic_descriptor tag="0x86">
        E4 65 6E 67 7E 3F FF 73 70 61 7E 3F FF 65 6E 67
        C1 3F FF 73 70 61 C2 3F FF
      </generic_descriptor>
    </component>
    <component elementary_PID="0x0101" stream_type="0x03">
      <ISO_639_language_descriptor>
        <language code="eng" audio_type="0x00"/>
      </ISO_639_language_descriptor>
    </component>
  </PMT>
</tsduck>
```

#### Proposed solution:

I suspect that the `<caption_service_descriptor>` was implemented in TSDuck specifically for the EIT, and it is perhaps a fortuitous coincidence that named elements can be (unintentionally) used to write to the PMT. 

The good news is that patching the PMT with TSDuck works (GREAT!).  A file containing both captions and service descriptors will get identified as having service descriptors in the PMT by mediainfo.  [media-analyzer.pro/app](https://media-analyzer.pro/app) can also read the service descriptors injected into the PMT by TSDuck(GREAT!).  media-analyzer.pro is a relatively recent new WASM/browser tool in the world of TS Analysis, and can be a useful independently-developed companion to TSDuck.  According to ATSC A/65, the `caption_service_descriptor()` is a valid node of a PMT.

**It would be most helpful if TSDuck considered the `caption_service_descriptor()` to be a supported node when reading from  a PMT, rather than a `generic_descriptor()`.**

I believe that there are other descriptors in ATSC A/65-2013 Table 6.25a & 6.25a (not pasted here to respect ATSC's copyright) which are _already_ supported for reading by TSDuck from the EIT but would also be perfectly valid in a PMT, such as `caption_service_descriptor()`, `content_advisory_descriptor()`, `redistribution_service_descriptor()` and `E-AC-3_audio_stream_descriptor()`.  According to ATSC A/65-2013, support for these tags within the PMT is considered mandatory in SCTE Cable Systems, although optional (and still valid) in ATSC Terrestrial Systems.

I hope that this is not an unreasonable ask and may be just an update of the XML Schema to allow these descriptors to be considered valid nodes of the PMT when reading.  It looks like the heavy lifting to support reading these nodes in an EIT has already been done - my ask is that the same nodes are supported when reading from a PMT.

The writing functionality works (GREAT!); this request is more a usability improvement on the reading/analysis side of the toolset.

#### Additional information:
As ever, big thanks to Thierry and the team.  Thierry was kind enough to help me a few years ago when attempting to write service descriptors to an ATSC EIT, and I remain forever grateful for a previous polite and supportive interaction with the TSDuck Team.  This question relates to caption_service_descriptors within a PMT rather than an EIT.

Love your work, Team TSDuck!

lelegard, 2023-11-16:
Hi @bbgdzxng1,

There is no need to enhance anything. This is the normal behavior. Let me explain why.

The short answer is: use option `--atsc`. The detailed answer follows.

This is basically a matter of ambiguities between standards. All MPEG-defined tables and descriptors are recognized by default. Then, DVB, ATSC and ISDB define their own tables and descriptors with their own id, sometimes overlapping ids.

We know that all TS implement the MPEG standard. But is a TS a DVB, ATSC or ISDB one, we cannot know by default.

When a descriptor 0x86 is found, what is this descriptor? Maybe an ATSC caption_service_descriptor, maybe a DVB private descriptor from the previous private data specifier. By default, we cannot know and a generic binary descriptor is reported.

TSDuck does its best to "guess" the context. For instance, if a table id 0xC7 is found in PID 0x1FFB, we know that this must be an ATSC-defined MGT and, from that point, we consider that the TS is an ATSC one. If your PMT is seen _after the MGT_, we know that descriptor 0x86 is an ATSC-defined caption_service_descriptor. Otherwise, we don't.

On the other hand, if the same table id 0xC7 is found in PID 0x0025, then this must be an ISDB-defined LDT and, from that point, we consider that the TS is an ISDB one.

If you know that your stream is an ATSC one and some ATSC-specific descriptor can be seen before the ATSC context is confirmed, use option `--atsc` on the command line. TSDuck will know from the beginning that this an ATSC stream and descriptor 0x86 is immediately interpreted as a caption_service_descriptor.

You mention that it worked from the beginning with an ATSC EIT. This is normal. An ATSC EIT has table id 0xCB with is interpreted as an ATSC-defined EIT without ambiguity because, by chance, no other common standard defines a table with id 0xCB. Therefore, the ATSC context is established as soon as the ATSC EIT section is read. The PMT, on the other hand, is an MPEG-defined table. In the absence of other prior indication, a descriptor 0x86 is ambiguous and unknown.

You also mention that an XML file containing `<caption_service_descriptor>` properly generates the descriptor. This is normal because you use the full name of an ATSC descriptor and we know what to generate.

If you always work with ATSC streams, you can preset options in the user's TSDuck configuration file. See Appendix A in the TSDuck user's guide. Section A.5.2 gives a typical user configuration file for Japanese users working on ISDB streams. You can build the equivalent for American users working on ATSC streams.

Demonstration using 3 XML files with simplified tables:

pat.xml
~~~
<tsduck>
  <PAT transport_stream_id="1">
    <service service_id="1" program_map_PID="100"/>
  </PAT>
</tsduck>
~~~

pmt.xml
~~~
<tsduck>
  <PMT service_id="1">
    <component elementary_PID="110" stream_type="6">
      <caption_service_descriptor>
        <service language="eng" digital_cc="false" caption_service_number="1" easy_reader="false" wide_aspect_ratio="false"/>
      </caption_service_descriptor>
    </component>
  </PMT>
</tsduck>
~~~

mgt.xml
~~~
<tsduck>
  <MGT>
    <table type="TVCT-current" PID="0x1FFB" version_number="11" number_bytes="218"/>
  </MGT>
</tsduck>
~~~

Build one TS file with PAT and PMT and one TS file with PAT, MGT, PMT, in that order:
~~~
tspacketize pat.xml --pid 0 -o pat.ts
tspacketize pmt.xml --pid 100 -o pmt.ts
tspacketize mgt.xml --pid 0x1FFB -o mgt.ts

cat pat.ts pmt.ts >pat-pmt.ts
cat pat.ts mgt.ts pmt.ts >pat-mgt-pmt.ts
~~~

When you dump the file `pat-pmt.ts`, a descriptor 0x86 is seen without contextual information. We do cannot know which kind of descriptor this is (DVB, ATSC, etc.):
~~~
$ tstables pat-pmt.ts
[... 37 lines cut ...]
~~~
$ tstables pat-mgt-pmt.ts 

* PAT, TID 0 (0x00), PID 0 (0x0000)
  Version: 0, sections: 1, total size: 16 bytes
  - Section 0:
    TS id:       1 (0x0001)
    Program:     1 (0x0001)  PID:  100 (0x0064)

* MGT (ATSC), TID 199 (0xC7), PID 8187 (0x1FFB)
  Version: 0, sections: 1, total size: 28 bytes
  - Section 0:
    Protocol version: 0, number of table types: 1
    - Table type: TVCT-current (0x0000)
      PID: 0x1FFB (8187), version: 11, size: 218 bytes

* PMT, TID 2 (0x02), PID 100 (0x0064)
  Version: 0, sections: 1, total size: 30 bytes
  - Section 0:
    Program: 1 (0x0001), PCR PID: none
    Elementary stream: type 0x06 (MPEG-2 PES private data), PID: 110 (0x006E)
    - Descriptor 0: ATSC Caption Service (0x86, 134), 7 bytes
      Number of services: 1
      - Language: "eng", digital: false, line 21: false, easy reader: false, wide: false
~~~

bbgdzxng1, 2023-11-16:
That makes perfect sense, and thankyou for replying with such a clear worked example.  Thanks for the tip about the --atsc option.  I never really understood why that was ever needed, because I had never needed that flag, but it works perfectly with the --atsc flag.  Very cool indeed.  I shall close, gratefully.

closed 2023-11-16 by bbgdzxng1