tsduck/tsduck #1357: [BUG] continuity plugin - many cc reported when included in chain

enricovittorini, 2023-11-17:
#### Bug description:
If I use the continuity plugin with the following command on a ts file:

`tsp -I file skyde.ts -P regulate -P continuity  -O ip 239.0.0.1:5555 -l 172.16.0.100
* continuity: packet index: 1,706,831, PID: 0x06FF, missing 4 packets
* continuity: packet index: 1,706,832, PID: 0x02FF, missing 5 packets
* continuity: packet index: 1,706,833, PID: 0x04FF, missing 1 packets
* continuity: packet index: 1,706,835, PID: 0x00FF, missing 2 packets
* continuity: packet index: 1,706,836, PID: 0x0EFF, missing 3 packets
* continuity: packet index: 1,706,840, PID: 0x05FF, missing 2 packets
* continuity: packet index: 1,706,852, PID: 0x06FF, missing 1 packets
* continuity: packet index: 1,706,853, PID: 0x0EFF, missing 1 packets
* continuity: packet index: 1,706,854, PID: 0x02FF, missing 2 packets
* continuity: packet index: 1,706,858, PID: 0x05FF, missing 1 packets
* continuity: packet index: 1,706,861, PID: 0x00FF, missing 1 packets
* continuity: packet index: 1,706,866, PID: 0x04FF, missing 1 packets
* continuity: packet index: 1,706,889, PID: 0x0303, missing 1 packets
* continuity: packet index: 1,706,920, PID: 0x0F02, missing 1 packets`

I have a total of 14 CC.

I then tried  to join the above multicast using another tsp command:

`tsp -I ip 239.0.0.1:5555 -l 172.16.0.100 -P continuity -O drop`

but i get much mode CC reported:

`tsp -I ip 239.0.0.1:5555 -l 172.16.0.100 -P continuity -O drop
* continuity: packet index: 1,703,899, PID: 0x06FF, missing 8 packets
* continuity: packet index: 1,703,900, PID: 0x0EFF, missing 6 packets
* continuity: packet index: 1,703,901, PID: 0x05FF, missing 3 packets
* continuity: packet index: 1,703,902, PID: 0x02FF, missing 5 packets
* continuity: packet index: 1,703,903, PID: 0x00FF, missing 3 packets
* continuity: packet index: 1,703,904, PID: 0x04FF, missing 2 packets
* continuity: packet index: 1,703,906, PID: 0x0EFF, missing 12 packets
* continuity: packet index: 1,703,907, PID: 0x02FF, missing 10 packets
* continuity: packet index: 1,703,908, PID: 0x04FF, missing 3 packets
* continuity: packet index: 1,703,910, PID: 0x00FF, missing 6 packets
* continuity: packet index: 1,703,911, PID: 0x06FF, missing 1 packets
* continuity: packet index: 1,703,915, PID: 0x05FF, missing 6 packets
* continuity: packet index: 1,703,917, PID: 0x0012, missing 4 packets
* continuity: packet index: 1,703,930, PID: 0x0F02, missing 2 packets
* continuity: packet index: 1,703,937, PID: 0x0102, missing 1 packets
* continuity: packet index: 1,703,943, PID: 0x06FF, missing 2 packets
* continuity: packet index: 1,703,944, PID: 0x0EFF, missing 2 packets
* continuity: packet index: 1,703,946, PID: 0x02FF, missing 1 packets
* continuity: packet index: 1,703,950, PID: 0x00FF, missing 1 packets
* continuity: packet index: 1,703,954, PID: 0x04FF, missing 1 packets
* continuity: packet index: 1,703,956, PID: 0x02FF, missing 1 packets
* continuity: packet index: 1,703,957, PID: 0x0EFF, missing 2 packets
* continuity: packet index: 1,703,958, PID: 0x06FF, missing 3 packets
* continuity: packet index: 1,703,961, PID: 0x00FF, missing 1 packets
* continuity: packet index: 1,703,965, PID: 0x0603, missing 2 packets
* continuity: packet index: 1,703,978, PID: 0x06FF, missing 2 packets
* continuity: packet index: 1,703,979, PID: 0x0EFF, missing 1 packets
* continuity: packet index: 1,703,981, PID: 0x02FF, missing 1 packets
* continuity: packet index: 1,703,982, PID: 0x05FF, missing 1 packets
* continuity: packet index: 1,703,983, PID: 0x0EFF, missing 4 packets
* continuity: packet index: 1,703,985, PID: 0x04FF, missing 2 packets
* continuity: packet index: 1,703,986, PID: 0x06FF, missing 6 packets
* continuity: packet index: 1,703,989, PID: 0x02FF, missing 4 packets
* continuity: packet index: 1,703,990, PID: 0x00FF, missing 3 packets
* continuity: packet index: 1,703,992, PID: 0x05FF, missing 2 packets
* continuity: packet index: 1,704,007, PID: 0x06FF, missing 2 packets
* continuity: packet index: 1,704,008, PID: 0x0EFF, missing 1 packets
* continuity: packet index: 1,704,009, PID: 0x02FF, missing 1 packets
* continuity: packet index: 1,704,018, PID: 0x0504, missing 1 packets
* continuity: packet index: 1,704,019, PID: 0x04FF, missing 1 packets
* continuity: packet index: 1,704,053, PID: 0x0304, missing 1 packets
* continuity: packet index: 1,704,056, PID: 0x06FF, missing 2 packets
[... 139 lines cut ...]
* continuity: packet index: 2,084,243, PID: 0x0703, missing 2 packets
* continuity: packet index: 2,084,640, PID: 0x0504, missing 1 packets
* continuity: packet index: 2,084,862, PID: 0x0012, missing 2 packets
* continuity: packet index: 2,085,070, PID: 0x0001, missing 1 packets
* continuity: packet index: 2,085,071, PID: 0x0000, missing 1 packets
* continuity: packet index: 2,085,072, PID: 0x0060, missing 1 packets
* continuity: packet index: 2,085,073, PID: 0x0064, missing 1 packets
* continuity: packet index: 2,085,074, PID: 0x0062, missing 1 packets
* continuity: packet index: 2,085,075, PID: 0x0066, missing 1 packets
* continuity: packet index: 2,085,076, PID: 0x0065, missing 1 packets
* continuity: packet index: 2,085,077, PID: 0x006E, missing 1 packets
* continuity: packet index: 2,089,799, PID: 0x0011, missing 1 packets
`

#### How to reproduce:
with the steps above

#### Expected behavior:
same CC number on the two commands

#### Environment:
 - OS: Windows
 - OS version: 10
 - TSDuck full version: 3.36-3417
 - Installation type: official binary

lelegard, 2023-11-18:
Hi @enricovittorini 

First, as usual, UDP is not reliable and losing packets is part of the contract. Most of the time, this is a matter of system tuning, buffer sizes, etc.

Since the local address is identical in the sender and receiver commands, I assume that the two run on the same machine. Losing multicast UDP packets on a local interface is definitely a matter of system tuning.

I tried your commands (with another TS file) on macOS, Linux and Windows. On macOS and Linux, I see no packet loss at all. On Windows, on the other hand, I see a lot of packet losses.

So, this seems a Windows system tuning issue. Quickly googling on Windows + multicast + local, I see a lot of options:

- Several listeners on the same group on the same machine.
- UDP buffer tuning (use buffer size options in `ip` plugins or see various registry hacks).
- Presence of VirtualBox network adapters.
- Windows is known to be a piece of crap for multicast.

Anyway, this is probably not a TSDuck issue. More generally, I suggest to investigate a Windows tuning issue.

lelegard, 2023-11-18:
I would also appreciate that users do not systematically declare a "bug" in TSDuck when they see something they did not expect. A problem becomes a bug when a deep analysis _demonstrates_  a deviation of the application behaviour from its specification. If losing UDP packets was a "bug", then all applications on Earth are buggy.

enricovittorini, 2023-11-18:
Tha la @lelegard ... I will investigate. 

With latest build it seems not reproducible. 

Regarding the bug topic: how should I have opened this issue?

lelegard, 2023-11-18:
> With latest build it seems not reproducible.

The latest builds did not change anything about IP, they were just refactoring. The system environment must have changed (system load, presence of other listeners, shutdown of virtual machine, etc).

> Regarding the bug topic: how should I have opened this issue?

As a question. You have something that does not work as you expected and you ask for help.

A bug could be
- A table should contain 1 and it contains 2
- The program crashes

Even a problem such as "a table should be inserted but it is not" is 99% of the time a user error because there are no free space (no stuffing) in the stream. To present something as a bug, you must provide a technical demonstration that the behaviour is incorrect. I know many projects and products where such a bug report would be abruptly closed in the first place with some terse comment such as "bug not demonstrated", full stop, not even trying to understand what is wrong.

enricovittorini, 2023-11-18:
@lelegard As the issue was systematic, I thought it was a bug.

Same stream joined at the same time with Dektec on the same laptop was not reporting cc issue.

I though about env, and spent some time to simplify the command but still was getting cc.

So, the question remains: in such case what do you expect an end user to open? Or should we not open anything?

Maybe next time better to open a "question"? Or "investigation" type  would be more appropriate?

Fake issues have been there since ever and will always be....

enricovittorini, 2023-11-22:
answer is ok.

closed 2023-11-22 by enricovittorini