tsduck/tsduck #1359: [Q] error in tstabcomp processing  EIT  UNICODE charset

rogers-zou, 2023-11-21:
Hi lelegard,
    I want to use the tsduck tool to create a multi-language EPG. The command is as follows. There is no problem with the binary compiled into UTF-8, but UNICODE reports an error. Can you help me find out what the problem is? 
Are there any missing options for my command? 
Thanks.
[eit_schedule_0x50_utf8_modify.zip](https://github.com/tsduck/tsduck/files/13424714/eit_schedule_0x50_utf8_modify.zip)
eit_schedule_0x50_utf8_modify.xml file  attached.
tstabcomp -c eit_schedule_0x50_utf8_modify.xml --default-charset UTF-8  -o eit_schedule_0x50_utf8.bin
tstabcomp -c eit_schedule_0x50_utf8_modify.xml --default-charset UNICODE -o eit_schedule_0x50_unicode.bin

tsduck ver:3.35-3258
[image]

lelegard, 2023-11-21:
Hi @rogers-zou 

Thanks for reporting this. I must admit that the error message is not quite explanatory. However, the error is real. Your file cannot be compiled using UNICODE character set. Let me explain why.

First, the option `--default-charset` in `tstabcomp` in compilation mode (from XML to binary) indicates the charset to use when generating binary strings.

Here, as explained in the user's guide, UNICODE means UTF-16. Each character is then encoded as 2 bytes (or 4 if the character is represented as a surrogate pair).

Second, the size of any descriptor is limited to 255 bytes (simply because the descriptor length is encoded on one byte).

Then, look at your XML file. The error occurs each time the two strings in the short_event_descriptor collectively use more than 127 characters, meaning 254 bytes (approximately because there are other small overhead in the descriptor).

Therefore, these descriptors, with the content you specify, cannot be represented using UTF-16 character set.

UTF-8 is more compact, most of the time (in practice, it depends on the characters). This is why there is no error.

rogers-zou, 2023-11-22:
Hi @lelegard，
   Thanks for your quick reply. I really didn't realize how much of a length increase this was. 
It would be better if the error message could be more friendly.
I checked bellow franch text ,after encode into unicode. The length increased several times. so this is why no mpeg ts using unicode 
tstabcomp: Error in descriptor <short_event_descriptor> at line 21
     <short_event_descriptor language_code="fra">
        <event_name>[7 Digital] C'est le premier événement</event_name>
        <text>[7 Digital] Les dernières nouvelles de toute l’Australie et d’outre-mer. Présenté par Ann Sanders.</text>
      </short_event_descriptor>

   [7 Digital] C'est le premier événement   ----> it is total 76 bytes
5B00370020004400690067006900740061006C005D0020004300270065007300740020006C00650020007000720065006D006900650072002000E9007600E9006E0065006D0065006E007400
[7 Digital] Les dernières nouvelles de toute l’Australie et d’outre-mer. Présenté par Ann Sanders. ---> it is total 196 bytes
[0x5B,0x00,0x37,0x00,0x20,0x00,0x44,0x00,0x69,0x00,0x67,0x00,0x69,0x00,0x74,0x00,0x61,0x00,0x6C,0x00,0x5D,0x00,0x20,0x00,0x43,0x00,0x27,0x00,0x65,0x00,0x73,0x00,0x74,0x00,0x20,0x00,0x6C,0x00,0x65,0x00,0x20,0x00,0x70,0x00,0x72,0x00,0x65,0x00,0x6D,0x00,0x69,0x00,0x65,0x00,0x72,0x00,0x20,0x00,0xE9,0x00,0x76,0x00,0xE9,0x00,0x6E,0x00,0x65,0x00,0x6D,0x00,0x65,0x00,0x6E,0x00,0x74,0x00]
both are not exceed 255 bytes. Is there any problem with my understanding?

rogers-zou, 2023-11-22:
Hi @lelegard 
I just checked  EIT Short event descriptor.  
So  event_name + text char should not exceet 256 bytes. I will  close ticket.
[image]

closed 2023-11-22 by rogers-zou