tsduck/tsduck #1360: [Q] Analyze Plugin  pes-stream-id

enricovittorini, 2023-11-21:
Am I wrong of before the pes-stream-id was listed in analyze.pids json output of the analyze plugin?

lelegard, 2023-11-22:
The plugin analyze outputs what the stream contains. Looking at some streams in the [repository](https://tsduck.io/streams/), Teletext streams have private 1 as stream id, indeed. I do not know if other values are valid or not.

enricovittorini, 2023-11-22:
I think the issue was a txt PID declared but with 0 bps.

lelegard, 2023-11-22:
> I think the issue was a txt PID declared but with 0 bps.

The stream id is extracted from the PES header, it cannot be found in the signalisation. So, indeed, 0 pbs = 0 packet = unknown stream id.

enricovittorini, 2023-11-22:
Yes of course. Didn't notices there were 0 packets is that PID. Sorry...

closed 2024-04-21 by github-actions[bot]