tsduck/tsduck #1361: [Q] Please give the main command for eitinject

KradinKYu, 2023-11-22:
I'm trying to figure out EITInject, but I'm having problems. Please, help.

There is an xml file with EPG. Each channel has its own ID:

        <channel id="000001336">
                <display-name lang="ru">BTN</display-name>
        </channel>
        <channel id="000000360">
                <display-name lang="ru">Vers</display-name>
        </channel>
        <channel id="000000350">
                <display-name lang="ru">Terra</display-name>
        </channel>
        <channel id="000000361">
                <display-name lang="ru">Openair</display-name>
        </channel>

You need to match the channel ID from EPG.xml with the corresponding channel ID on the output.
Here's an example analysis from another broadcaster's EPG:

TS      Services  With EITp/f  With EITs  EPG days
------  --------  -----------  ---------  --------
Actual         0            0          0         0
Other         14           14         14         3

A/O  TS Id   Srv Id  Name  EITp/f  EITs  EPG days
---  ------  ------  ----  ------  ----  --------
Oth  0x0002  0x0001        yes     yes          3
Oth  0x0002  0x0002        yes     yes          3
Oth  0x0002  0x0003        yes     yes          3
Oth  0x0002  0x0004        yes     yes          3
Oth  0x0002  0x0005        yes     yes          3
Oth  0x0002  0x0006        yes     yes          3
Oth  0x0002  0x0007        yes     yes          3
Oth  0x0002  0x0008        yes     yes          3
Oth  0x0002  0x0009        yes     yes          3
Oth  0x0002  0x000A        yes     yes          3
Oth  0x0002  0x000B        yes     yes          3
Oth  0x0002  0x000C        yes     yes          3
Oth  0x0002  0x000D        yes     yes          3
Oth  0x0002  0x000E        yes     yes          3

I understand how to make a table using the sdt plugin, but how to make correspondence between epg.xml and output pids.
How to bring everything to the table above?

Unfortunately, I didn’t find any examples of using the eitinject plugin in the documentation.

Maybe there is some kind of universal template for these purposes?

lelegard, 2023-11-22:
> Maybe there is some kind of universal template for these purposes?

EPG injection is complicated and there is no "universal template" for that. It depends on the transport stream source, its destination, your EPG event provisioning source and strategy, etc.

So, you need to understand first how it works, not apply some "universal template".

If you are not yet familiar with TSDuck and the `tsp` command (transport stream processor), I recommend to have a look at [this presentation](https://tsduck.io/download/docs/tsduck-presentation.pdf), a bit old but still relevant, and browse through the [TSDuck user's guide](https://tsduck.io/download/docs/tsduck.pdf), at least the table of contents, the reference section on the `tsp` command and the `eitinject` plugin.

KradinKYu, 2023-11-24:
tsp -P sdt --patch-xml pack.xml -P inject -p 000001336 --eit-pf -b 5000000 TV_pack.xml  -P eitinject -f TV_Pack.xml --bitrate 5000000 --actual --other --ts-id 1 --time system -O ip 224.205.1.1:1234