tsduck/tsduck #1362: [Q] How to get the hidden field from ts::VCT exposed to demux

MorelloCherry, 2023-11-23:
Hi,

my use case is to parse a live or recorded MPEG transport stream and run that via ts::TSAnalyze and ts::SignalizationDemux.
This is used to make report available services with quirks for ATSC and DVB.

One issue I came across is the VCT support a flag called "hidden" which in essence hides the service but that is no exposed/used outside of ts::VCT and not visible in ts::TSAnalyze nor ts::SignalizationDemux.

I do not want to copy the existing ts::SignalizationDemux code and make the changes exposing the VCT that is otherwise just used in ts::SignalizationDemux::handleTable. This could be done in the same way as ts::SignalizationDemux::_last_nit is handled. Of course this only valid for ATSC and perhaps there is a much better way and in addition does not require me to track upstream changes / bug fixes.

Any suggestion would be helpful.

- Thank you

lelegard, 2023-11-23:
Hi @MorelloCherry 

I am not 100% sure to understand your need.

I assume that you develop a C++ application and you want to handle the hidden flags in the VCT. So, if you simply get the VCT, either using ts::SignalizationDemux or ts::SectionDemux, you can deserialize it as an instance of CVCT or TVCT. These are C++ classes which list all channels and all fields, including the one named hidden.

And since the VCT are all in PID 0x1FFB, it is probably simpler to use a ts::SectionDemux using that PID.

Could you describe what is your final purpose?

MorelloCherry, 2023-11-24:
Thank you for your suggestion, I will give it a try.

The purpose is to get a generic channel list of services that are not specific to a TV standard per se.
What I need to have is:
- Channel number (LCN or Major-Minor ...)
- Channel name
- Program number
- Service Id
- PMT pid
- PCR pid
- Hidden flag (ATSC - from VCT, not sure about others)
- CA flag
- per elementary stream
  - Type
  - PID
  - CA flag

lelegard, 2023-11-25:
I added the hidden flag (and the LCN) in `tsanalyze` and plugin `analyze`. It is also available in normalised and JSON format.

With DVB, there is no "hidden" flag. However, most operators include a "visible" flag in their private LCN descriptor. Therefore, in a DVB service, the "hidden" flag is set to true when a "visible" flag is present and set to false in an private LCN descriptor.

Does this help? 

If you want to analyze this into your own application, a possible way is to subclass `ts::TSAnalyzer`. This class only collects data and synthesizes them. The results are in protected data structures. The use of the collected data is done in a subclass. The command `tsanalyze` and the plugin `analyze` use the subclass `ts::TSAnalyzerReport` which creates the text grids or JSON reports. You may develop your own subclass of `ts::TSAnalyzer` to use or present the collected data the way you like.

MorelloCherry, 2023-11-27:
That is quite helpful, thank you.

MorelloCherry, 2023-11-28:
Did not mean to reopen the issue.

closed 2023-11-28 by MorelloCherry

lelegard, 2023-11-28:
> The addition to the c++17 requirement was a bit of a surprise and sadly broke my build. My current target compiler is GCC 8.3 so I had to hand-crank in the changes relating to the hidden flag...

See this: https://tsduck.io/doxy/building.html#cpprequnix

Your distro, even if it is a bit old, may have GCC 11 packages (as explained for RedHat 8 in the link above). Most so-called "stable" (i.e. 10 years old) distros have realized that they cannot lock users with prehistoric compilers forever and they provide alternative packages for more recent compilers.

Honestly, C++23 is already implemented, some features of C++26 are on the way. It does not seem too much to require C++17.