tsduck/tsduck #1363: [BUG] error: call of overloaded ‘__is_path_src(ts::UString, int)’ is ambiguous

ikor7, 2023-11-24:
When compiling on RPI OS 64bits (Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux), it gives me the following error:

`/usr/include/c++/10/bits/fs_path.h:119:29: error: call of overloaded ‘__is_path_src(ts::UString, int)’ is ambiguous
  119 |     : decltype(__is_path_src(std::declval<_Source>(), 0))
      |                ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/10/bits/fs_path.h:99:5: note: candidate: ‘std::filesystem::__cxx11::__detail::__is_encoded_char<_CharT> std::filesystem::__cxx11::__detail::__is_path_src(const std::__cxx11::basic_string<_CharT, _Traits, _Allocator>&, int) [with _CharT = char16_t; _Traits = std::char_traits<char16_t>; _Alloc = std::allocator<char16_t>; std::filesystem::__cxx11::__detail::__is_encoded_char<_CharT> = std::filesystem::__cxx11::__detail::__is_encoded_char<char16_t>]’
   99 |     __is_path_src(const basic_string<_CharT, _Traits, _Alloc>&, int);
      |     ^~~~~~~~~~~~~
/usr/include/c++/10/bits/fs_path.h:107:5: note: candidate: ‘std::false_type std::filesystem::__cxx11::__detail::__is_path_src(const _Unknown&, ...) [with _Unknown = ts::UString; std::false_type = std::integral_constant<bool, false>]’
  107 |     __is_path_src(const _Unknown&, ...);
      |     ^~~~~~~~~~~~~
make[2]: *** [../../Makefile.inc:584: /home/pi/tsduck/bin/release-aarch64-raspberrypi/objs-libtsduck/tsAACDescriptor.o] Error 1
make[1]: *** [Makefile:20: install-tools] Error 2
make: *** [Makefile:134: install] Error 2
`

lelegard, 2023-11-24:
Hi @ikor7,

Which version of gcc is this ?

The code base of TSDuck is currently migrated to C++17. In theory, this means gcc 8 or more. The current state of the repo is successfully built on recent versions of Linux, macOS and Windows, with their respective recent compilers. At some point, we will drop support for old compilers but I currently do not know the exact frontier between non-supported compilers, compilers which pretend to support C++17 but have bugs, compilers which actually support C++17. All recent versions, including in the GitHub CI/CD pipeline, are in the latter category.

ikor7, 2023-11-24:
pi@raspberrypi:~ $ gcc --version
gcc (Debian 10.2.1-6) 10.2.1 20210110
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

lelegard, 2023-11-24:
Ubuntu 22.04 on the GitHub CI/CD has gcc 11 and it works. All distros I have include gcc 11, 12 or 13 and it works also.

I found an Alma Linux 8 with gcc 8 and I got the same error. Gcc is completely out of its mind. The full message is enlightening. It reports an error on a multimap on strings in class ts::Enumeration, a class that didn't change for years, about calling an std::filesystem function which is never used in TSDuck. In fact, TSDuck now uses std::filesystem, this is new, but other functions, not the one in the error message. This unrelated change seems to confuse gcc.

So, this is a gcc 8-10 bug which was fixed on gcc 11 onwards.

lelegard, 2023-11-25:
On Alma Linux 8.8 (same as RedHat 8.8), with gcc 8.5 as default, it is possible to install other versions of gcc, namely gcc 9, 10, 11, 12, 13. I installed gcc 13. There is a script to set enable it in the current process. Then, TSDuck compiles.

Check the packages on RPI OS for a similar kind of setup.

My Raspberry Pi 4 uses Ubuntu 23 and TSDuck compiles.

ikor7, 2023-11-25:
OK, thanks.

closed 2023-11-25 by ikor7