tsduck/tsduck #1364: T2MI PLP replacer[FEAT]

izharm, 2023-11-25:
Hi,

I'm looking for a way to replace one PLP in a multiPLP T2MI stream with another PLP.  Obviously the PLPs have the same L1 parameters, so replacing the PLP should keep the T2MI stream intact.

The replacement PLP could come for a second TS or a different PID in the same TS.

I guess it is not an addition to the current T2MI plugin, as the output of the plugin, after the replacement, need to encapsulate the data stream back to T2MI format, which I don't think tsduck currently can do.

Thanks 

Izharm

lelegard, 2023-11-25:
Hi @izharm 

Since you want to replace one PLP in the T2MI PID, you need to decapsulate all PLP, keep all of them but one, remux them with the new TS and rebuild a new multi-PLP T2MI PID.

The plugin `t2mi` can only decapsulate a TS from a T2MI PID. With TSDuck, there is currently no way to build a T2MI PID from one or more TS to encapsulate. There is no plan to so either, sorry.

izharm, 2023-11-25:
lelegard,

how about an enhencment to the current plugin to output the full T2 packet and not just the description (when using the log 
option)?

lelegard, 2023-11-25:
> how about an enhencment to the current plugin to output the full T2 packet and not just the description

Do you mean an hexadecimal dump ? A binary dump in a file ?

That can be possible but it won't help you rebuilding a T2MI PID with one PLP replaced by another one from outside.

izharm, 2023-11-28:
I can get two binary files (of the two recapitulated transport streams) and
do the replacement and encapsulation outside taduck….

izharm, 2023-11-30:
Amazing, thanks !

closed 2024-05-05 by github-actions[bot]