tsduck/tsduck #1367: [Q] Regarding the tsp -P pmt plugin

rnarasappa, 2023-11-26:
Hi, 

We are trying to combine the functionalities of splicemonitor and the pmt plugins in our custom plugin. That is, we want to pass all the PMT PIDs (that contain the Splice Info sections) as input to the pmt plugin from within our custom plugin (so as to update all these PMTs with certain additional descriptors).  

From the command line, we would normally do that similar to the below.
**tsp --debug=3 -I file input_stream.ts \
                        -P custom_plugin --bitrate 677280 \
                        -P pmt --pmt-pid 272 --add-pid 8002/0x0C --set-stream-identifier 8002/9  \
                        -P pmt --pmt-pid 268 --add-pid 8001/0x0C --set-stream-identifier 8001/9 \
                        -O file output_stream.ts**

Here 2 sets of inputs (PMT PIDs 272 & 268, new PIDs 8002 &8001) are being specified explicitly to the PMT plugin. 
Is there a way to pass these input parameter values from within our custom_plugin itself (rather than specifying explicit PID values at the command line).?  
That is, in our custom_plugin we are identifying the PMTs that contain Splice Info sections and want to pass these PMT PID values (along with new PIDs & stream identifier values) to the PMT plugin internally (not from command line) so that we could use this method for a Live TS on the fly.  Please suggest.

regards,
Raghunath

lelegard, 2023-11-26:
Hi @rnarasappa 

There is no communication mechanism between plugins. Each plugin is independent.

Why not modifying the various PMT's yourself, in your custom plugin, without any additional `pmt` plugin? You have already done most of the work: coding a plugin in C++, getting all PMT's, deseriliazing them as C++ object. You just have to add the required descriptors in the PMT and packetise them. The only additional step is replacing packets from the PMT PID's using packets from the packetizers of the modified PMT's.

rnarasappa, 2023-11-27:
Hi Thierry, 

Thanks for your response. Our custom plugin is similar to the splicemonitor plugin, i.e., the handlePMT() is receiving the PMTs but this is in "const PMT" form. Are you suggesting to copy this PMT, deserialize, modify, packetize and then reinsert.?
Is there any closest reference from any other plugin.?

Earlier, we had tried to inherit the AbstractTablePlugin in our custom plugin but that appeared to have some conflicts as there were many build errors.

regards,
Raghunath

lelegard, 2023-11-27:
> Are you suggesting to copy this PMT, deserialize, modify, packetize and then reinsert.?
Is there any closest reference from any other plugin.?

That's it. Many plugins share this pattern.

In practice:

- Use a `SectionDemux` to get notified of tables. Initially set `PID_PAT`.
- In your `handleTable()` method, keep PAT and PMT, deserialize them as C++ objects.
  - PAT is only used to identify PMT PID's. Add all PMT PID's to the demux.
  - PMT is used for your own transformations. Modify the `PMT` C++ object, serialize it in a `CyclingPacketizer`. Create one context object per PMT PID. This context object contains the `CyclingPacketizer`, as well as any info you would need. 
- In your `processPacket()` method:
  - Feed the demux with all packets.
  - If the current PID is one of the modified PMT PID's, replace the packet with the next one from the correponding packetizer.

I suggest to have a look at plugin `tsrename`. It is very simple and does most of this (except creating one context object per modified PID).

The plugin `remap` creates a packetizer per PMT PID.

> Earlier, we had tried to inherit the AbstractTablePlugin in our custom plugin 

This one is dedicated to specialized plugins such as `pmt`, `pat`, `sdt`, etc. which perform transformations on one specific table on one specific PID. In practice, all these modules were initially independently developed until I realized that they shared the same high-level logic. Then, I refactored them to push the common high-level logic in the same superclass.

rnarasappa, 2023-11-27:
Thanks Thierry. That is really helpful. I will try this approach.

regards,
Raghunath

rnarasappa, 2023-12-12:
Hi Thierry,

Based on your suggestions above, we are able to successfully add the required descriptors into the PMT from our custom plugin.
However, we noticed the older PMT section is also still present in the output stream along with the new PMT.  Is there something else specifically to be taken care to avoid this duplication.?

And, how do we update the version of this modified PMT from our custom plugin.? 

regards,
Raghunath

lelegard, 2023-12-12:
> And, how do we update the version of this modified PMT from our custom plugin.?

You must _replace_ the packets in the PMT PID with the packets from the packetizer of your modified PMT. See [this example](https://github.com/tsduck/tsduck/blob/6deed7f372151ad65559a3658437a735f9a8d207/src/tsplugins/tsplugin_svrename.cpp#L559) in the plugin `svrename` where several tables are replaced.

rnarasappa, 2023-12-13:
Hi Thierry, 

Yes, we have added that support for using the packetizer (like the CyclingPacketizerPtr in tsplugin_remap.cpp). The packetizer method in svrename plugin appears to be same.  
But still we see the original copy of PMTs being present in the output stream along with the modified PMTs.

In our custom plugin (which is similar to splicemonitor plugin), both the SectionDemux and SignalizationDemux (for getting PMTs) are being used.  

In start(), we have the below
....
    _sig_demux.reset();
    _sig_demux.addFilteredTableId(TID_PMT);
    _section_demux.reset();
    _section_demux.setPIDFilter(NoPID);
....

In handlePMT(), we are adding specific PMTs into the SectionDemux using _section_demux.addPID() call.

In handleTable(), we are modifying those specific PMTs and then adding into packetizer as below:
            const CyclingPacketizerPtr pzer = getPacketizer(pmtPID, true);
            pzer->removeSections(table.tableId(), pmt.service_id);
            pzer->addTable(duck, pmt);

In the processPacket(), we have the TS packets being fed to both these demuxes as below.  Is this correct.?
.......
.......
      _section_demux.feedPacket(pkt);
      _sig_demux.feedPacket(pkt);
       const CyclingPacketizerPtr pzer = getPacketizer(pid, false);
        if (!pzer.isNull()) 
        {
            pzer->getNextPacket(pkt);
            return TSP_OK;
        }
.......
......
Should we feed the modified PMTs only to one of these demuxes.?

regards,
Raghunath

lelegard, 2023-12-13:
I think you missed the point that pulling a packet from the packetizer for a table on PID XXX shall be done when and only when the input packet is from the same PID XXX. Otherwise, you will probably overwrite other PID's (resulting in corrupted stream) and the previous table is not overwritten.

rnarasappa, 2023-12-13:
Hi Thierry, 

I believe you meant the below (I didn't paste the code fully in my previous comment). Below is the actual way we pull from the packetizer in processPacket().

 const PID pid = pkt.getPID();
 _section_demux.feedPacket(pkt);
 _sig_demux.feedPacket(pkt);
   
if (pid is one of our specific PMTs)
{
    const CyclingPacketizerPtr pzer = getPacketizer(pid, false);
    if (!pzer.isNull())
    {
         pzer->getNextPacket(pkt);
         return TSP_OK;
    }
}

We are checking if the TS packet pid is present in our PMT PID list and then we invoke getPacketizer(). 
Even with this, the issue is still there.

regards,
Raghunath

lelegard, 2023-12-13:
OK, so you must have some logical bug in the selection of "one of our specific PMTs" or in `getPacketizer()` being unexpectedly null, or this piece of code not being executed in some cases.

There is no magic. The `processPacket()` method of the plugin is called for _all packets_ in the TS at that point of the chain (except if you use `--only-label`). The call to `getNextPacket(pkt)` _always_ replaces the packet, at a last resort with a null packet if the packetizer has no packet to return. So, if packets from selected PIDs survived, there is no other option that this piece of code is _not_ executed, or not always executed. You need to debug and/or trace execution.

closed 2024-05-12 by github-actions[bot]