tsduck/tsduck #1371: [Q] Using ts::SignalizationDemux and test when processing complete

MorelloCherry, 2023-11-28:
How does one know when ts::SignalizationDemux has acquired all data for a successful service scan?
I like to stop the processing as soon as the service information is complete.

The code below work except it fails to record any of the elementary PIDs.
How do I ensure that the processing of elementary PIDs s included?

Any suggestions, thanks a lot in advance.

```ts::DuckContext tsDuck;
ts::SignalizationDemux tsDemux(tsDuck);

while (true) {
    tsDemux.feedPacket(pkt) // feed with MPEG_TS from file

    // check if enough data has been received...
    ts::ServiceList list;
    ts::SignalizationDemux::getServices(list);

    if (!list.empty()) {
        bool complete = true;
        for (const auto &it : list) {
            if (!it.hasId()) {
                complete = false;
                break;
            }

            const auto sid = it.getId();
            addFilteredServiceId(sid);
            if (!isFilteredServiceId(sid)) {
                complete = false;
                break;
            }
        }
        if (complete) {
            break;
        }
    }
}
```

lelegard, 2023-11-29:
Hi @MorelloCherry 

First, I assume that this is not the original code because it won't compile (one instance method is used as a static one and two others are used without object).

Using the constructor `ts::SignalizationDemux tsDemux(tsDuck)`, you filter everything, all basic signalization tables and all services. There is no need to `addFilteredServiceId()`. Explicitly filtering a service is fine when you need just one service that you already know by id but you don't know its PMT PID. So, you say "I just want that service, if you eventually find it".

Generally speaking, in a broadcast stream, scanning is never "complete". It is always incremental. New information can always arrive to update the description of a service. For instance, assume that you got the PAT and all corresponding PMT's. Is the scan "complete"? Yes and no. Yes because you know all services. But no because additional information on these services may arrive later, for instance the name in an SDT or the LCN in a NIT. So, when do you decide that the scan is complete? When the name and LCN is known for all services? However, the SDT and NIT may not be present. In that case, there is no need to wait forever. Worse, the PAT may contains service id and PMT PID for non-present services, with nothing on the corresponding PMT PID.

Therefore, the concept of "complete scan" does not exist. You must decide when to stop. When the file is fully read, in case of a file. After a number of seconds, in a broadcast stream. You decide but no one will tell you "it's complete".

You use the `SignalizationDemux` in polling mode. You poll its list of services after each packet. This is very inefficient.

The class `SignalizationDemux` was designed to be used in notification mode. Your application class shall implement the interface `SignalizationHandlerInterface` and override the virtual methods it needs, for instance `handleService()` if you want to be notified of all modifications in the description of a service. The same handler will be called each time the description of the service is updated. Typically, it will be called the first time the PAT is received, in which case the only available information is service id and PMT PID. Then, it will be called again when the PMT is received, with much more information. Later (or even before the PMT), it can be called again when the service name or LCN become known. Etc.

MorelloCherry, 2023-11-29:
Thank you for getting back to me so quickly.

As you have pointed out that the processing of the service information (SI) never finishes but my aim was to know at the earliest possible time when a service scan is complete and I could move on.
The purpose my code is to extract service information from a MPEG-TS transport stream and the platform I am using provides multiple tuners and I am intending to scan all of them in parallel. Speed of this exercise is of essence. Thankfully, TSDUCK is very CPU and memory efficient which I really appreciate and polling was not my ideal solution but a mere test case. Using a timeout is by and large is a fallback but shall not be the primary solution when acquiring the service information - well, I am digressing here...

Now I have started to change the polling code into the more favourable callback methodology and I am quite happy that I get most of the details I used to have from the ts::Analyser APIs. Unfortunately some details are missing on the ts::Service interface; please note this is not about the bitrate I am talking about.

The details I was looking for are the
- elementary stream whether it is encrypted or not (perhaps the NDS ECM PID, that is missing at all)
- the hidden (ATSC)/visibility (DVB) flag as available on the ts::TSAnalyzer::ServiceContextPtr vs. ts::Service

On the positive note I can use the handleService callback to get an understanding to check if all services are acquired and then with the combination of ts::TSAnalyzer get the details I do need. Sadly though I cannot replace the ts::TSAnalyzer code with the callback solution just yet.

Thank you.

lelegard, 2023-11-29:
> my aim was to know at the earliest possible time when a service scan is complete and I could move on.

I don't know how to explain that but there is no _guaranteed_ way to get a complete scan.

You want the elementary streams of each service. You will get them in the PMT. The `handleService()` virtual method receives a `Service` and a `PMT` object. Probably the first time it is called for a service, the PMT is invalid (use `isValid()` to check) because the PMT is not received yet and you got minimal information on the service from the PAT. At some point, when the PMT is received, you get a valid PMT which contains all elementary streams.

However, how do you know that you have all services? In the `handleService()` method, you can call `demux.lastPAT()` to get the last PAT and verify that you have received enough description on all services. However, it is possible (rare but possible) that you never get the PMT of one service from the PAT, simply because the service is down is nothing is sent. In that case, you will wait forever. So, you have to decide to give up at some point. This is only your decision.

Similarly, you need the hidden flag. With DVB, this is in the NIT. Currently, I don't think that `SignalizationDemux` extracts it. This can be improved. However, there is no guarantee that the NIT is received or that the services are listed in a private LCN descriptor. Again, in some streams, you will wait for the hidden flags forever and never get them.

Encrypted streams cannot be detected from the signalisation. The presence of a CA_descriptor in the PMT is not sufficient. If it is present at program level, it gives no information on which ES is encrypted. If it is present at component level, it means that it is highly probable that the component is encrypted, not a guarantee. So, you also have to check the transport scrambling control bits in the header of the TS packets of that ES. Again, some component (audio description for instance) may be muted from time to time, either with no packet or clear empty packets, giving no information on the ES encryption.

So, don't count on TSDuck or anybody else to say "it's complete". You need to listen for as many information as possible, knowing that information on services comes in pieces, through several calls to `handleService()`, depending on the incoming information (PAT, PMT, SDT, NIT). So, at some point, you have to decide "it's over, don't wait any longer, be happy with what we got so far".

lelegard, 2023-12-01:
SignalizationDemux now tracks visibility/hidden on services and encrypted PID's.
See an example here: https://github.com/tsduck/tsduck/tree/master/sample/sample-sig-demux

However, the information on services remains incremental, with more information each time a table adds some information (PAT, PMT, SDT, NIT) and still without a way to define what "complete scan" means. You still have to decide when to stop, either when you got all information that _you_ consider important, or when some large enough number of packets have been processed and the missing information will likely never arrive.

MorelloCherry, 2023-12-04:
That is a fair assessment, I can see multiple calls to handleService with more and more details being acquired. You are correct to mention it very difficult to know when one is done. The best approach I think is to acquire the most important details and then what a little longer...

A little excursion from the initial question, perhaps you don't mind.

If I use the the example you have kindly made available to me, how do I show services that are signalled as not running?
Thanks a lot.

MorelloCherry, 2023-12-04:
The SDT contains ServiceEntry that gives the running_status of the service. The RST does not appear to be present in my test data. Now I feel I am getting somewhere...

Thanks a lot.

closed 2024-05-12 by github-actions[bot]