tsduck/tsduck #1376: [Q] FFmpeg tripping up on TSDuck HLS output

tab1293, 2023-12-07:
We are attempting to use TSDuck to ingest a SRT stream and pass it through to FFmpeg via HLS. This usually works great but often we see issues where FFmpeg gets "tripped up" on what seems to be corrupt packets in the MPEG TS stream. When this happens, FFmpeg transcoding speed starts to decline and does not recover unless the FFmpeg process is restarted.  

Here is the tsp command we're using to convert SRT to HLS:
```
tsp -I srt --caller [srt_uri] --transtype live --latency 2000 --statistics-interval 2000  \
    -P splicemonitor -a --json -o - \
    -P analyze \
    -O hls -d 3 --live 5 -p playlist.m3u8 --live-extra-segments 10000 out.ts
```
And here is the FFmpeg command:
```
ffmpeg -i http://localhost:8123/playlist.m3u8 \
    -ac 2 -b:a 256K -c:a aac \
    -c:v libx264 -b:v 5M -max_delay 0 -pix_fmt yuv420p -profile:v baseline -r 30 -x264-params keyint=120 -bf 0 \
    -f flv -flvflags no_duration_filesize [rtmp_url] 
```
When the issue occurs we see logs that look like the following in FFmpeg:
```
[mpegts @ 0x7f889df822c0] PES packet size mismatch
[mpegts @ 0x7f889df822c0] Packet corrupt (stream = 3, dts = 5781579583).
[hls @ 0x7f889df82040] Packet corrupt (stream = 3, dts = 5781576703).
http://localhost:8123/playlist.m3u8: corrupt input packet in stream 3
[mpegts @ 0x7f889df822c0] PES packet size mismatch
[mpegts @ 0x7f889df822c0] Packet corrupt (stream = 1, dts = 5781579703).
[hls @ 0x7f889df82040] Packet corrupt (stream = 1, dts = 5781576823).
http://localhost:8123/playlist.m3u8: corrupt input packet in stream 1
[ac3 @ 0x7f889df42040] delta bit allocation strategy reserved [ac3 @ 0x7f889df42040] error decoding the audio block Error while decoding stream #0:1: No error information
```
Can anyone explain what might be going on here? Is this something that maybe the continuity plugin may help with?

The interesting thing is that when playing back the SRT stream directly with VLC, we never see any decode issues. Also if we generate a TS file with tsduck and perform the same transcode with FFmpeg on the non live TS file, we don't see the same errors.

I'm happy to provide additional details to help clarify any details around our set up. Any help is appreciated, thanks!

lelegard, 2023-12-07:
None of the tsp plugins you use modifies packets. There is no unreliable I/O plugin such as `ip`. So, I see no obvious reason. 

Investigation tracks:
- Add a `continuity` plugin to look for missing packets. 
- Since you keep up to 1000 output segments on disk, just `cat` them into one single TS file and `tsanalyze` it.

lelegard, 2023-12-08:
Additional suggested test:

- Pipe the `tsp -I srt` live command directly into `ffmpeg`.

tab1293, 2023-12-08:
@lelegard so I still am seeing the corruption errors, and the continuity plugin does not seem to be reporting any issues when these corruption periods occur.

I am testing out passing tsp stdout to ffmpeg stdin now and will report back if I continue seeing the corruption logs.

Maybe a clue here.. whenever the corruption logs are thrown, ffmpeg reports a large increase in the number of frames it is processing: 

[image]

You can see the 3 instances that see the corruption logs starting around 9:30 - 10 drop in speed and FPS while seeing a large increase in the frames being processed. 

(These graphs are plotted using the ffmpeg progress output `frame=  269 fps= 66 q=6.0 size=    1536kB time=00:00:05.44 bitrate=2313.0kbits/s speed=1.33x`)

tab1293, 2023-12-08:
Okay seeing a different issue now using tsp to directly pass data to ffmpeg stdin:

```
{"program":"tsduck","time":"2023-12-08T17:34:41Z","message":"17:34:41.225942/SRT:RcvQ:w1!W:SRT.qr: @735152877:No room to store incoming packet seqno 1906824841, insert offset 32831. Space avail 25497/25600 pkts. Packets ACKed: 102 (TSBPD ready in -11575 : -11540 ms), not ACKed: 16592, timespan 5859 ms. GETTIME_MONOTONIC drift -1 ms."}
{"program":"tsduck","time":"2023-12-08T17:34:41Z","message":"17:34:41.229150/SRT:RcvQ:w1!W:SRT.qr: @735152877:No room to store incoming packet seqno 1906824849, insert offset 32839. Space avail 25497/25600 pkts. Packets ACKed: 102 (TSBPD ready in -11579 : -11543 ms), not ACKed: 16592, timespan 5859 ms. GETTIME_MONOTONIC drift -1 ms.\n17:34:41.229184/SRT:RcvQ:w1!W:SRT.qr: @735152877:No room to store incoming packet seqno 1906824850, insert offset 32840. Space avail 25497/25600 pkts. Packets ACKed: 102 (TSBPD ready in -11579 : -11543 ms), not ACKed: 16592, timespan 5859 ms. GETTIME_MONOTONIC drift -1 ms."}
{"program":"tsduck","time":"2023-12-08T17:34:41Z","message":"17:34:41.232325/SRT:RcvQ:w1!W:SRT.qr: @735152877:No room to store incoming packet seqno 1906824858, insert offset 32848. Space avail 25497/25600 pkts. Packets ACKed: 102 (TSBPD ready in -11582 : -11547 ms), not ACKed: 16592, timespan 5859 ms. GETTIME_MONOTONIC drift -1 ms."}
{"program":"tsduck","time":"2023-12-08T17:34:41Z","message":"17:34:41.235536/SRT:RcvQ:w1!W:SRT.qr: @735152877:No room to store incoming packet seqno 1906824868, insert offset 32858. Space avail 25533/25600 pkts. Packets ACKed: 66 (TSBPD ready in -11572 : -11550 ms), not ACKed: 16592, timespan 5846 ms. GETTIME_MONOTONIC drift -1 ms.\n17:34:41.235563/SRT:RcvQ:w1!W:SRT.qr: @735152877:No room to store incoming packet seqno 1906824869, insert offset 32859. Space avail 25533/25600 pkts. Packets ACKed: 66 (TSBPD ready in -11572 : -11550 ms), not ACKed: 16592, timespan 5846 ms. GETTIME_MONOTONIC drift -1 ms."}
{"program":"tsduck","time":"2023-12-08T17:34:41Z","message":"17:34:41.236760/SRT:TsbPd!W:SRT.br: @735152877:RCV-DROPPED 4 packet(s), packet seqno %1906792014 delayed for 11548.350 ms"}
{"program":"tsduck","time":"2023-12-08T17:34:41Z","message":"17:34:41.238752/SRT:RcvQ:w1*E:SRT.qr: @735152877:SEQUENCE DISCREPANCY. BREAKING CONNECTION. seq=1906824876 buffer=(1906792014:1906808601+1906817613), 7264 past max. Reception no longer possible. REQUESTING TO CLOSE."}
{"program":"ffmpeg","time":"2023-12-08T17:34:47Z","message":"frame= 3474 fps= 46 q=9.0 size=   19217kB time=00:00:58.56 bitrate=2688.3kbits/s speed=0.77x    \r"}
{"program":"ffmpeg","time":"2023-12-08T17:34:47Z","message":"[mpegts @ 0x58055577e180] changing packet size to 204"}
{"program":"ffmpeg","time":"2023-12-08T17:34:47Z","message":"[mpegts @ 0x58055577e180] changing packet size to 188"}
```
A ton of `No room to store incoming packet seqno` errors from tsduck and then tsduck exits with `SEQUENCE DISCREPANCY. BREAKING CONNECTION` 

Does this provide anymore insight in to what the issue could be?

lelegard, 2023-12-09:
> A ton of "No room to store incoming packet seqno" errors from tsduck 

That is interesting and probably the key to the solution.

This is a message from the SRT layer in the input plugin. 

When tsp is directly piped into ffmpeg, there is a complete flow control all along the chain. When one of the elements in the chain is too slow, it slows the output downstream and sporadicly blocks the input upstream.

The SRT message shows that incoming packets arrive faster than the rest of the chain can consume them.

All tsp plugins you use consume close to zero CPU. Therefore, the probable cause of your issue is that ffmpeg cannot transcode at the speed of input and blocks the input from time to time.

The root causes may include too small CPU, lack of memory, too aggressive encoding parameters (needing more CPU than available), etc. So, this is a global system capacity and tuning issue, not a TSDuck issue.

In your original problem description with an intermediate HLS layer, there was no blocking in the tsp chain but the HLS playlist was renewed faster than ffmpeg could read it. You specified 1000 segment files on disks but only 5 are referenced in the playlist. Ffmpeg continuously reads the playlist, looking for more segments. But since ffmpeg seems too slow, it missed intermediate segments. For instance, at some point, the HLS playlist references segments 1001-1005, ffmpeg gets and encodes the 5 segments, then reads the playlist again, but because it was too slow, the playlist now references segments 1008-1012. In this example, 2 segments were missed and part of the video is corrupted.

You may specify a longer playlist (please RTFM on output plugin hls). However, this would work only if the encoding process is limited in time. The media segments on disk would act as a buffer to compensate ffmpeg lower speed. Otherwise, if this is an endless continuous process, this will only delay the beginning of problems, not solve them.

Get a bigger CPU or use less greedy transcoding parameters in ffmpeg. It depends on the nature of the input stream. If your input is a complex high bitrate stream, then you need a bigger CPU anyway because the decoding process inside ffmpeg is slow. If the input is less complex to decode, you may simply use a simpler output encoding to fit in your current CPU.

tab1293, 2023-12-09:
@lelegard thank you for the detailed response, I really appreciate your help here. 

I believe you are correct in the assessment that at some point ffmpeg is not keeping up with processing the SRT stream, and therefore I see the "no room to store incoming packet" from the SRT layer. But I don't think this is because lack of resources. This is issue is happening when I am using NVIDIA GPUs to do the decoding and re-encoding of the stream. I've also tried using machines that have 16 [C2](https://cloud.google.com/compute/docs/compute-optimized-machines#c2_machine_types) CPUs. Here is the ffmpeg I am using with the GPU:

lelegard, 2023-12-09:
> What's really confusing me though is that transcoding the concatenated TS files after the fact with the same command does not have any of these issues. It's only happening in real time

The next test is quite simple. Use an input file containing N seconds of that exact same video content and transcode it using your ffmpeg command. If the processing time is greater than N seconds, it won't work in real time on that machine. No way, as simple as that.

From your command, we see that you transcode a 27 Mb/s AVC video into 5 Mb/s AVC. This is huge, especially decoding the 27 Mb/s AVC.

You didn't mention that you had plenty of CPU resources. Your Google Cloud instances may be impressive on paper but you should verify that the right resources are actually used. I cannot help here. I suggest to verify what does `ffmpeg -hwaccel nvdec -hwaccel_output_format cuda` actually do on your system. Using the above mentioned N-second test, try with and without these options. How is the performance ratio? From the options, it seems that a Nvidia GPU is requested for encoding the 5 Mb/s. But what about the 27 Mb/s decoding? As another test, I suggest to retry the same ffmpeg command on a 5 Mb/s file (the result of a previous transcoding for instance). How do 27 and 5 Mb/s decoding compare?

Anyway, this is an ffmpeg tuning issue. On a TSDuck perspective, I cannot help further.

tab1293, 2023-12-13:
Marking this as closed. We were able to avoid these TS packet issues in our flow by having TSDuck pass through to ffmpeg via UDP. ffmpeg then transcodes and outputs HLS and our upstream service is able to consume this rather than using RTMP. 

Thanks for your help @lelegard

closed 2023-12-13 by tab1293

jinnkhan88, 2024-04-24:
hey @tab1293 can you please share how you are doing the pass through to ffmpeg ? I am having sort of similar issues i am consuming an srt and creating a hls .ts packets however i get packet corrupt error. any help would be great.