tsduck/tsduck #1377: [BUG] tsduck identifies DTS-HD MA sound stream as SCTE-35 Splice Info

svlobanov, 2023-12-09:
[dts-hd-ma.ts.gz](https://github.com/tsduck/tsduck/files/13624202/dts-hd-ma.ts.gz)
#### Bug description:
tsduck identifies DTS-HD MA streams as SCTE-35 Splice Info

#### How to reproduce:
Run `tsanalyze dts-hd-ma.ts`
```|  0x0100  PMT .......................................... C       13,997 b/s  |
|  0x1001  PCR (not otherwise referenced) ............... C       18,080 b/s  |
|  0x1011  MPEG-2 Video (1920x1080i, @25 Hz, 16:9, 4:2:2  C   27,335,274 b/s  |
|  0x1100  SCTE 35 Splice Info (eng) .................... C    3,200,682 b/s  |
|  0x1101  MPEG-2 Audio (eng, Audio layer II, 384 kb/s,   C      428,663 b/s  |
```

#### Expected behavior:
Identify PID=0x1100 as 'DTS-HD MA' audio

#### Errors and logs:
no errors, no logs

#### Environment:
 - OS: macOS
 - OS version: 14.1
 - TSDuck full version: 3.35-3258
 - Installation type: homebrew

#### Additional information:
ffprobe (6.1) output:
```
Input #0, mpegts, from 'dts-hd-ma.ts':
  Duration: 00:00:02.70, start: 4200.000000, bitrate: 29599 kb/s
  Program 1 
  Stream #0:0[0x1011]: Video: mpeg2video (4:2:2) ([2][0][0][0] / 0x0002), yuv422p(tv, top first), 1920x1080 [SAR 1:1 DAR 16:9], 29.97 fps, 29.97 tbr, 90k tbn
    Side data:
      cpb: bitrate max/min/avg: 26580000/0/0 buffer size: 6651904 vbv_delay: N/A
  Stream #0:1[0x1100](eng): Audio: dts (DTS-HD MA) ([134][0][0][0] / 0x0086), 48000 Hz, 5.1(side), s32p (24 bit)
  Stream #0:2[0x1101](eng): Audio: mp2 ([4][0][0][0] / 0x0004), 48000 Hz, stereo, fltp, 384 kb/s
```

mediainfo output (audio stream related):
```
Audio #1
ID                                       : 4352 (0x1100)
Menu ID                                  : 1 (0x1)
Format                                   : DTS XLL
Format/Info                              : Digital Theater Systems
Commercial name                          : DTS-HD Master Audio
Muxing mode                              : Stream extension
Codec ID                                 : 134
Duration                                 : 2 s 507 ms
Bit rate mode                            : Variable
Channel(s)                               : 6 channels
Channel layout                           : C L R Ls Rs LFE
Sampling rate                            : 48.0 kHz
Frame rate                               : 93.750 FPS (512 SPF)
Bit depth                                : 24 bits
Compression mode                         : Lossless
Delay relative to video                  : 21 ms
Language                                 : English
```

lelegard, 2023-12-09:
Hi @svlobanov 

Which standard is this stream supposed to be? DVB?

MPEG-2 video for 1920x1080i at 27 Mb/s is not a common broadcast or streaming encoding. They all use AVC or HEVC. A master feed maybe?

The signalisation is quite minimal:
~~~
$ tspsi dts-hd-ma.ts 

* PAT, TID 0 (0x00), PID 0 (0x0000)
  Version: 0, sections: 1, total size: 20 bytes
  - Section 0:
    TS id:       1 (0x0001)
    NIT:         0 (0x0000)  PID:   31 (0x001F)
    Program:     1 (0x0001)  PID:  256 (0x0100)

* PMT, TID 2 (0x02), PID 256 (0x0100)
  Version: 0, sections: 1, total size: 55 bytes
  - Section 0:
    Program: 1 (0x0001), PCR PID: 4097 (0x1001)
    Program information:
    - Descriptor 0: Registration (0x05, 5), 4 bytes
      Format identifier: 0x48444D56 ("HDMV")
    - Descriptor 1: unknown (0x88, 136), 4 bytes
      0000:  0F FF FC FC                                ....
    Elementary stream: type 0x02 (MPEG-2 Video), PID: 4113 (0x1011)
    Elementary stream: type 0x86 (SCTE 35 Splice Info), PID: 4352 (0x1100)
    - Descriptor 0: ISO-639 Language (0x0A, 10), 4 bytes
      Language: eng, Type: 0x00 (undefined)
    Elementary stream: type 0x04 (MPEG-2 Audio), PID: 4353 (0x1101)
    - Descriptor 0: ISO-639 Language (0x0A, 10), 4 bytes
      Language: eng, Type: 0x00 (undefined)
~~~

The PID has stream type 0x86, which is commonly used for SCTE 35. This is the reason why TSDuck identifies it as such.

Ffprobe and mediainfo are audio/video tools. They look into the audio data and identify it as DTS-HD. TSDuck is a transport stream tool. It looks into the signalisation and identifies it as SCTE 35.

This is of course DTS-HS audio because mediainfo and ffprobe say so, and the bitrate, and the language descriptor. But its signalisation is at best unusual and unknown to me.

On an MPEG standpoint, the stream type 0x86 is "user private". So, we have to look into higher-level standards. In DVB, the stream type for DTS-xx shall be 0x06 (not 0x86), see ETSI TS 101 154 section 4.1.6.1.

For other standards, I do not know. This is why I asked which standard is that TS supposed to be compliant with.

svlobanov, 2023-12-09:
The only thing I know regarding the source of this file is "C-band satellite capture", sorry no more info

lelegard, 2023-12-09:
After googling a bit, the registration descriptor with "HDMV" seems to indicate a BluRay disk. However, I am not aware of BluRay using MPEG2 encoding, they should use AVC/H.264.

Master Google also says that DTS-HD should have stream type 0x86 in the PMT of a BluRay disk with HDMV convention. Thus, your stream looks like a BDrip with the weird usage of a massive MPEG2 video encoding.

Anyway, not really a "standard" stuff.

svlobanov, 2023-12-09:
Looks like, audio stream was transcoded and maybe some PIDs were deleted. If you don't have plans to support hdmv convention, then let's close the issue

I've just searched for yuv422 video and found this sample, then I observed this strange behaviour and decided and decided to report a bug.

lelegard, 2023-12-10:
I have added the specific interpretation of stream type per registration id, including "HDMV". However, there is no public specification of BluRay and no way to know the exact list of stream types for "HDMV" registration id. Based on several other open source projects, I have found partial lists, never the same and sometimes contradictory. So, the result is not 100% accurate.

In your case, DTS-HD Master Audio is now correctly identified.

svlobanov, 2023-12-10:
Thanks a lot! Will compile from git and recheck next week

svlobanov, 2023-12-15:
It works, thank you!

closed 2023-12-15 by svlobanov