tsduck/tsduck #1378: [Q] DVB-SIS

EnoSat, 2023-12-10:
5 West - 12522 V tests DVB-SIS. It is assigned SID-700.
[image]
PID 3002 - descriptor 0xF0
PID-3003 - descriptor 0x90
PID-3005 - descriptor 0x00 - PAT
PID-3006 - descriptor 0x02 - PMT for FRANCE 2
PID-3007 - descriptor 0x02 - PMT for FRANCE 3
PID-3015 - descriptor 0x01 - CAT
PID-3016 - descriptor 0x40 - NIT for DVBT
PID-3017 - descriptor 0x42 - SDT
PID-3018 - descriptor 0x4E - EIT
PID-3019 - descriptor 0x70/0x73 - TDT/TOT
Is it possible to create a TS for DVBT via TSDuck?

lelegard, 2023-12-10:
Hi @EnoSat 

I do not know DVB-SIS. Is it something equivalent to T2-MI? Some encapsulation of a DVB-T2 TS inside a satellite TS?

There is no way to extract the DVB-T2 stream with TSDuck, no equivalent to the `t2mi` plugin.

EnoSat, 2023-12-10:
DVB-SIS one TS for DVBT/DVBS, in this particular case DVB-T2 is used (they don't use DVB-T2-MI yet)
https://www.enensys.com/solution/dvb-t2-single-illumination-dvb-t2-with-dth-co-existence/
https://www.youtube.com/watch?v=dsiV3u-q8pg
TS file - https://mega.nz/file/MN11nTTT#3JvpI5uNeexqZumx0_b-n2Q-qU-8EOzSjAtMTk51o70

EricBerendsen, 2024-01-14:
@EnoSat I do not want to steal someone else's thread, but this might be a interesting standard to add to DVB Inspector. Not to create a DVBT TS from it (sorry), but to show all meta data. Would you have any more test data (other streams?)

skunge2000, 2024-01-15:
More information here : https://dvb.org/?standard=metadata-generation-and-deterministic-dvb-t-megaframe-dvb-t2-mi-stream-generation-from-mpeg-2-transport-streams-for-a-dvb-single-illumination-system