tsduck/tsduck #1379: [FEAT] bitrate_monitor add json-line

enricovittorini, 2023-12-11:
It would be great to add the JSON line output format for the bitrate monitor plugin.

Possibly in addition to the total TS and Net bitrate would be great to attnthe stuffing bitrate

lelegard, 2023-12-11:
> It would be great to add the JSON line output format for the bitrate monitor plugin.

That should be easy.

> Possibly in addition to the total TS and Net bitrate would be great to attnthe stuffing bitrate

The stuffing bitrate is simply the total minus net bitrate. Seems redundant, isn't it?

enricovittorini, 2023-12-11:
thanks @lelegard, I agree regarding the stuffing, but was just to have something complete at the output. As you prefer, in my case would not be be an issues to do the math in my code instead of accessing the key value.

lelegard, 2023-12-11:
Option `--json-line` added.

enricovittorini, 2023-12-11:
thanks @lelegard

enricovittorini, 2023-12-12:
hello @lelegard, would it be possible to implement also the json-line for the continuity plugin?

lelegard, 2023-12-12:
> would it be possible to implement also the json-line for the continuity plugin?

Done

closed 2024-05-12 by github-actions[bot]