tsduck/tsduck #1386: [FEAT] show pcap timestamp in the output of pcrextract plugin

svlobanov, 2023-12-15:
#### Is your feature request related to a problem?
I would like to check clock generator accuracy of the source signal via collecting pcap dumps of the signal for several days and compare the difference of PCR values with expected difference.

#### Proposed solution:
Using: tsp -I pcap dump.pcap -P pcrextract -O drop
My solution is to collect pcap dumps during the whole day(s), extract PCR along with pcap timestamps (on the host synchronized with world time), then recalculate PCR value based on pcap timestamps and compare the difference between expected PCR value and PCR value from the signal.
For me, the accuracy of pcap timestamps is enough, because we are talking about the error of the frequency generator about several seconds per day (like general PC clock without ntp sync).
If pcrextract shows pcap timestamp of the packet contained extracted ts packet, then it will be enough for me.

#### Alternatives:
Calculate pcap packet number by ts packet number shown in the output of pcrextract plugin (might be not easy if we have ip packet with less than 7 ts packets) and extract timestamp from pcap by pcap packet number.

#### Additional information:
no additional info

lelegard, 2023-12-16:
There is often a confusion between PCR, the Program Clock Reference of the transport stream, and the wall-clock time, in your case the capture time stamps. The PCR exists primarily because there is no reason for it to be synchronous with the wall-clock time (except maybe in a pure lossless synchronous modulated broadcast reception). The reception of packets can be desynchronised from the TS PCR, especially in an UDP/IP reception where TS packets are received in bursts of UDP datagrams. The purpose of the PCR is precisely to resynchronise the playout in the receiver or player.

Resynchronizing the PCR with the wall-clock time is always a mistake because it corrupts the playout clock and results in incorrect video rendering.

Let's assume I developed this feature:

1. It will take a lot of my time.
2. It will brings nothing to me.
3. It will fool users into thinking that this is a legitimate operation.
4. They will use it, the result will be buggy, they will report issues here, it will take a lot of my time explaining why they should not do this.

svlobanov, 2023-12-16:
Thank you for your reply and great tool you provide to the community.

My case is a special one, not like TV or STB receiver that's why I'm comparing PCR clock and receiver clock.

Anyway, I've found a simple way with tshark: `tshark -r dump.pcap -Y "mp2t.af.pcr" -T fields -e mp2t.pid -e frame.time_epoch -e mp2t.af.pcr`

closed 2023-12-16 by svlobanov