tsduck/tsduck #1387: [Q] How to keep srt port active until hls m3u8 #EXT-X-ENDLIST is found in file

shivambhatia5432, 2023-12-16:
I am running command:
`tsp --debug -I hls --infinite --receive-timeout 200000000  --connection-timeout 20000000 foo.m3u8 -O srt --listener  127.0.0.1:9999 -m `
I am processing ts videos and adding to a foo.m3u8 file like this:
```
#EXTM3U
#EXT-X-VERSION:3
#EXT-X-INDEPENDENT-SEGMENTS
#EXT-X-TARGETDURATION:10
#EXT-X-MEDIA-SEQUENCE:0
#EXT-X-PLAYLIST-TYPE:EVENT
#EXTINF:2.912,
#EXT-X-BITRATE:2683
foo-000000.ts
#EXTINF:1.752,
#EXT-X-BITRATE:4460
foo-000001.ts
#EXTINF:1.366,
#EXT-X-BITRATE:5718
foo-000002.ts
```

I don't want my srt stream to stop until #EXT-X-ENDLIST in m3u8 file is found because I will keep adding ts chunks in file till then. Is there anyway this is possible using tsduck?

lelegard, 2023-12-16:
Hi @shivambhatia5432 

HLS is a streaming protocol, meaning that the segments shall be delivered on time.

Looking at the code (I don't remember my mindset at the time I wrote it), a playlist file is regularly reloaded, looking for more segments, when this playlist is "updatable", meaning it is of type LIVE or EVENT. It stops being updatable when `#EXT-X-ENDLIST` is found.

However, the streaming must be continuous. Assuming that:
- a given version of a playlist was read at time T0,
- it contains N segments,
- the total announced duration of all segments is D.

Then, we compute the end of the current playlist at T0 + D. If at time T0 + D the reloaded version of the playlist has no more segment, then the playlist is considered completed because streaming is interrupted. In real conditions, segments are delivered slightly in advance to guarantee a continuous streaming.

So, if you declare your playlist as LIVE or EVENT and you provide your segments on-time and correctly update the playlist, you should be safe. Otherwise, this is not considered as valid HLS and the stream is terminated.

shivambhatia5432, 2023-12-16:
Hey @lelegard,
Thank you so much for replying, I am adding chunks in same duration only, but still new chunks are not getting to srt output.

akshhack, 2023-12-16:
Hi @lelegard can you please point me to the code you're referring to and also if there is a way to increase D without necessarily guaranteeing very strict delivery?

lelegard, 2023-12-16:
>  I am adding chunks in same duration only

The problem is not having equal duration (they are always slightly different in real HLS anyway), the problem is to provide new segments and update the playlist in time.

For instance, if your playlist currently references two segments of 10 seconds and if 20 seconds later the playlist of not updated with new segments, then the streaming is broken and stops.

lelegard, 2023-12-16:
> can you please point me to the code you're referring to 

See [here](https://github.com/tsduck/tsduck/blob/1fbdfd645c72cdfaa59bed68264a4b0ac773e1df/src/libtsduck/plugins/plugins/tshlsInputPlugin.cpp#L399), especially this:

~~~
while (... && Time::CurrentUTC() <= _playlist.terminationUTC() && ...) {
~~~

This test means that when the current time is after time of playlist load + duration of all segments, we stop.

closed 2024-05-19 by github-actions[bot]