tsduck/tsduck #1388: [Q] Stream series of dynamically created mp4 files out as SRT

akshhack, 2023-12-16:
Hey, 

I'm quite new to the area of streaming. I'd like to know if there's a reliable way for me to stream a series of mp4 files out as SRT using tsduck.

Caveat: These mp4 files are being generated dynamically and I want to keep an SRT connection open that "adds" every new mp4 to the SRT out stream. 

There isn't an obvious way at the moment except for me to keep recreating the SRT connection, but that requires the client to keep reconnecting which isn't a scalable solution. 

Thank you so much @lelegard, this is one of the best Github projects I've ever seen!

lelegard, 2023-12-16:
Hi @akshhack 
TSDuck is a transport stream processing tools. It can manipulate TS streams only. MP4 files are not TS files, the container format is different. You may try ffmpeg to convert the files and, possibly, directly stream the result in SRT.

akshhack, 2023-12-16:
Hey @lelegard, yes let's say I have a bunch of .ts files that are being dynamically generated. What would you suggest the best approach to take these and stream through SRT without needing the client to reconnect to fresh connections each time?

Thank you for the prompt response!

akshhack, 2023-12-16:
Hey @lelegard, 

`tsp -I hls foo.m3u8 --live -O srt -l 127.0.0.1:9999 -m` is the ``--live` flag equivalent in functionality to the `--event` flag when HLS is in input, in effect?

lelegard, 2023-12-16:
Hi @akshhack 

If I remember correctly, on input, the only difference between LIVE and EVENT is the starting point. The first time you load the playlist, TSDuck will start with the first segment of the playlist when it is EVENT and with the last one when it is LIVE.

An event shall be viewed from the beginning, even if it started a long time ago. Its playlist grows: new segments are added but the first ones are never deleted.

A live stream shall be viewed at "current" point, starting with the last available segment. The playlist is typically a sliding window of N segments. At each update, the first segment is removed and a new one is added at the end.

If you do something that is not really HLS but you want it to be ingested by the HLS input plugin (or any other HLS player), then you should follow the above rules. See all #1387 for a similar situation.

closed 2024-05-19 by github-actions[bot]