tsduck/tsduck #1389: [BUG] Segfault or unknown Error with MPE depacketisation

Aang23, 2023-12-17:
#### Bug description:
I am attempting to process a stream being received from a TBS5930 as MPE : 
`tsp --verbose --debug=3 -I dvb --device-name /dev/dvb/adapter0 --lnb Extended --delivery-system "DVB-S2" --frequency 11262000000 --polarity vertical --modulation 8-PSK --symbol-rate 33000000 --local-address 127.0.0.1 -P mpe --udp-forward -O drop --log`

However, it will only result in the follow logs, or a straight up Segfault when not providing --local-address.

```
tsp: Debug[3]: debug level set to 3
tsp: Debug: ====> tsp --verbose --debug=3
* Debug[3]: debug level set to 3
* Debug: trying to load "/usr/bin/tsplugin_dvb.so"
* Debug: /usr/bin/tsplugin_dvb.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/bin/dvb.so"
* Debug: /usr/bin/dvb.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsduck/tsplugin_dvb.so"
* Debug: /usr/lib64/tsduck/tsplugin_dvb.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsduck/dvb.so"
* Debug: /usr/lib64/tsduck/dvb.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsplugin_dvb.so"
* Debug: /usr/lib64/tsplugin_dvb.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/dvb.so"
* Debug: /usr/lib64/dvb.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib/tsduck/tsplugin_dvb.so"
* Debug: registering input plugin "dvb", status: ok
* Debug[3]: dvb: debug level set to 3
* Error: dvb: unknown option --local-address
* Error: dvb: no parameter allowed, use options only
* Debug: dvb: ====> tsp -I dvb --device-name /dev/dvb/adapter0 --lnb Extended --delivery-system DVB-S2 --frequency 11262000000 --polarity vertical --modulation 8-PSK --symbol-rate 33000000 --local-address 127.0.0.1
* Debug[3]: drop: debug level set to 3
* Error: drop: unknown option --log
* Debug: drop: ====> tsp -O drop --log
* Debug: trying to load "/usr/bin/tsplugin_mpe.so"
* Debug: /usr/bin/tsplugin_mpe.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/bin/mpe.so"
* Debug: /usr/bin/mpe.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsduck/tsplugin_mpe.so"
* Debug: /usr/lib64/tsduck/tsplugin_mpe.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsduck/mpe.so"
* Debug: /usr/lib64/tsduck/mpe.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsplugin_mpe.so"
* Debug: /usr/lib64/tsplugin_mpe.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/mpe.so"
* Debug: /usr/lib64/mpe.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib/tsduck/tsplugin_mpe.so"
* Debug: registering processor plugin "mpe", status: ok
* Debug[3]: mpe: debug level set to 3
* Debug: mpe: ====> tsp -P mpe --udp-forward
* Debug: error when initializing the plugins
* Debug: Report logging thread terminated
```

#### Environment:
 - OS: Linux 6.5.6
 - OS version: Debian Sid
 - TSDuck full version: 3.36-3528 & Latest commit at the time of writing
 - Installation type: Both

#### Additional information:
This was not an issue in the past, but I'm not sure of the version I was using at the time.

lelegard, 2023-12-17:
Hi @Aang23 

It is of course impossible to do anything with so little information.

First, we need to know if this is a segfault (a crash) or an "unknown error" message.

Second, we need to know which plugin generates an error. So, try with `dvb` but without `mpe`. If it works, then this must be an issue with `mpe`. In that case, first save a few seconds or one minute of content using `tsp -I dvb ... -P until --seconds 30 -O file ...` and then use that file as input with the `mpe` plugin. If you can reproduce the error, then please upload the input TS file somewhere so that I can reproduce the problem.

Aang23, 2023-12-17:
Sorry if I wasn't clear : 
- Everything works fine without MPE, and I get a TS on stdout. The issue is only with `mpe`
- Without `--local-adress` it will segfault
- With `--local-adress` it will error out without any useful log

From the TS file : 
`tsp -I file ./outpufile.ts -P mpe --udp-forward -O drop` will segfault.
`tsp -I file ./outpufile.ts -P mpe --local-address 127.0.0.1 --udp-forward -O drop` also segfaults.

The TS : 
https://we.tl/t-JeQMdUXozt

lelegard, 2023-12-17:
Thanks for the details. This is a regression (for which there was no non-regression test). The plugin mpe is broken in version 3.36 in the analysis of its arguments and cannot be used at all.

The fix is quite simple. I just fixed it in the repo. So, you have to rebuild from the repo.

This is a severe issue which may need an anticipated release for version 3.37.

Aang23, 2023-12-18:
Thanks, I will test later today.

closed 2023-12-18 by Aang23