tsduck/tsduck #1392: [Q] clarification on splicemonitor plugin

enricovittorini, 2023-12-19:
tsduck version 3.36-3528 on windows 10

I run the following command:

and I notice a difference when i run it with or without specifying the video pid of the service wiht "-t" option.

With "-t 517" option:

` splicemonitor: { "#name": "event", "count": 1, "event-id": 61602001, "event-pts": 1664448970, "event-type": "out", "packet-index": 796573, "progress": "pending", "splice-pid": 961, "time-to-event-ms": 5811 }
* splicemonitor: { "#name": "splice_information_table", "#nodes": [ { "#name": "metadata", "pid": 961 }, { "#name": "splice_insert", "#nodes": [ { "#name": "break_duration", "auto_return": true, "duration": 1800000 } ], "avail_num": 0, "avails_expected": 0, "out_of_network": true, "pts_time": 1664349856, "splice_event_cancel": false, "splice_event_id": 61602001, "splice_immediate": false, "unique_program_id": 0 }, { "#name": "splice_avail_descriptor", "identifier": 1129661769, "provider_avail_id": 61602001 } ], "protocol_version": 0, "pts_adjustment": 99114, "tier": 4095 }
* splicemonitor: { "#name": "event", "count": 1, "event-id": 61602001, "event-pts": 1664448970, "event-type": "out", "packet-index": 1002301, "pre-roll-ms": 5812, "progress": "occurred", "splice-pid": 961, "status": "normal" }`

Without -t option

`* splicemonitor: { "#name": "event", "count": 1, "event-id": 61602001, "event-pts": 1664448970, "event-type": "out", "packet-index": 796573, "progress": "pending", "splice-pid": 961, "time-to-event-ms": -28968561 }
* splicemonitor: { "#name": "splice_information_table", "#nodes": [ { "#name": "metadata", "pid": 961 }, { "#name": "splice_insert", "#nodes": [ { "#name": "break_duration", "auto_return": true, "duration": 1800000 } ], "avail_num": 0, "avails_expected": 0, "out_of_network": true, "pts_time": 1664349856, "splice_event_cancel": false, "splice_event_id": 61602001, "splice_immediate": false, "unique_program_id": 0 }, { "#name": "splice_avail_descriptor", "identifier": 1129661769, "provider_avail_id": 61602001 } ], "protocol_version": 0, "pts_adjustment": 99114, "tier": 4095 }
* splicemonitor: { "#name": "event", "count": 1, "event-id": 61602001, "event-pts": 1664448970, "event-type": "out", "packet-index": 797687, "pre-roll-ms": 31, "progress": "occurred", "splice-pid": 961, "status": "normal"`

A couple of additional questoins:

- Would it be possible to a a "time" property in the logged command to collect when the event has been registered by tsduck?  Something like "event-time": "2023/12/19 10:20:30"

- From my understanding, the first line logged is the command reception. I guess it would be useful to name it "event-type": "command".

thanks!

lelegard, 2023-12-19:
Hi @enricovittorini 

> I notice a difference when i run it with or without specifying the video pid of the service

The only difference I can spot between the two displays is the `packet-index` in the last line. Is there something else?

About this one, the last line has `"pre-roll-ms": 5812, "progress": "occurred"`. This means that the event already occured and this line is only here to display the pre-roll time. Therefore, this line is not added on reception of a splice information table. It is displayed when the plugin receives a PTS which is after the programmed event. At that time, and only at that time, the pre-roll time can be computed.

Now, the difference is that the pre-roll time is displayed earlier without `--time-pid` option (lower packet index in the TS). This is normal. When you specify `-t 517`, the video PID, you mean "only look at PID 517 for PTS". By default, the plugin reads all PMT's and associates each SCTE 35 PID with all audio/video PID's of the same service. So, in your case, after the event occurred, the first packet in the service with a PTS is probably in an audio packet. So, without `-t 517`, this audio packet is used to determine that the event is past. With `-t 517`, you need to wait for the next PTS on the video PID, which occurred later. In general, audio PES packets are smaller and more frequent than video packets. So, in general, you get new PTS more often on audio PID's. This explains the difference.

> Would it be possible to a a "time" property in the logged command to collect when the event has been registered by tsduck?

Everything is possible, it's just code. However, TSDuck is designed to be independent of real-time in the general case. When analyzing a stream file, the wall-clock time is meaningless. Only a few plugins such as `time` and `timeref` use the wall-clock time for something.

> From my understanding, the first line logged is the command reception. I guess it would be useful to name it "event-type": "command".

No, it shouldn't be. The "event" is not the reception of an SCTE 35 command (everything is a command anyway). The "event" refers to the future "in" or "out" event which is described in the command.

The second line is just an XML-translated-to-JSON dump of the splice information table because you specified `--select-commands`, which is a superset of `--display-commands` (RTFM for details). Without any of these two options, the details of the splice information tables are not displayed, only information about events (which, again, are not the same thing as "receiving an SCTE 35 command").

enricovittorini, 2023-12-19:
hello @lelegard ,

I would not expect a big difference between the pre-roll evaluation when splicemonitor look at the audio or video.
Being the pre-roll the time a scte-35 message precedes that insertion, I would expect the reported value to be very similar in both cases:
"pre-roll-ms": 31 without -t
"pre-roll-ms": 5812 with -t

lelegard, 2023-12-19:
I didn't notice that. 5812 is very realistic. 31 is not at all. My hypothesis for the 31 ms in the second case:
- Get several splice insert before.
- Got PTS from an audio PID > event PTS => declare the event as completed.
- Got a last splice event for same event id => reopen a new future event (same PTS in fact)
- Got PTS from video PID (or audio PID) > event PTS) => declare the event as completed with a very short pre-roll time

We could imagine other scenarios. Is this reproduceable at will from a TS file or just some random results from a live stream?

Anyways, the plugin has a very simple way of working. Have a look at the code and compare this with the output, you will find the explanation.

The plugin just reports events on a TS and display results. Now, SCTE 35 is not a fully synchronized mechanism. The PTS of the various audio/video PID's are not synchronized in the TS. This is the job of the receiver to resynchonize them. So, several interpretations are possible when the server inserts a last splice information table too close to the actual event.

enricovittorini, 2023-12-19:
hello @lelegard 
these result are replicable as they come from a recorded TS. results are always the same.

The splice commands are back-to-back, 20s avail slot each.
For sure the command is sent enough in advance between 5 to 7 seconds before the splice is happening.

These are the sequence of commands:

--------------------------- DPI overview ---------------------------
Avails: 8

------------------------ DPI Avail #1 ---------------------------
Number / expected: 0 / 0
Splice mode: Program
Splice Out Program ID / Event ID: 0 / 61601401
Splice In Program ID / Event ID: no Splice In message.
------------------------ Out picture
Immediate: No
PTS (90kHz): 1662648970
PTS (s): 18473.877
Pic index / type: 207 / I
------------------------ In picture
Immediate: No
PTS (90kHz): 1664448970
PTS (s): 18493.877
Pic index / type: 707 / I
------------------------ Avail Length
Pictures: 500
Duration (90kHz): 1800000
Duration (s): 20.000

------------------------ DPI Avail #2 ---------------------------
Number / expected: 0 / 0
Splice mode: Program
Splice Out Program ID / Event ID: 0 / 61602001
Splice In Program ID / Event ID: no Splice In message.
------------------------ Out picture
Immediate: No
PTS (90kHz): 1664448970
PTS (s): 18493.877
Pic index / type: 707 / I
------------------------ In picture
Immediate: No
PTS (90kHz): 1666248970
PTS (s): 18513.877
Pic index / type: 1207 / I
------------------------ Avail Length
Pictures: 500
Duration (90kHz): 1800000
Duration (s): 20.000

------------------------ DPI Avail #3 ---------------------------
Number / expected: 0 / 0
Splice mode: Program
Splice Out Program ID / Event ID: 0 / 61602201
Splice In Program ID / Event ID: no Splice In message.
------------------------ Out picture
Immediate: No
PTS (90kHz): 1666248970
PTS (s): 18513.877
Pic index / type: 1207 / I
------------------------ In picture
Immediate: No
PTS (90kHz): 1668048970
PTS (s): 18533.877
Pic index / type: 1707 / I
------------------------ Avail Length
Pictures: 500
Duration (90kHz): 1800000
Duration (s): 20.000

------------------------ DPI Avail #4 ---------------------------
Number / expected: 0 / 0
Splice mode: Program
Splice Out Program ID / Event ID: 0 / 61602401
Splice In Program ID / Event ID: no Splice In message.
------------------------ Out picture
Immediate: No
PTS (90kHz): 1668048970
PTS (s): 18533.877
Pic index / type: 1707 / I
------------------------ In picture
Immediate: No
PTS (90kHz): 1669848970
PTS (s): 18553.877
Pic index / type: 2207 / I
------------------------ Avail Length
Pictures: 500
Duration (90kHz): 1800000
Duration (s): 20.000

------------------------ DPI Avail #5 ---------------------------
Number / expected: 0 / 0
Splice mode: Program
Splice Out Program ID / Event ID: 0 / 61603001
Splice In Program ID / Event ID: no Splice In message.
------------------------ Out picture
Immediate: No
PTS (90kHz): 1669848970
PTS (s): 18553.877
Pic index / type: 2207 / I
------------------------ In picture
Immediate: No
PTS (90kHz): 1671648970
PTS (s): 18573.877
Pic index / type: 2707 / I
------------------------ Avail Length
Pictures: 500
Duration (90kHz): 1800000
Duration (s): 20.000

------------------------ DPI Avail #6 ---------------------------
Number / expected: 0 / 0
Splice mode: Program
Splice Out Program ID / Event ID: 0 / 61603201
Splice In Program ID / Event ID: no Splice In message.
------------------------ Out picture
Immediate: No
PTS (90kHz): 1671648970
PTS (s): 18573.877
Pic index / type: 2707 / I
------------------------ In picture
Immediate: No
PTS (90kHz): 1673448970
PTS (s): 18593.877
Pic index / type: 3207 / I
------------------------ Avail Length
Pictures: 500
Duration (90kHz): 1800000
Duration (s): 20.000

------------------------ DPI Avail #7 ---------------------------
Number / expected: 0 / 0
Splice mode: Program
Splice Out Program ID / Event ID: 0 / 61603401
Splice In Program ID / Event ID: no Splice In message.
------------------------ Out picture
Immediate: No
PTS (90kHz): 1673448970
PTS (s): 18593.877
Pic index / type: 3207 / I
------------------------ In picture
Immediate: No
PTS points to a picture that is out of stream.
------------------------ Avail Length
PTS points to a picture that is out of stream.

------------------------ DPI Avail #8 ---------------------------
Number / expected: 0 / 0
Splice mode: Program
Splice Out Program ID / Event ID: 0 / 61604001
Splice In Program ID / Event ID: no Splice In message.
------------------------ Out picture
Immediate: No
PTS points to a picture that is out of stream.
------------------------ In picture
Immediate: No
PTS points to a picture that is out of stream.
------------------------ Avail Length
PTS points to a picture that is out of stream.

lelegard, 2023-12-19:
Please upload a subset of the TS which reproduces the problem.

lelegard, 2023-12-22:
> Would it be possible to a a "time" property in the logged command to collect when the event has been registered by tsduck? Something like "event-time": "2023/12/19 10:20:30"

Option `--time-stamp` has been added.

enricovittorini, 2023-12-22:
> > Would it be possible to a a "time" property in the logged command to collect when the event has been registered by tsduck? Something like "event-time": "2023/12/19 10:20:30"
> 
> Option `--time-stamp` has been added.

Thanks Thierry  will give it a try.

enricovittorini, 2023-12-23:
@lelegard  thanks the "--time-stamp" works.

one more question, to have the full splice command what should I do?

something like what is produced by [threefive ](https://github.com/futzu/scte35-threefive)

`{
  "info_section": {
    "table_id": "0xfc",
    "section_syntax_indicator": false,
    "private": false,
    "sap_type": "0x3",
    "sap_details": "No Sap Type",
    "section_length": 158,
    "protocol_version": 0,
    "encrypted_packet": false,
    "encryption_algorithm": 0,
    "pts_adjustment_ticks": 0,
    "pts_adjustment": 0,
    "cw_index": "0x2",
    "tier": "0xfff",
    "splice_command_length": 5,
    "splice_command_type": 6,
    "descriptor_loop_length": 136,
    "crc": "0xe5204b77"
  },
  "command": {
    "command_length": 5,
    "command_type": 6,
    "name": "Time Signal",
    "time_specified_flag": true,
    "pts_time": 27640.4891,
    "pts_time_ticks": 2487644019
  },
  "descriptors": [
    {
      "tag": 2,
      "descriptor_length": 31,
      "name": "Segmentation Descriptor",
      "identifier": "CUEI",
      "components": [],
      "segmentation_event_id": "0x2193446e",
      "segmentation_event_cancel_indicator": false,
      "program_segmentation_flag": true,
      "segmentation_duration_flag": true,
      "delivery_not_restricted_flag": false,
      "web_delivery_allowed_flag": true,
      "no_regional_blackout_flag": true,
      "archive_allowed_flag": true,
      "device_restrictions": "No Restrictions",
      "segmentation_duration": 20,
      "segmentation_duration_ticks": 1800000,
      "segmentation_message": "Break Start",
      "segmentation_upid_type": 1,
      "segmentation_upid_type_name": "Deprecated",
      "segmentation_upid_length": 11,
      "segmentation_upid": "first_Avail",
      "segmentation_type_id": 34,
      "segment_num": 0,
      "segments_expected": 0
    },
    {
      "tag": 2,
      "descriptor_length": 33,
      "name": "Segmentation Descriptor",
      "identifier": "CUEI",
      "components": [],
      "segmentation_event_id": "0x5d55cc33",
      "segmentation_event_cancel_indicator": false,
      "program_segmentation_flag": true,
      "segmentation_duration_flag": true,
      "delivery_not_restricted_flag": false,
      "web_delivery_allowed_flag": true,
      "no_regional_blackout_flag": true,
      "archive_allowed_flag": true,
      "device_restrictions": "No Restrictions",
      "segmentation_duration": 20,
      "segmentation_duration_ticks": 1800000,
      "segmentation_message": "Provider Placement Opportunity Start",
      "segmentation_upid_type": 1,
      "segmentation_upid_type_name": "Deprecated",
      "segmentation_upid_length": 11,
      "segmentation_upid": "first_Avail",
      "segmentation_type_id": 52,
      "segment_num": 0,
      "segments_expected": 0,
      "sub_segment_num": 0,
      "sub_segments_expected": 0
    },
    {
      "tag": 2,
      "descriptor_length": 33,
      "name": "Segmentation Descriptor",
      "identifier": "CUEI",
      "components": [],
      "segmentation_event_id": "0x53724e2e",
      "segmentation_event_cancel_indicator": false,
      "program_segmentation_flag": true,
      "segmentation_duration_flag": true,
      "delivery_not_restricted_flag": false,
      "web_delivery_allowed_flag": true,
      "no_regional_blackout_flag": true,
      "archive_allowed_flag": true,
      "device_restrictions": "No Restrictions",
      "segmentation_duration": 20,
      "segmentation_duration_ticks": 1800000,
      "segmentation_message": "Distributor Placement Opportunity Start",
      "segmentation_upid_type": 1,
      "segmentation_upid_type_name": "Deprecated",
      "segmentation_upid_length": 11,
      "segmentation_upid": "first_Avail",
      "segmentation_type_id": 54,
      "segment_num": 0,
      "segments_expected": 0,
      "sub_segment_num": 0,
      "sub_segments_expected": 0
    },
    {
      "tag": 2,
      "descriptor_length": 31,
      "name": "Segmentation Descriptor",
      "identifier": "CUEI",
      "components": [],
      "segmentation_event_id": "0x3b5665de",
      "segmentation_event_cancel_indicator": false,
      "program_segmentation_flag": true,
      "segmentation_duration_flag": true,
      "delivery_not_restricted_flag": false,
      "web_delivery_allowed_flag": true,
      "no_regional_blackout_flag": true,
      "archive_allowed_flag": true,
      "device_restrictions": "No Restrictions",
      "segmentation_duration": 20,
      "segmentation_duration_ticks": 1800000,
      "segmentation_message": "Provider Advertisement Start",
      "segmentation_upid_type": 1,
      "segmentation_upid_type_name": "Deprecated",
      "segmentation_upid_length": 11,
      "segmentation_upid": "first_Avail",
      "segmentation_type_id": 48,
      "segment_num": 0,
      "segments_expected": 0
    }
  ]
}`

lelegard, 2023-12-23:
The presentation may be different but everything is there. The TSDuck JSON output is reversible, it can be used to recreate the binary section. Everything is explained as XML in appendix D of the user's manual. The JSON format is an automated translation of the XML format. 

The splice commands have several alternative formats and optional fields. It is possible that the threefive format includes everything, including absent fields. But the TSDuck format includes by design everything in the actual section.

enricovittorini, 2023-12-24:
@lelegard thanks for the feedback.

Could you guide me on how to get a base64 or binary representation of the command in the "metadata" filed of the JSON?
even using a customer plugin like mysplicemonitor

something like:
```
{
	"#name": "metadata",
	"pid": 961,
	"time": "2023-12-23 08:14:10",
	"command": "/DCeAAAAAAAAAv/wBQb+lEZvcwCIAh9DVUVJIZNEbn/fAAAbd0ABC2ZpcnN0X0F2YWlsIgAAAiFDVUVJXVXMM3/fAAAbd0ABC2ZpcnN0X0F2YWlsNAAAAAACIUNVRUlTck4uf98AABt3QAELZmlyc3RfQXZhaWw2AAAAAAIfQ1VFSTtWZd5/3wAAG3dAAQtmaXJzdF9BdmFpbDAAAOUgS3c="
}
```

thanks!

lelegard, 2023-12-24:
What do you mean by "command"?

If you mean the tsp command, there is no obvious or portable way. A plugin is independent of the context. It can run as part of a tsp command or any custom Python, Java or C++ application using the TSProcessor class (which exists in all three languages).

enricovittorini, 2023-12-24:
@lelegard i mean "splice command". My goal is to have the base64 representation of the whole splice_info_section

lelegard, 2023-12-24:
Ok, you mean section. Quite feasible since the `handleTable()` method of the plugin receives a binary table. Could be implemented after the Christmas period.

lelegard, 2023-12-30:
Option `--meta-sections` added in `splicemonitor` (and `tstables` for consistency).

The `<metadata>` XML or JSON structure contains one `<section>` structure per section in the captured binary table. The content is hexadecimal data, not base64, for consistency with all other uninterpreted binary data in XML and JSON format in TSDuck.

closed 2024-06-02 by github-actions[bot]