tsduck/tsduck #1393: [Q] TSP dektec input plugin blocks the process if the port is not connected to any ASI stream

sathish25071992, 2023-12-20:
Hi,

We are trying to control the tsp's dektec input plugin via the control port to switch between channels on some user input. What we observe is, if we start the tsp dektec input plugin with channel where no input is connected, then the plugin simply blocks the process and controlling from the tspcontrol will not work (connection refused). I think the tsp hasn't started the server yet. In this case we are left with no choice but to restart the entire process manually.

So can you help us on any way to properly handle this situation? Below are the debug log of running the tsp with the dektec input plugin on the channel where no input is connected,

```
$ tsp --debug --control-port 5050 -I dektec --device 1 --channel 0
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp --debug --control-port 5050
* Debug: debug level set to 1
* Debug: dektec: debug level set to 1
* Debug: dektec: ====> tsp -I dektec --device 1 --channel 0
* Debug: file: debug level set to 1
* Debug: file: ====> tsp -O file 
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* dektec: using Dektec device 1, input channel 0 (DTA-2144B port 1)
* Debug: dektec: attaching to device DTA-2144B serial 0x000000007FCB0BC5
* Debug: dektec: attaching to port 1
* Debug: dektec: resetting channel, mode: 1
* Debug: dektec: setting RxControl, mode: 0
* Debug: dektec: clearing FIFO and flags
* Debug: dektec: getting FIFO max size
* Debug: dektec: max FIFO size: 33,554,432 bytes
* Debug: dektec: using FIFO size: 33,554,432 bytes
* Debug: dektec: setting RxMode, mode: 17
* Debug: dektec: setting RxControl, mode: 1
```

lelegard, 2023-12-20:
Hi @sathish25071992 

Yes, your observations are correct. The first thing that tsp does is initializing the input and trying to read the initial load. The idea is to avoid starting / creating / locking other resources when there is nothing to read from input or the input is invalid / incorrect.

In the case of a Dektec ASI board, if reading from an unplugged cable blocks forever, then the tsp initialization will wait forever. The initial behavior really depends on the input plugin and device.

TSDuck is a generic lab tool for experimentation. All input/output plugins have the same generic interface and basic expected behavior. If you want a precise behavior which depends on specificities of an input device, then you have to develop your own specialized application.

lelegard, 2023-12-20:
If you want the details of the tsp initialization sequence, you may have look at the [source code here](https://github.com/tsduck/tsduck/blob/1700f5f49e0f7e1d0638c55147bccb374fdcb658/src/libtsduck/plugins/apps/tsTSProcessor.cpp#L84). The initial load from the input plugin is performed [around the middle](https://github.com/tsduck/tsduck/blob/1700f5f49e0f7e1d0638c55147bccb374fdcb658/src/libtsduck/plugins/apps/tsTSProcessor.cpp#L188), after all plugins are _initialized_, but before _starting_ them. The control server is started at the very end, when all plugin threads are in a stable state.

sathish25071992, 2023-12-20:
Thank you for the quick response @lelegard . Will go through the code further on understanding the flow. Actually, we are planning to use the TSDuck library for the customized application. So as per the code, if we write our custom sub-class derived from TSProcessor and create a custom start method where we start the server as very first thing then we start plugin should be able to handle this condition right?.

lelegard, 2023-12-20:
> So as per the code, if we write our custom sub-class derived from TSProcessor and create a custom start method where we start the server as very first thing then we start plugin should be able to handle this condition right

Not really. This is not as simple as that. The interactions between the plugins, their executor thread and the global buffer are rather complex. I do not recommend changing the order of things or you would risk diving into an endless debugging cycle. Moreover, if reading from the Dektec ASI board blocks when there is no stream, restarting the input won't do anything since the previous input operation still blocks the input thread (note that in the current scheme, all initializations are done even before the plugin threads are started).

The Dektec input and output plugin were designed to interface the generic Dektec API. However, distinct types of devices have distinct behaviors : ASI, modulators, demodulators, Ethernet, etc.

If your really want to address specific behaviors of the ASI boards, you may need to directly interface DTAPI, the Dektec-API.

Alternatively, instead of using tsp, you may have a look at tsswitch. This application takes several input plugins. You may start it using multiple Dektec plugins, one per port, and then switch between them. Note that I am not 100% sure that accessing the same Dektec boards at the same time works. You need to test.

sathish25071992, 2023-12-20:
Ok got it. Let me check the tsswitch option.

Also as I check the source there is a timeout based dektec read which gets executed if the user provides `--receive-timeout` option. Using that makes the entire tsp process to exit if it timesout. Since we use custom application that invokes the tsProcessor::start, I think if it exits with error I can intimate the user and wait for proper channel input from user or retry.

sathish25071992, 2023-12-21:
Closing this issue as the `--receive-timeout` was able to get us going. Thank you for your support @lelegard

closed 2023-12-21 by sathish25071992