tsduck/tsduck #1398: [BUG] tsduck hangs and uses 100% on broken TS packet

svlobanov, 2023-12-30:
#### Bug description:
tsduck hangs and uses 100% CPU when trying to extract PID using "filter" plugin 

#### How to reproduce:
Run `tsp -I file 482.ts -P filter -p 0x58 -O file 0x58.ts` 
or `tsp -I file 482.ts -P filter -p 0x58 --valid -O file 0x58.ts`

TS sample:[482.zip](https://github.com/tsduck/tsduck/files/13799248/482.zip)

#### Expected behavior:
tsduck extracts pid 0x58, tsduck doesn't hang

#### Errors and logs:
~~~
* Error: file: synchronization lost after 185 packets, got 0xFF instead of 0x47
~~~

#### Environment:
 - OS: macOS
 - OS version: 14.1
 - TSDuck full version: 3.36-3528
 - Installation type: homebrew

#### Additional information:
TS packets 186-190 are broken (sync byte is not 0x47). you can use wireshak filter: `mp2t.sync_byte != 0x47`

Merry Christmas and Happy New Year!

lelegard, 2023-12-31:
> Merry Christmas and Happy New Year!

A nice bug as Christmas present! Thank you @svlobanov 😄 

OK, this is a bug. It can be seen without any filter (`tsp -I file 482.ts -O drop`). I just fixed it (a simple one-liner). You may pull the repo and rebuild.

This kind of corruption used to be properly handled. I don't know when that regression happened. I have added the first 300 TS packets of your file in the test suite as `test-142` to detect any further regression.

svlobanov, 2023-12-31:
Thanks @lelegard! Now it works as expected (with --valid option for 'filter' plugin)

closed 2023-12-31 by svlobanov