tsduck/tsduck #1400: [BUG] Teletext plugin crashes on broken packets

svlobanov, 2024-01-01:
#### Bug description:
Teletext plugin crashes (segfault) on broken packets

#### How to reproduce:
`tsp -I file out-fuzzing-seed1-change1_2.ts -P teletext -O drop`

TS sample: [out-fuzzing-seed1-change1_2.zip](https://github.com/tsduck/tsduck/files/13806344/out-fuzzing-seed1-change1_2.zip)

#### Expected behavior:
tsp doesn't crash

#### Errors and logs:
~~~
% tsp -I file out-fuzzing-seed1-change1_2.ts -P teletext -O drop
zsh: segmentation fault tsp -I file out-fuzzing-seed1-change1_2.ts -P teletext -O drop

% lldb -- tsp -I file out-fuzzing-seed1-change1_2.ts -P teletext -O drop
(lldb) target create "tsp"
Current executable set to '/Users/sergey/atsc/tsduck/bin/release-arm-Sergeys-MacBook-Air/tsp' (arm64).
(lldb) settings set -- target.run-args  "-I" "file" "out-fuzzing-seed1-change1_2.ts" "-P" "teletext" "-O" "drop"
(lldb) run
Process 4986 launched: '/Users/sergey/atsc/tsduck/bin/release-arm-Sergeys-MacBook-Air/tsp' (arm64)
Process 4986 stopped
* thread #4, name = 'teletext', stop reason = EXC_BAD_ACCESS (code=1, address=0x158000000)
    frame #0: 0x000000010107c464 libtsduck.dylib`ts::TeletextDemux::handlePESPacket(ts::PESPacket const&) + 432
libtsduck.dylib`ts::TeletextDemux::handlePESPacket:
->  0x10107c464 <+432>: ldrb   w24, [x9, #0x1]
    0x10107c468 <+436>: sub    x26, x8, #0x2
    0x10107c46c <+440>: add    x25, x9, #0x2
    0x10107c470 <+444>: and    w8, w3, #0xfe
Target 0: (tsp) stopped.
(lldb) bt
* thread #4, name = 'teletext', stop reason = EXC_BAD_ACCESS (code=1, address=0x158000000)
  * frame #0: 0x000000010107c464 libtsduck.dylib`ts::TeletextDemux::handlePESPacket(ts::PESPacket const&) + 432
    frame #1: 0x0000000100f15fc0 libtsduck.dylib`ts::PESDemux::processPESPacket(unsigned short, ts::PESDemux::PIDContext&) + 292
    frame #2: 0x0000000100f15e10 libtsduck.dylib`ts::PESDemux::processPacket(ts::TSPacket const&) + 776
    frame #3: 0x0000000100f15ad4 libtsduck.dylib`ts::PESDemux::feedPacket(ts::TSPacket const&) + 92
    frame #4: 0x0000000101081704 libtsduck.dylib`ts::TeletextPlugin::processPacket(ts::TSPacket&, ts::TSPacketMetadata&) + 80
    frame #5: 0x0000000101129248 libtsduck.dylib`ts::tsp::ProcessorExecutor::processIndividualPackets() + 628
    frame #6: 0x0000000101128e84 libtsduck.dylib`ts::tsp::ProcessorExecutor::main() + 568
    frame #7: 0x0000000101089848 libtsduck.dylib`ts::Thread::mainWrapper() + 536
    frame #8: 0x00000001010895dc libtsduck.dylib`ts::Thread::ThreadProc(void*) + 40
    frame #9: 0x000000018b9a5034 libsystem_pthread.dylib`_pthread_start + 136
(lldb) quit
~~~

#### Environment:
 - OS: macOS
 - OS version: 14.1
 - TSDuck full version: 3.37-3563
 - Installation type: rebuilt binary using git master code 

#### Additional information:
This issue was found by automatic fuzzing test, I'll send a proposal and source code for fuzzing test a bit later

lelegard, 2024-01-02:
Hi @svlobanov 

Thanks for reporting this. I have fixed this (it was a length check issue).

I am interested in your fuzzing tests. I have seen that you also reported #1401. Fuzzing tests are extremely precious to find weird bugs which can be eventually turned into security issues through code injection (possibly this one). How do you proceed? Do you use any standard fuzzing tool? In the past, I was interested in AFL but I didn't use it much.

svlobanov, 2024-01-02:
@lelegard thank you for the fix! I've started a new thread with my proposal regarding fuzzing tests https://github.com/tsduck/tsduck/issues/1402

closed 2024-01-03 by svlobanov