tsduck/tsduck #1402: [FEAT] fuzzing tests

svlobanov, 2024-01-02:
#### Is your feature request related to a problem?
tsduck might crash in case of non-expected input data

#### Proposed solution:
make fuzzing test to emulate broken data/garbage

#### Alternatives:
use static analyzers (might help in some cases, but not for all)

#### Additional information:
I've developed very simple PoC quality fuzzing tests, it's here https://github.com/svlobanov/tsduck-fuzzing , to run the tests, it is required to put some realistic ts dump of SPTS/MPTS normal DVB/ATSC signal (50-500MB+ is the best choice) with the name "in.ts" and the run the tests via `make test`

Points to discuss:
1. It would be nice to implement fuzzing tests as a part of tsduck or tsuck-tests repo.
2. Fuzzing might be an option for normal tests of tsduck-tests (use the same dumps (after fuzzing) and the same tsduck commands, but check only for segfaults, aborts, etc and timeouts). Right now I'm working on this
3. Fuzzing tests require a lot of computation time (the more the better) so it can't be a part of free github actions CI (or maybe very limited - e.g. use just one seed value)
4. I would like to use GNU "timeout" tool to catch infinite loops like #1398, but I'm not sure if it's OK to use non-posix tools. Please suggest
5. I have a prototype for fuzzing ts files written in bash, it works but it is extremely slow due to bash limitations so it is not realistic to write everything in bash like it's done in tsduck-tests repo. I don't want to write fuzzing tool in python, perl and similar languages because it will cause compatibility headache. As I can see you support a lot of OS so I think pure C99 + bash is a good choice
6. Currently I don't use any framework for tsduck fuzzing so I'm not sure if you like the idea of implementation from scratch

lelegard, 2024-01-02:
Hi @svlobanov 

Thank you for this excellent idea. Let me share some thoughts and then let's see how we can proceed.

Doing fuzzing tests means two things: 1) generating corrupted input data, 2) running tests and handling errors. This is roughly what you already do in your tsduck-fuzzing project with 1) fuzzing.c and 2) fuzzing.sh. Let's see the two steps separately.

### Step 1: generating corrupted TS files

Generating corrupted TS files, based on selected criteria, can be useful to anyone. It can be used to stress the resistance of TSDuck but it can also be useful to test a set-top box, a TV set, any intermediate reception equipment or media tools.

So, it could be a good idea to add in TSDuck a `tsfuzz` tool to corrupt an existing TS file, just like your fuzzing.c. We need to specify a list of features / options for that tool. I am not sure to fully understand all command line parameters of your fuzzing.c program.

The `tsfuzz` tool should be developed in C++17, just like the rest of TSDuck. If you agree, I can provide a skeleton of tool with basic features (command line options, reading / modifying a TS file). Then, you could update it with new ideas, new options, new TS corruptions, using pull requests.

### Step 2: running fuzzing tests

This part is about TSDuck testing and would use the `tsfuzz` tool. As you mention, these tests should be part of the tsduck-test repo but shall not be run on any push or PR.

First, because they would use too many resources, as you pointed out.

But, second, the automated test suite only contains reproducible tests, tests which do the same thing, all the time, on all platforms. By design, fuzzing tests are random and not reproducible.

In the tsduck-test repo, there is an `extra` directory which contains tests which are not part of the automated test suite, mainly because they need some form of manual operation or specific hardware. I think that a `fuzzing` subdirectory could be placed there.

The fuzzing tests should be run from time to time, in "fuzzing campaigns", in the background of some powerful system (typically an Arm64 Mac).

For a start, a simple script such as your fuzzing.sh can be sufficient, based on the `tsfuzz` tool.

Question for your next sleepless night: how to fuzz `tsfuzz` ? 😉 

### The tsfuzz specifications

Let's start with this and please provide your ideas.

Syntax (based on `tsfclean`): `tsfuzz [options] filename ...`

Options (also based on `tsfclean`):

`-o path`, `--output path`: Output file or directory. If more than one input file is specified, the output name shall specify a directory.

`-p pid`, `--pid pid`: Only corrupt the specified PID's (several options allowed).

Please formalize in terms of command options your ideas from your fuzzing.c (and other ideas).

svlobanov, 2024-01-02:
>So, it could be a good idea to add in TSDuck a tsfuzz tool to corrupt an existing TS file, just like your fuzzing.

I'm 100% agree.

>We need to specify a list of features / options for that tool

I think we can start from two easy things: corrupt random bytes and drop random TS packets like it's done in tcprewrite (https://tcpreplay.appneta.com/wiki/tcprewrite-man.html , see fuzz-factor and fuzz-seed). Frankly, I don't like tcprewrite fuzzing because there is no possibility to split drop and corrupt features so I have my own tools for pcap drops/corrupting, but in general it is likely the same as tcprewrite.
I have some more idea like: changing random bytes in the tables and recalculate CRC32 mpeg value to pass crc32 filters

>If you agree, I can provide a skeleton of tool with basic features (command line options, reading / modifying a TS file)

For sure, I agree

> Please formalize in terms of command options your ideas from your fuzzing.c (and other ideas)

I can explain fuzzing.c args:

INPUT_FILE ($1) - no need to describe

OUTPUT_FILE ($2) - no need to describe

SEED_VALUE ($3) - very important option for reproducibility. Also, it is highly important to use OS- and CPU-independent random generator like I use in fuzzing.c. Otherwise, it will be required to save produced .ts/pcap/etc file every time when running script founds an issue.

CHANGE_PROB_X ($4) and CHANGE_PROB_Y ($5) - these two options defines probability of byte corruption. the probability of corruption is CHANGE_PROB_X/CHANGE_PROB_Y . You can use just one float value, I just prefer to avoid floating point calculations.

IGNORE_SYNC_BYTE ($6) - if it equal to 1 then sync_byte (every 188 byte) is ignored and not touched. this options is needed to pass sync_byte validations

lelegard, 2024-01-05:
An initial implementation is available. You may [propose new features through pull requests](https://tsduck.io/doxy/contributing.html).

To remain consistent with the rest of TSDuck, the fuzzing features are available through C++ classes in the TSDuck library and can be used from any application. A tool named `tsfuzz` and a plugin named `fuzz` are available and are simple wrappers to these C++ classes. They are documented in the user's guide (which then may need to be updated when new features are added).

The fuzzing is implemented in two separate classes.

The class [`TSFuzzingArgs`](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/dtv/transport/tsTSFuzzingArgs.h) defines the fuzzing options. They can be set in a dedicated application. The class may also define and decode them as command line options.

These command line options are used in `tsfuzz` and plugin `fuzz`. This means that the source code of `tsfuzz` and plugin `fuzz` is final and do not need to be changed. If you want to add new options, simply update the class `TSFuzzingArgs`.

The fuzzing itself is done packet per packet in the class [`TSFuzzing`](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/dtv/transport/tsTSFuzzing.h).

Currently, the features of `TSFuzzing` and `TSFuzzingArgs` are exactly the same as in your code. For this reason, I added your name in the copyright of these two classes.

The only difference with your code is the PRNG. I used a more cryptographically secure PRNG (derived from Fortuna without external entropy pools). It is still strictly deterministic based on the initial seed. It is probably much slower though.

If you look at the [source code of the `fuzz` plugin](https://github.com/tsduck/tsduck/blob/master/src/tsplugins/tsplugin_fuzz.cpp), you can see the advantage this structure. You can implement fuzzing anywhere at virtually zero development cost. The same principles are used in TSDuck with utilities `tsanalyze`, `tstables`, `tspsi` and the corresponding plugins `analyze`, `tables`, `psi`.

lelegard, 2024-01-05:
OK, we do not need cryptographic security to fuzz. I changed the PRNG to xoshiro256** and the result is much faster of course.

Concerning the differences between Arm64 and Intel (not macOS vs. Linux), the implementation of SHA-256 and AES (used by the crypto PRNG) are accelerated using specialized instructions on Arm64, not on Intel. In practice, the insertion of crypto algos in TSDuck dates back to the very beginning (circa 2005). The code was grabbed from LibTomCrypt, a fast (for the time) library with a very liberal license. The code is pure C/C++. Recently, I wrote an accelerated version for Arm64 because I work on security on Arm64 CPU's. This is why TSDuck crypto is much faster on Arm64 than Intel.

lelegard, 2024-01-06:
Let's close this one, if you agree. The new feature (tsfuzz) is in place. Please submit a specific PR or issue for new features inside the `TSFuzzing` class.

svlobanov, 2024-01-06:
Yes, let's close this one.

closed 2024-01-06 by svlobanov