tsduck/tsduck #1404: [Q] How can I get splicemonitor to decode more than just the first cue?

thep0et, 2024-01-02:
I have an issue where, when subscribing to an active multicast, only the first SCTE 35 cue gets decoded. 
Here is an example:

```* splice PID 0x0100 (256), event 0x00000003 (3) out, occurrence #1
* SCTE 35 Splice Information, TID 252 (0xFC), PID 256 (0x0100)
  Short section, total size: 57 bytes
  - Section 0:
    Protocol version: 0
    Encryption: none
    PTS adjustment: 0x000000000
    CW index: 0x00 (0), tier: 0xFFF (4095)
    Command type: 0x06 (TimeSignal), size: 5 bytes
    Time: 0x0F04CD0E8 (4,031,566,056, 44,795,178 ms)
    - Descriptor 0: SCTE 35 Segmentation (0x02, 2), 30 bytes
      Identifier: 0x43554549 ("CUEI")
      Segmentation event id: 0x00000003, cancel: 0
      Program segmentation: 1, has duration: 1, not restricted: 1
      Segment duration: 10800000
      Segmentation upid type: 0x08 (TI)
      Upid data (8 bytes): 00 00 00 00 02 A6 2B 1C
      Segmentation type id: 0x34 (Provider Placement Opportunity Start)
      Segment number: 1, expected segments: 1
      Sub-segment number: 1, expected sub-segments: 1
    CRC32: 0x3ED8C883 (OK)
GA▒06▒▒▒▒L▒▒ CUE▒▒+4>▒ȃ▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒PuTTYGA▒06▒▒▒􈥯 CUE▒▒U4>▒$▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒PuTTY```

This issue seems unique to this particular provider (Adcuratio), where the first cue comes through and everything after doesn't seem to be recognized as a SCTE-35 message.

lelegard, 2024-01-03:
Hi @thep0et 

Please provide the complete tsp command and, if possible, a partial TS file which reproduces the problem.

I am worried by the garbage which is displayed after the CR.

thep0et, 2024-01-03:
`tsp -I ip {source}@{dest}:{port} -P filter -p 256 -P splicemonitor -a -s 256`

This is the command that produced the above result.  I've tried various versions of this command with no luck.  It's always the first cue and then garbage after.  My current working theory is that, somehow, tsp isn't recognizing the package headers as SCTE-35 on subsequent ques, but I can't figure out why.

TS file containing a cue:
https://drive.google.com/file/d/1FLuJs1x38F-UQPfVA3sVoa8n4KIhb1xA/view?usp=sharing

lelegard, 2024-01-03:
tsp is a transport stream processor, it reads an input and writes to an output, both being defaulted to the standard input and output (to be used in pipes).

You do not specify any output plugin. Therefore, tsp write the _binary transport packets_ to the standard output. The garbage you see on screen is the binary transport stream.

If you do not need the output TS, use `-O drop`, an output plugin which discards the TS.

Your stream has only one section. If your actual stream contains several SCTE-35 sections, `-P splicemonitor -a` will display them all.

thep0et, 2024-01-03:
Apologies.  I probably should have specified that this is just an excerpt of the full command that I'm using.  
The full command already uses -O drop to drop the binary ts garbage, outputting as JSON before running it through some other text processing to clean it up before logging.

My issue is that, if I use -O drop, it drops every cue after the first one.  I removed the output so that I could see everything.
In the above capture, each time the word "cue" appears, I should have received another SCTE-35 cue, but it's coming across as more garbage.  The data is the same as if I didn't specify a PID

thep0et, 2024-01-03:
I should also note, that this is not an in-band stream that I'm monitoring.  
These cues are being sent out-of-band for merging.  I'm not sure if that would make a difference.

lelegard, 2024-01-03:
> My issue is that, if I use -O drop, it drops every cue after the first one. I removed the output so that I could see everything.

There is a misunderstanding on how tsp works. This is a chain of plugins. The intermediate plugins such as splicemonitor work exactly the same way regardless of the output plugin downstream. Conversely, the drop output plugin discards all packets at the end of the chain. In _no circumstances_, the drop output plugin can let one cue pass and drops the others. This is not the way tsp works.

Please provide a full TS file with several cues but for which you see only one.

thep0et, 2024-01-03:
I will have to capture one which may take some time.
In the meantime, I may have hit upon the issue.  The provider is not incrementing the Continuity Counter in the transport header.  
Could this be causing an issue where tsp and/or the plugin is bundling the packets together?

lelegard, 2024-01-03:
> The provider is not incrementing the Continuity Counter in the transport header.
Could this be causing an issue where tsp and/or the plugin is bundling the packets together?

Yes, this is definitely a bug from the provider, a huge bug. This is basic MPEG transport. Their system is simply broken and cannot work with MPEG-compliant receivers.

When the CC is not incremented in a PID, the ISO 13818-1 spec says that the packet is a duplicate of the previous one from the same PID. This is detected by the TSDuck demux and duplicated packets are ignored. Thus, indeed, only the first TS packet is analyzed. All other TS packets on the same PID are considered as duplicated and ignored.

thep0et, 2024-01-03:
This is exactly the smoking gun I've been looking for for weeks.  I really appreciate the feedback here, it definitely helped and gave me a deeper understanding of how tsp works.
Hopefully I can convince the provider to update their specs.

closed 2024-01-03 by thep0et

lelegard, 2024-01-03:
> Hopefully I can convince the provider to update their specs.

Well, the spec is ISO 13818-1 / ITU H.222, this is the "MPEG Bible", the root of all Digital TV standards. If their product is not compliant with this, this is a bug, a huge bug and there is no one to "convince". They **must** fix it, full stop. Do not hesitate to direct them to this issue.