tsduck/tsduck #1406: [Q] Is there a way to reduce source bitrate before merge when remapping?

assallay, 2024-01-03:
Hi! @lelegard !.
I am new here.... and was working around with tsduck and i found that when remapping the channels from source using MERGE PLUGIN it doesn't EFFECT the bitrate. IS there a way to reduce source bitrate or that should be done with other settings.

or if there is any wrong in the code.. what i am doing wrong!!.

"tsp -t -b 70960784 -I null \ 
-P regulate -b 70960784 \ 
-P inject "$SDT" -b 5000 -s -p 17 \ 
-P inject "$PAT" -b 5000 -p 0 \ 
-P inject "$channel_1" -b 2500 -s -p 1010 \ 
           -P merge -t --no-pcr-restamp "dvb1 -I dvb -a 0 --frequency 12283000000 --symbol-rate 27500000 --fec-inner auto --spectral- 
           inversion auto --polarity vertical --delivery-system DVB-S --modulation QPSK --pilots off --roll-off 0.35 --lnb 9750,10600,11700 -- 
          satellite-number 0 --receive-timeout 0 --signal-timeout 0 \
          -P analyze -i 10 -o /home/max/sat/lnb1.txt \ 
          -P remap -u 601=1101 810=1102 602=1111 820=1112 604=1121 840=1122 \ 
-P filter -p 1101 -p 1102 -p 1111 -p 1112 -p 1121 -p 1122 " \
-P regulate -b 70960784 \ 
-P pcradjust -b 70960784 \
-P analyze -i 10 -o /var/log/output.txt \ 
-O dektec --channel 0 --device 0 -r 7/8 -f 1150000000 -l -10 -s --symbol-rate 44000000 -m DVB-S-QPSK"

THANKS !!!!

lelegard, 2024-01-04:
Merging replaces null packets with incoming packets and does not change the bitrate.

Additional observations: One regulate plugin is enough. Once the processing speed is aligned on the playout speed, there is no need to do it twice. At best it does nothing. At worst it introduces glitches.

assallay, 2024-01-04:
So reducing bitrate for incoming source is not acheivable!!!
Thanks..

closed 2024-02-08 by assallay