tsduck/tsduck #1407: [BUG] tsduck hangs in case of std::bad_alloc in 'pes' plugin

svlobanov, 2024-01-03:
#### Bug description:
tsduck hangs in case of std::bad_alloc in 'pes' plugin 

#### How to reproduce:
1. Use Ubuntu 20.04 x86_64
2. Compile tsduck using clang-12 and libstdc++ from gcc-13 (http://ppa.launchpad.net/ubuntu-toolchain-r/test/ubuntu) (libstdc++ packages: `libstdc++6:amd64 13.1.0-8ubuntu1~20.04.2`  and `libstdc++-13-dev:amd64 13.1.0-8ubuntu1~20.04.2`). Command for building: `make -j32 NOTEST=1 NODEKTEC=1 NOHIDES=1 NOVATEK=1 NOPCSC=1 NOPCSTD=1 NOJAVA=1 NOGITHUB=1 CC=clang-12 CXX=clang++-12 USE_LLVM=1 `
Most probably, the issue will be reproduced on actual Ubuntu, in this case it is not required to play with compiler and libstdc++
3. Use bash shell
4. Run `ulimit -m 4000000 -v 16000000`
5. Run `tsp -d2 -I file mem_18125_300_100000_xac.ts -P pes --avc-access-unit --audio-attributes --intra-image --sei-avc --video-attributes -O drop`
6. After some time tsp hangs due to std::bad_alloc exception in pes plugin thread

TS sample: [mem_18125_300_100000_xac.zip](https://github.com/tsduck/tsduck/files/13824342/mem_18125_300_100000_xac.zip)

#### Expected behavior:
I think std::bad_alloc exception means that pes plugin uses too much memory (like in #1405) so it will be great to investigate why so much memory is required (it is required about 20GB RAM to process attached 10Mb file)
Regardless the previous idea, from my point of view it is not normal that tsp process hangs in case of exception in one plugin thread. Looks like it should be fixed, e.g. tsp process should exit with non-zero exit code and some error message if one exception occurs in plugin's thread.

#### Errors and logs:
~~~
# tsp -d2 -I file mem_18125_300_100000_xac.ts -P pes --avc-access-unit --audio-attributes --intra-image --sei-avc --video-attributes -O drop
... after some time: ...
*** Internal error, thread aborted: std::bad_alloc
... after CTRL+C: ...
^C* tsp: user interrupt, terminating...
* Debug: aborting all plugins...
* Debug: aborting plugin file
* Debug: aborting plugin pes
* Debug: aborting plugin drop
* Debug: drop: stopping the output plugin
* Debug: drop: output thread aborted after 10,000 packets (10,000 output)
* Debug: Report logging thread terminated
~~~

#### Environment:
 - OS:  Linux
 - OS version: Ubuntu 20.04
 - TSDuck full version: 3.37-3571
 - Installation type: rebuilt binary using git master code (using clang-12 and libstdc++ from gcc-13)

#### Additional information:
This issue was found by automatic fuzzing test.
I use ulimit memory limits to catch memory issues and avoid unpredictable OOM killer which can stop random processes. ulimit -m 4GB -v 16GB is enough for allmost all cases, but this particular ts sample requires much more
P.S. Apple removed -m and -v ulimit options some time ago so it's not easy to reproduce this issue on mac (it is required to play with launchd)

lelegard, 2024-01-04:
Hi @svlobanov 

This is a similar issue as #1405, with a similar fix (now done).

The HEVC data structures contain several "lists of data". When the HEVC stream is corrupted, the read or computed size of the list can become extremely large, leading to memory allocation failures. I have tracked all these allocations and checked the sizes against valid values. I hope I have not missed any.

svlobanov, 2024-01-04:
@lelegard, thanks for the fix, but what do you think about inter-thread communication issue? Looks like, if any unhandled exception happens in plugin thread, it will be the same issue - tsp will hang instead of "crash"/exit with non-zero code. From my experience, exit/crash is much better than hang, because usually it will be auto-restarted by systemd, monit, etc.

lelegard, 2024-01-04:
Yes, I will look into it. However, not all threads are equal. In the case of a plugin thread, it is clearly better to exit the application because the loss of a plugin thread breaks the chain and the application hangs because the packets no longer flow through the dead plugin. However, ancillary plugins such as logging or system monitoring are not vital to the operations and crashing the application is not a good idea. I need to add some new "exit application on exception" property in the thread class and carefully apply it to selected threads only.

lelegard, 2024-01-05:
The exit on plugin thread exception has been implemented.

However, not all exceptions can be handled. For instance, a memory allocation failure such as the one in this issue is not handled as a `bad_alloc` exception, as it should be. On Linux, the application hangs for a while without eating CPU and is then killed by the OS. On macOS, the application also hangs without eating CPU, possibly forever.

Any regular exception in a plugin thread, on the other hand, terminates the application.