tsduck/tsduck #1409: [BUG] invalid start of video PES payload on VVC video

svlobanov, 2024-01-05:
#### Bug description:
pes plugin reports false positive warning on VVC video: "WARNING: PID 0x1011, invalid start of video PES payload: 00 00 00 01 00 79 00 AD"

#### How to reproduce:
`tsp  -I file vvc-720p.ts -P pes --avc-access-unit --sei-avc -h -v -i --packet-index  -O drop`

TS sample: [vvc-720p.zip](https://github.com/tsduck/tsduck/files/13847311/vvc-720p.zip)

#### Expected behavior:
Pes plugin doesn't report 'invalid start of video PES payload' warnings

#### Errors and logs:
~~~
* PID 0x1011, TS packets 4-5, stream_id 0xE1 (Video 1), size: 348 bytes (header: 19, payload: 329)
  PES header:
    0000:  00 00 01 E1 01 56 84 C0 0A 31 0C DF F3 01 11 0C
    0010:  DF 66 61
WARNING: PID 0x1011, invalid start of video PES payload: 00 00 00 01 00 79 00 AD
* PID 0x1011, TS packets 4-5, VVC H.266 access unit type 0x0F (Sequence parameter set)
  Offset in PES payload: 4, size: 156 bytes
  VVC H.266 access unit:
    0000:  00 79 00 AD 02 33 80 00 00 80 0A 02 00 B4 46 A0
    0010:  07 37 44 6E 88 46 99 49 93 38 4D 95 8C 10 20 9E
    0020:  00 CA 0C 58 42 18 31 10 81 92 20 D4 87 A3 D5 A9
~~~

#### Environment:
 - OS: macOS
 - OS version: 14.1
 - TSDuck full version: 3.37-3578
 - Installation type: rebuilt binary from git master

#### Additional information:
Looks like it is required to add `&& !pkt.isVVC()` here: https://github.com/tsduck/tsduck/blob/392bc19e1d33b153e53b18a46416d8cc5e3025f5/src/tsplugins/tsplugin_pes.cpp#L570-L576 (and maybe some more modern video codecs)

To play this TS you can use ffmpeg with fraunhofer vvc decoder: https://github.com/fraunhoferhhi/vvenc/wiki/FFmpeg-Integration (ffmpeg 6.1 + libvvdec) or elmedia player (macos only). ffplay works fine, elmedia has playback issues
```
% ./ffmpeg -codecs | grep vvc
ffmpeg version n6.1.1 Copyright (c) 2000-2023 the FFmpeg developers
...
 DEV.L. vvc                  H.266 / VVC (Versatile Video Coding) (decoders: libvvdec) (encoders: libvvenc)
```

lelegard, 2024-01-06:
Hi @svlobanov 

Thank you for pointing to the source of the root cause. I fixed it.

svlobanov, 2024-01-07:
Thanks for the fix

closed 2024-01-07 by svlobanov