tsduck/tsduck #1411: pcredit Plugin Miscalculation with PCR Subtraction below 0 [BUG]

mattgriffindevelopment, 2024-01-09:
#### Bug description:
Hi there,

I believe to have identified an issue with TSDuck Version 3.36-3528 (and previous versions) regarding the overflow calculations of the PCR value.

I have noticed that when using subtraction with the pcredit --add-pcr function, if it drops below 0, it incorrectly wraps around to pcr_base value 7101012595, instead of 8589934591.

When using addition, this isn't a problem and adding 8589934591 to a ts file with pcr_base value 0 can reach the max 33-bit value as expected. And, adding any further value to this wraps back around to 0 correctly. But any subtractions going below 0 appear to be calcuted incorrectly.

To note, I am using "-u pts", but have also tested in multiple ways using "-u pcr" and "-u milliseconds" with the relevant unit conversions (aka multiplying by 300 when using "-u pcr" for example)

As a final note, it appears that the calculations are perfectly correct for PTS values with --add-pts and I imagine likewise with --add-dts (haven't tested dts). PTS, when subtracting 1 with -u pts, does change to 8589934591 pts as expected.

#### How to reproduce:
My testing is as follows:

I have a transport stream file with exactly 0 pcr_base, 0 pcr_ext and 0 pts.

Running:
	tsp -I file input.ts -P pcredit --pid 201 --add-pcr -1 -u pts -O file output.ts

Produces a transport stream file with pcr_base value of 7101012595, when I would expect 8589934591. For some reason it also sets the pcr_ext to 16 which I would expect to remain at 0 as the units are set to PTS (33-bit).

Now if I run the following on the original ts file with 0 pcr_base/pcr_ext:
	tsp -I file input.ts -P pcredit --pid 201 --add-pcr 8589934591 -u pts -O file output.ts
	
It correctly has the pcr_base set to 8589934591 and the pcr_extension at 0.

#### Expected behavior:
pcr_base value (when subtracted below 0) should wrap around to 2^33 (8589934591) instead of 7101012595.

#### Errors and logs:
No Errors or Logs for this issue.
~~~
[be sure to surround verbatim log lines with "~~~" lines]
~~~

#### Environment:
 - OS: 
 - OS version: 
 - TSDuck full version: 
 - Installation type:

lelegard, 2024-01-09:
Hi @mattgriffindevelopment 

Thanks for reporting this.

This can be reproduced without input file as follow:
~~~
tsp -I null 1\
    -P craft --pid 100 --pcr 0 --no-payload \
    -P dump --headers-only \
    -P pcredit --add-pcr -1 -u pts \
    -P dump --headers-only \
    -O drop
~~~
Which gives this output:
~~~
* Packet 0
  ---- TS Header ----
  PID: 100 (0x0064), header size: 188, sync: 0x47
  Error: 0, unit start: 0, priority: 0
  Scrambling: 0, continuity counter: 0
  Adaptation field: yes (184 bytes), payload: no (0 bytes)
  Discontinuity: 0, random access: 0, ES priority: 0
  PCR: 0x00000000000

* Packet 0
  ---- TS Header ----
  PID: 100 (0x0064), header size: 188, sync: 0x47
  Error: 0, unit start: 0, priority: 0
  Scrambling: 0, continuity counter: 0
  Adaptation field: yes (184 bytes), payload: no (0 bytes)
  Discontinuity: 0, random access: 0, ES priority: 0
  PCR: 0x1EFFFFFFED4
~~~
The adjusted PCR `0x1EFFFFFFED4 ` is the one you see:
~~~
$ echo $(( 0x1EFFFFFFED4 ))
2130303778516
$ echo $(( 0x1EFFFFFFED4 / 300 ))
7101012595
$ echo $(( 0x1EFFFFFFED4 % 300 ))
16
~~~
This seems to be an issue with the way C/C++ handles the modulo operator on negative numbers with unsigned modulus. The modulus of PCR values is `(1<<33)*300 = 2576980377600`.

After adjustment, the unbounded value of the PCR is -300.
- Using Python: -300 % 2576980377600 == 2576980377300
- Using C/C++ (signed modulus): -300 % 2576980377600 == -300
- Using C/C++ (unsigned modulus): -300 % 2576980377600u == 2130303778516

I will fix this.

lelegard, 2024-01-10:
>  As a final note, it appears that the calculations are perfectly correct for PTS values with --add-pts and I imagine likewise with --add-dts (haven't tested dts). 

That is easy because PTS and DTS are pure 33-bit binary value. Because the modulus is a power of two, the result is simply binary-masked with a binary and `&` operator. The problem with PCR is that the modulus is not a power of 2 and a modulo `%` operator must be used instead of the binary and `&`.

lelegard, 2024-01-10:
And this has been fixed

mattgriffindevelopment, 2024-01-10:
Hi @lelegard ,
Thanks for the fast turn around. That's great to hear, thanks for the help. Will test when available,
Matt