tsduck/tsduck #1418: [Q] Are Big Endian platforms still supported?

svlobanov, 2024-01-14:
Hi @lelegard 

I've found that utest fails on s390x platform (which is big-endian). Do you still support big-endian platforms? Will you accept fixes for them?
I'm asking this question because BE platforms become less popular. Popular generic distros (rhel, debian, ubuntu) officially support only s390x.

Current issues:
~~~
*** JsonTest::testQuery, basic_string::substr: __pos (which is 3) > this->size() (which is 3)

*** JsonTest::testRunningDocument, basic_string::substr: __pos (which is 3) > this->size() (which is 3)

*** Xoshiro256ssTest::testRandom, assertion failure, utestXoshiro256ss.cpp, line 73
condition: expected == generated

*** FAILURES (3 tests FAILED, 656 passed, 1 assertions FAILED, 31357 passed)
~~~

debug log 
[utest-debug-s390x.txt](https://github.com/tsduck/tsduck/files/13931104/utest-debug-s390x.txt)
also attached

lelegard, 2024-01-14:
Hi @svlobanov 

It is a goal of TSDuck code to be endianness agnostic. This is why there are so many tweaked conversion functions in [tsMemory.h](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/base/types/tsMemory.h). The problem is that I never really had the opportunity to test TSDuck on big endian systems. It is surprisingly good news that you didn't find more issues.

The last times I used BE systems were Motorola 68k and Sun Sparc, decades ago, before I started to work on TSDuck.

What kind of s390x are you using? Qemu or a real system?

I think I understand what's the problem with Xoshiro. I will provide a fix. The JSON test failure is more surprising. There must be some hidden crappy side effect which needs some investigation.

Fixes are welcome. I won't be able to test them (I have no intention to build Qemu platforms). If they look fine to me and don't break the test suite on Intel and Arm, I will accept them.

lelegard, 2024-01-14:
I have pushed a proposed fix for the Xoshiro issue. Could you please test it on s390x?

I will have a look at potential root cause for the JSON issue, without guarantee.

svlobanov, 2024-01-14:
Thanks, Xoshiro256ss utest is passed in commit c9a9b19a020105f34701531b3bf23bd67cbbd8c7

>What kind of s390x are you using? Qemu or a real system?

I'm using qemu. Normal s390x VMs are available in IBM Cloud (with 120 days trial period)

lelegard, 2024-01-14:
What is the guest OS? Some Linux distro? Is this a VM running in IBM Cloud? Or a VM running on your own machine with an initial system image that you downloaded from IBM?

svlobanov, 2024-01-14:
Guest OS is Ubuntu 23.10 (https://cdimage.ubuntu.com/releases/23.10/release/ubuntu-23.10-live-server-s390x.iso ). I run s390x ubuntu using cpu-emulated qemu VM on x86_64 host. It should work on arm64 macos/linux too.
I used this manual: https://community.ibm.com/community/user/ibmz-and-linuxone/blogs/timothy-sipples1/2020/04/28/run-ubuntu-z-linuxone (with only two changes: remove ',msa5-base=off' and fixed /etc/fstab after installation (in recovery mode), '0.0002' changed to '0.0001'). Another way is docker over qemu: https://til.simonwillison.net/docker/emulate-s390x-with-qemu

lelegard, 2024-01-14:
I am confused with the error message in the two failed JSON tests. The message does not come from an explicitly failed assertion in the test suite. It seems to come from an exception in the C++ runtime.

And here the exception message is confusing:
~~~
basic_string::substr: __pos (which is 3) > this->size() (which is 3)
~~~

Does it complain that 3 is greater than 3 ????

svlobanov, 2024-01-14:
I will provide more info using gdb.

svlobanov, 2024-01-14:
>Does it complain that 3 is greater than 3 ????

_pos=3 means that it reads (or writes) 4th element, but size is 3

svlobanov, 2024-01-14:
(possibly) the same issue during test-001

~~~
> *** Aborting application
> timeout: the monitored command dumped core
> tests/../common/testrc.sh: line 481: 11607 Segmentation fault      timeout -s9 180s $(tspath tsp) --synchronous-log --bitrate-adjust-interval 10,000 "$@"

Core was generated by `tsp --synchronous-log --bitrate-adjust-interval 10,000 -I file /home/sergey/tsd'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x000003ff9d1b6590 in ts::UCharacteristics(char16_t) () from /lib/libtsduck.so
[Current thread is 1 (Thread 0x3ff9d6568c0 (LWP 11612))]
(gdb) bt full
#0  0x000003ff9d1b6590 in ts::UCharacteristics(char16_t) () from /lib/libtsduck.so
No symbol table info available.
#1  0x000003ff9d1dd3b4 in ts::UString::SuperCompare(char16_t const*, char16_t const*, unsigned int) () from /lib/libtsduck.so
No symbol table info available.
#2  0x000003ff9cf0f10e in ts::NamesFile::decodeDefinition(ts::UString const&, ts::NamesFile::ConfigSection*) () from /lib/libtsduck.so
No symbol table info available.
#3  0x000003ff9cf0fb18 in ts::NamesFile::loadFile(ts::UString const&) () from /lib/libtsduck.so
No symbol table info available.
#4  0x000003ff9cf105e0 in ts::NamesFile::NamesFile(ts::UString const&, bool) () from /lib/libtsduck.so
No symbol table info available.
#5  0x000003ff9cf10c78 in (anonymous namespace)::AllInstances::getFile(ts::UString const&, bool) () from /lib/libtsduck.so
No symbol table info available.
#6  0x000003ff9cf117ba in ts::NamesFile::Instance(ts::NamesFile::Predefined) () from /lib/libtsduck.so
No symbol table info available.
#7  0x000003ff9d0e1030 in ts::TSAnalyzerReport::reportPIDs(ts::Grid&, ts::UString const&) () from /lib/libtsduck.so
No symbol table info available.
#8  0x000003ff9d0e9702 in ts::TSAnalyzerReport::report(std::basic_ostream<char, std::char_traits<char> >&, ts::TSAnalyzerOptions&, ts::Report&) () from /lib/libtsduck.so
No symbol table info available.
#9  0x000003ff9d605a16 in ts::AnalyzePlugin::produceReport() () from /usr/lib/tsduck/tsplugin_analyze.so
No symbol table info available.
#10 0x000003ff9d605a86 in ts::AnalyzePlugin::stop() () from /usr/lib/tsduck/tsplugin_analyze.so
No symbol table info available.
#11 0x000003ff9d17f0b8 in ts::Thread::mainWrapper() () from /lib/libtsduck.so
No symbol table info available.
#12 0x000003ff9d17f4fa in ts::Thread::ThreadProc(void*) () from /lib/libtsduck.so
No symbol table info available.
#13 0x000003ff9c29c18a in start_thread (arg=0x3ff9d6568c0) at pthread_create.c:444
        ret = <optimized out>
        start = <optimized out>
        pd = 0x3ff9d6568c0
        out = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {{__gregs = {2, 4396392212672, 4393751543808, -4294967295, 4396392212672, 4393751544831,
                    4396372827408, 4397765457863, 45544236157678728, 45544236144273696}, __fpregs = {163520, 0, 2929314104496, 0, 167936, 0,
                    4396392212672, 4397250614959}}}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0,
              cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
#14 0x000003ff9c321550 in thread_start () at ../sysdeps/unix/sysv/linux/s390/s390-64/clone3.S:71
No locals.

~~~

lelegard, 2024-01-14:
Does it work when you use a simpler single-thread process, such as tstables or tsanalyze?

svlobanov, 2024-01-14:
Looks like, the issue is not related to single or multi thread difference:

~~~
$ tsp --synchronous-log  -I file input/test-001.ts -P tables -O drop 1>/dev/null
$ tsp --synchronous-log  -I file input/test-001.ts -P psi --all -O drop 1>/dev/null
$ tsp --synchronous-log  -I file input/test-001.ts -P analyze --json --deterministic -O drop 1>/dev/null
*** Internal error, thread aborted: basic_string::substr: __pos (which is 2) > this->size() (which is 2)
*** Aborting application
$ tsp --synchronous-log  -I file input/test-001.ts -P analyze  --deterministic -O drop 1>/dev/null
$ tsanalyze input/test-001.ts  1>/dev/null
$ tsanalyze input/test-001.ts --json --deterministic   1>/dev/null
Program aborted: basic_string::substr: __pos (which is 2) > this->size() (which is 2)
~~~

lelegard, 2024-01-14:
> $ tsanalyze input/test-001.ts --json --deterministic   1>/dev/null
Program aborted: basic_string::substr: __pos (which is 2) > this->size() (which is 2)

That should be easier to troubleshoot than a tsp session.

How did you install Ubuntu on the VM? I tried your link and the following command:
~~~
qemu-img create -f raw ubuntu-run.raw 30G
qemu-system-s390x \
    -machine s390-ccw-virtio \
    -cpu max,zpci=on \
    -serial telnet::4441,server \
    -display none \
    -m 4096 \
    --cdrom ubuntu-23.10-live-server-s390x.iso \
    -kernel kernel.ubuntu \
    -initrd initrd.ubuntu \
    -drive file=ubuntu-run.raw,format=raw \
    -nic user,hostfwd=tcp::2222-:22
~~~

On the console, the boot process runs up to some point where it completely stales. The console output ends here, or similar across various attempts:
~~~
[  OK  ] Listening on snap.lxd.daem…x for snap application lxd.daemon.
         Starting snap.lxd.activate…r snap application lxd.activate...
[  OK  ] Finished snap.lxd.activate…for snap application lxd.activate.
[  OK  ] Started snap.lxd.hook.conf…-476a-8269-8393dd10548f.scope.
[  OK  ] Started snap.subiquity.hoo…-4b5e-9c67-dd3e7f924a66.scope.
[  OK  ] Started snap.subiquity.sub…cation subiquity.subiquity-server.
[  OK  ] Started snap.subiquity.sub…ation subiquity.subiquity-service.
         Starting systemd-timedated…rvice - Time & Date Service...
[    **] (1 of 2) Job systemd-timedated.service/start running (4s / 1min 30s)
~~~

Specifically, the time at the end of the bottom line "Job ... running" stops increasing, indicating a possible hang in the boot process. 

The emulated system is not hung because the ssh server accepts connections but the usual user/passwd "ubuntu/ubuntu" or "ubuntu/" are rejected.

The qemu process uses 4 to 5% CPU time on the host (an Intel iMac).

svlobanov, 2024-01-14:
try this one:

```
qemu-system-s390x \
    -machine s390-ccw-virtio \
    -cpu max,zpci=on \
  -smp 8 \
  -nographic \
    -m 4096 \
    --cdrom ubuntu-23.10-live-server-s390x.iso \
    -kernel kernel.ubuntu \
    -initrd initrd.ubuntu \
    -drive file=ubuntu-run.raw,format=raw \
    -nic user,hostfwd=tcp::2222-:22
```

(most probably, `-smp 8` will the issue, I think I added this option during install phase)

svlobanov, 2024-01-14:
I checked on my mac arm64, the last command I provided works fine, installation process starts normally (TUI wizard)

lelegard, 2024-01-14:
It works for me also. I do not know if the difference is in the `-smp 8` or not using a telnet redirection for the console. Installation in progress. However, I am not sure to have the time to analyse the issue now. If you find the solution first, you win...

svlobanov, 2024-01-14:
You will need several hours to compile tsduck (if use just make -jX), but maybe you know some shortcuts how to minimize build time.

I will have some time next week to find the root cause if you don't fix it.

svlobanov, 2024-01-14:
This issue is getting more interesting. In debug-build (make DEBUG=1 -j16) there is no issue such issue:
~~~
$ bin/debug-s390x-ubuntu-s390x/tsanalyze --json --deterministic ../tsduck-test/input/test-001.ts 1>/dev/null
$ bin/release-s390x-ubuntu-s390x/tsanalyze --json --deterministic ../tsduck-test/input/test-001.ts 1>/dev/null
Program aborted: basic_string::substr: __pos (which is 2) > this->size() (which is 2)
~~~

utest also reports no errors in debug build. test-001 is also passed.
~~~
bin/debug-s390x-ubuntu-s390x/utest

OK (659 tests, 31405 assertions)
~~~

lelegard, 2024-01-14:
So this is probably not a plain endianness bug such as reading bytes in a non-portable order (that was the case with the xoshiro issue). 

On Windows, I have seen the opposite many times. The debug version contains much more assertions and we see failures like this one on debug builds only. 

I have seen this situation (error in release mode, none in debug mode) in two categories of situations:

- Very ugly, low level bug, using undefined behavior. Very hard to debug. 
- Incorrect code generation (compiler bug). 

Do we have a clang version on s390x? Just to compare the two. 

Do you have another big endian platform ? Sparc for instance. Just to test another big endian.

svlobanov, 2024-01-14:
Clang is available on Ubuntu 23.10 s390x.
I have mipsbe router openwrt-enabled, but it will be too hard to recompile due to openwrt toolchain, very limited flash and ram.

Will try to catch something using asan on s390x qemu host, then maybe clang. It's recompiling now.

svlobanov, 2024-01-15:
@lelegard , I've found that docker variant for running on s390x is much much faster, also it doesn't require long installation process.

It's very easy, just:
```
docker run --rm --privileged multiarch/qemu-user-static --reset -p yes
docker run -it s390x/ubuntu:rolling bash
```
(then install everything required using apt)

It doesn't emulate kernel (it uses host kernel). It uses qemu-s390x-static to run userspace applications and translate syscalls to host kernel. Of course it's not full emulation (e.g. you will see host /proc/cpuinfo and so on) and might have some new issues, but it works, at least I have the same failed test when run `utest` when I build and run tsduck in s390x docker.

Unfortunately it works only on x86_64 linux (because multiarch/qemu-user-static container is built only for x86_64), it should work on mac x86_64 with docker desktop, but I can't check because my intel mac is broken.

lelegard, 2024-01-16:
Hi @svlobanov 

I have seen in your PR #1420 that all tests pass when compiled with clang.

On my side, using your guidelines, I successfully built a s390x qemu system and built TSDuck in release mode using gcc. I reproduced the same failure as you. Using some traces, I see that the exception is thrown during the call to `ts::json::Object::query(const UString& path)`. A trace just before the call is displayed. The trace in the first line of the function is never displayed.

The problem occurs on one call to this function, not all. You saw that it didn't occur in non-optimized code (I didn't take the time to rebuild in debug mode). The problem also never appears on any other platform or with clang.

Passing a `const UString&` is just passing an address, something trivial. No manipulation of the string content should take place, as indicated in the failure message from the exception.

Therefore, we can safely say with a high degree of confidence that this is a gcc code generation bug on s390x. I have seen many compiler bugs, so I am not really surprised. However, the code generation bugs are the worst ones because you don't have any clue to start with. Considering that s390x is not the most popular open source platform, this is even less surprising.

On Arm64, I would have examined the generated code to find the actual bug. However, I don't know anything about the IBM s390x ISA. So, I just stay with the high-level observations.

I think we can close this one now.

svlobanov, 2024-01-16:
Yes, let's close this one. Anyway, I found something interesting using s390x platform https://github.com/tsduck/tsduck/issues/1421 that might be applicable to all platforms

closed 2024-01-16 by svlobanov

lelegard, 2024-01-17:
About the weird behavior on the s390x emulator, gcc is maybe not the only culprit. I have some doubt on the Qemu implementation. After successfully completed the build and test of TSDuck using clang, just as you did, I tried my [cryptobench](https://github.com/lelegard/cryptobench) project on s390x. It compiled (using gcc) but during the benchmark run, Qemu crashed with the following message:
~~~
ERROR:../target/s390x/tcg/fpu_helper.c:449:s390_swap_bfp_rounding_mode: code should not be reached
Bail out! ERROR:../target/s390x/tcg/fpu_helper.c:449:s390_swap_bfp_rounding_mode: code should not be reached
~~~
This is not an application crash in the VM, not a guest operating system crash either, this is a crash in the Qemu emulator.

More worrying, if I reboot the same s390x VM, Qemu crashes again with the same error before the end of the boot. All the time. My s390x VM has become completely unusable. It can no longer complete a boot.

So, I think it is safe to not trust Qemu too much on s390x.

svlobanov, 2024-01-17:
I had such issues on qemu-8.1, found them in their bug tracker, then switched to qemu-8.2 and now I don't have qemu crashes. Anyway, I agree that CPU emulation might have bugs, especially so specific and rare CPU. 
Maybe later I will try freebsd ppc64 on qemu, freebsd still supports ppc64 BE version

lelegard, 2024-01-17:
I use qemu 8.2.0 on macOS (M1 chip) and got the crashes.

lelegard, 2024-01-17:
I also tried SPARC64 in order to get another big endian platform. I got a Debian 12 ISO for SPARC64 (maybe the last distro to support SPARC) and tried to install it on Qemu. The installation always fail during the installation of the base packages. In the log, I find things such as this:
~~~
Jan 17 12:56:20 debootstrap: Unpacking libzstd1:sparc64 (1.5.2+dfsg2-3) ...
Jan 17 12:56:20 kernel: [ 1021.608227] dpkg-deb[12744]: segfault at 0 ip fffff8010178e30c (rpc 0000000000000000) sp fffff80102502221 error 1 in libc.so.6[fffff80101700000+18e000]
Jan 17 12:56:20 debootstrap: dpkg-deb: error: <decompress> subprocess was killed by signal (Segmentation fault)
Jan 17 12:56:20 debootstrap: dpkg: error processing archive /var/cache/apt/archives/libzstd1_1.5.2+dfsg2-3_sparc64.deb (--unpack):
Jan 17 12:56:20 debootstrap:  cannot copy extracted data for './usr/lib/sparc64-linux-gnu/libzstd.so.1.5.2' to '/usr/lib/sparc64-linux-gnu/libzstd.so.1.5.2.dpkg-new': unexpected end of file or stream
~~~

There are many segfaults during unpacking the packages.

Still on qemu 8.2.0. I give up qemu. That was a nice try, but too time consuming without reliability.

The boot command was this:
~~~
qemu-system-sparc64 \
    -machine sun4u,usb=off \
    -m 4G -nographic \
    -drive file=vm-sparc-hdd.qcow2,format=qcow2,if=ide,bus=0,unit=0 \
    -drive file=debian-12.0.0-sparc64-NETINST-1.iso,format=raw,if=ide,bus=1,unit=0,media=cdrom,readonly=on \
    -boot d
~~~

lelegard, 2024-02-16:
@svlobanov 

Just to keep you informed on this cold case. 
I wrote this above:
> About the weird behavior on the s390x emulator, gcc is maybe not the only culprit. I have some doubt on the Qemu implementation.
> ...
> ERROR:../target/s390x/tcg/fpu_helper.c:449:s390_swap_bfp_rounding_mode: code should not be reached
> Bail out! ERROR:../target/s390x/tcg/fpu_helper.c:449:s390_swap_bfp_rounding_mode: code should not be reached
> ...
> So, I think it is safe to not trust Qemu too much on s390x.

I was a bit worried and I kept this in my backlog. I rebuilt an s390x VM on Qemu and I get repeated crashes of the same kind, doing various stuff such as `apt update` or compiling TSDuck.

This is indeed a qemu bug for s390x. I submitted [a bug report](https://gitlab.com/qemu-project/qemu/-/issues/2169) for it.

In the meantime, I tested TSDuck on PowerPC64 (another big endian platform) using qemu and all tests pass.

svlobanov, 2024-02-19:
It sounds amazing how tsduck helps improving qemu and gcc.