tsduck/tsduck #1424: LATEST VERSION NOT COMILE WELL ON ARM AARCH64 ARCHITECTURE [BUG]

ntmedia2, 2024-01-17:
When i try to compile latest version of tsduck show this error

/usr/include/c++/10/bits/fs_path.h:119:29: error: call of overloaded '__is_path_src(ts::UString, int)' is ambiguous
  119 |     : decltype(__is_path_src(std::declval<_Source>(), 0))
      |                ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~

lelegard, 2024-01-17:
Hi @ntmedia2 

Could you specify which operating system, which distro, which version, and which compiler version?

From the path `/usr/include/c++/10/...`, I would vote for GCC 10 on Linux. But could you confirm?

Starting with version 3.37, TSDuck requires C++17. This is due to an ongoing effort to refactor the code to more modern techniques, to enforce long term maintenance. GCC is supposed to support C++17 from GCC 8. However, due to numerous GCC bugs, it is correctly supported from GCC 11 only. This is [explained here](https://tsduck.io/doxy/building.html#cpprequnix).

So, if you use GCC 10, yes your problem is a bug, but a GCC bug...

Old operating systems with old compilers are limited to TSDuck version 3.36 maximum.

ntmedia2, 2024-01-20:
I use arm 64 board aarch64 with dietpi distro (seem to be ubuntu based) i not find GCC 11 to install may you help me? currently was installed gcc 10

lelegard, 2024-01-20:
To manipulate specific versions of specific software, TSDuck or others, you need to understand your distro and its version. This will give you an idea on which versions of TSDuck and other software you can have.

If your distro is based on Ubuntu, you can get more details using the command `lsb_release -a`.

If you cannot upgrade it or cannot get GCC 11, then you need to build TSDuck 3.35, [source code here](https://github.com/tsduck/tsduck/archive/refs/tags/v3.35-3258.tar.gz).

lelegard, 2024-01-20:
And please post the output of the following commands:
~~~
cat /etc/debian_version 
lsb_release -a
gcc --version
~~~

ntmedia2, 2024-01-20:
i check and my distro was based on debian 11 may you help me to upgrage gcc

ntmedia2, 2024-01-20:
cat /etc/debian_version   =====> 11.8
lsb_release -a =======>No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye
gcc --version ======>gcc (Debian 10.2.1-6) 10.2.1 20210110

lelegard, 2024-01-20:
Try `sudo apt install g++-11`. Check which script shall be used to activate it (the default will probably remain GCC 10 and the two compilers are installed side by side).

Or upgrade your system to Debian 12 (Debian 11 is old).

Or build TSDuck 3.35.

This is now a system problem. This site is for TSDuck support, not for Debian, DietPi or GCC support. If you need help to upgrade your system or compiler, there are more appropriate places.

closed 2024-06-23 by github-actions[bot]