tsduck/tsduck #1425: [BUG] Undefined Behavior in ts::AVCParser::expColomb

svlobanov, 2024-01-17:
#### Bug description:
During fuzzing test (using make UBSAN=1 build) 'shift exponent' undefined behaviour was found:

~~~
src/libtsduck/dtv/codec/tsAVCParser.h:281:21: runtime error: shift exponent 590 is too large for 32-bit type 'unsigned int'
    #0 0x1033a50c4 in bool ts::AVCParser::expColomb<unsigned int, (void*)0>(unsigned int&) tsAVCParser.h:281
    #1 0x103710178 in ts::HEVCSequenceParameterSet::parseBody(ts::AVCParser&, std::initializer_list<unsigned int>) tsHEVCSequenceParameterSet.cpp:116
    #2 0x1034124ac in ts::AbstractVideoAccessUnit::parse(unsigned char const*, unsigned long, std::initializer_list<unsigned int>) tsAbstractVideoAccessUnit.cpp:35
    #3 0x10370e3e0 in ts::HEVCSequenceParameterSet::HEVCSequenceParameterSet(unsigned char const*, unsigned long) tsHEVCSequenceParameterSet.cpp:19
    #4 0x1036e8ef8 in ts::HEVCAttributes::moreBinaryData(unsigned char const*, unsigned long) tsHEVCAttributes.cpp:64
    #5 0x1038f4cd8 in ts::PESDemux::handlePESContent(ts::PESDemux::PIDContext&, ts::PESPacket const&) tsPESDemux.cpp:510
    #6 0x1038f205c in ts::PESDemux::processPESPacket(unsigned short, ts::PESDemux::PIDContext&) tsPESDemux.cpp:405
    #7 0x1038f1654 in ts::PESDemux::processPacket(ts::TSPacket const&) tsPESDemux.cpp
    #8 0x1038f0dac in ts::PESDemux::feedPacket(ts::TSPacket const&) tsPESDemux.cpp:156
    #9 0x103b60630 in ts::TSAnalyzer::feedPacket(ts::TSPacket const&) tsTSAnalyzer.cpp:1382
    #10 0x100d2bb00 in MainCode(int, char**) tsanalyze.cpp:100
~~~

#### How to reproduce:
1. Build tsduck with ubsan: `make -j9 UBSAN=1` (on macos arm64 or linux x86_64)
2. Run `UBSAN_OPTIONS=print_stacktrace=1 tsanalyze test-074.hevc-fuzz.ts`

TS file: [test-074.hevc-fuzz.zip](https://github.com/tsduck/tsduck/files/13967479/test-074.hevc-fuzz.zip)

#### Expected behavior:
no undefined behavior

#### Environment:
 - OS: macOS
 - OS version: 14.1
 - TSDuck full version: 3.37-3610
 - Installation type: rebuilt binary

#### Additional information:
this issue is similar to https://github.com/tsduck/tsduck/pull/1416

svlobanov, 2024-01-17:
You are fixing so fast. There is no commercial vendor with such speed...

lelegard, 2024-01-17:
The fix has been pushed.

Another one was in UString was also fixed (found in utest). It was another issue on iterator at end of container being nullptr (similar to #1421).

lelegard, 2024-01-17:
> You are fixing so fast. There is no commercial vendor with such speed...

I hate bugs. Especially when I created them.

Commercial vendor don't hate anything, they just love money...

closed 2024-01-17 by svlobanov