tsduck/tsduck #1426: [BUG] heap-buffer-overflow in ts::SpliceInsert::deserialize (found in asan+fuzz test)

svlobanov, 2024-01-17:
#### Bug description:
During fuzzing test (using `make ASAN=1` build) heap-buffer-overflow was found:

~~~
==59399==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x00010558eb52 at pc 0x000102ee83c0 bp 0x00016f7ae0f0 sp 0x00016f7ae0e8
READ of size 1 at 0x00010558eb52 thread T0
    #0 0x102ee83bc in ts::SpliceInsert::deserialize(unsigned char const*, unsigned long) tsSpliceInsert.cpp:179
    #1 0x102edc914 in ts::SpliceInformationTable::DisplaySection(ts::TablesDisplay&, ts::Section const&, ts::PSIBuffer&, ts::UString const&) tsSpliceInformationTable.cpp:436
    #2 0x1030583a4 in ts::TablesDisplay::displaySectionData(ts::Section const&, ts::UString const&, unsigned short) tsTablesDisplay.cpp:338
    #3 0x103056e84 in ts::TablesDisplay::displaySection(ts::Section const&, ts::UString const&, unsigned short, bool) tsTablesDisplay.cpp:320
    #4 0x103054b48 in ts::TablesDisplay::displayTable(ts::BinaryTable const&, ts::UString const&, unsigned short) tsTablesDisplay.cpp:234
    #5 0x103078b48 in ts::TablesLogger::handleTable(ts::SectionDemux&, ts::BinaryTable const&) tsTablesLogger.cpp:599
    #6 0x102e579f8 in ts::SectionDemux::ETIDContext::notify(ts::SectionDemux&, bool, bool) tsSectionDemux.cpp:157
    #7 0x102e59f50 in ts::SectionDemux::processPacket(ts::TSPacket const&) tsSectionDemux.cpp:509
    #8 0x102e58470 in ts::SectionDemux::feedPacket(ts::TSPacket const&) tsSectionDemux.cpp:211
    #9 0x1030773c4 in ts::TablesLogger::feedPacket(ts::TSPacket const&) tsTablesLogger.cpp:503
    #10 0x10064d450 in MainCode(int, char**) tstables.cpp:98
    #11 0x102b8f784 in MainWrapper(int (*)(int, char**), int, char**) tsMain.cpp:35
    #12 0x18b6290dc  (<unknown module>)

0x00010558eb52 is located 0 bytes after 18-byte region [0x00010558eb40,0x00010558eb52)
allocated by thread T0 here:
    #0 0x100f0d0ec in wrap__Znwm+0x74 (libclang_rt.asan_osx_dynamic.dylib:arm64e+0x610ec)
    #1 0x102689f6c in std::__1::vector<unsigned char, std::__1::allocator<unsigned char>>::vector(unsigned long) vector:1080
    #2 0x10268a8fc in ts::ByteBlock::ByteBlock(void const*, unsigned long) tsByteBlock.cpp:39
    #3 0x10281f1e8 in ts::DemuxedData::DemuxedData(void const*, unsigned long, unsigned short) tsDemuxedData.cpp:49
    #4 0x102e4e5b8 in ts::Section::Section(void const*, unsigned long, unsigned short, ts::CRC32::Validation) tsSection.cpp:32
    #5 0x102e59bd8 in ts::SectionDemux::processPacket(ts::TSPacket const&) tsSectionDemux.cpp:481
    #6 0x102e58470 in ts::SectionDemux::feedPacket(ts::TSPacket const&) tsSectionDemux.cpp:211
    #7 0x1030773c4 in ts::TablesLogger::feedPacket(ts::TSPacket const&) tsTablesLogger.cpp:503
    #8 0x10064d450 in MainCode(int, char**) tstables.cpp:98
    #9 0x102b8f784 in MainWrapper(int (*)(int, char**), int, char**) tsMain.cpp:35
    #10 0x18b6290dc  (<unknown module>)
~~~

#### How to reproduce:
1. Build tsduck with asan: `make -j9 ASAN=1` (on macos arm64 or linux x86_64)
2. Run `tstables  --no-pager test-084-fuzz.ts`

TS file: [test-084-fuzz.zip](https://github.com/tsduck/tsduck/files/13969686/test-084-fuzz.zip)

#### Expected behavior:
no heap-buffer-overflow issue

#### Environment:
 - OS: macOS
 - OS version: 14.1
 - TSDuck full version: 3.37-3611
 - Installation type: rebuilt binary

#### Additional information:
no additional info

svlobanov, 2024-01-17:
One more. I'm posting all of them because the issues are in the different lines of codes, but maybe all of them have the same root cause

~~~
==62014==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x0001059a703b at pc 0x000103170570 bp 0x00016f532010 sp 0x00016f532008
READ of size 1 at 0x0001059a703b thread T0
    #0 0x10317056c in ts::SpliceInsert::deserialize(unsigned char const*, unsigned long) tsSpliceInsert.cpp:218
    #1 0x103164ab8 in ts::SpliceInformationTable::DisplaySection(ts::TablesDisplay&, ts::Section const&, ts::PSIBuffer&, ts::UString const&) tsSpliceInformationTable.cpp:436
    #2 0x1032e0458 in ts::TablesDisplay::displaySectionData(ts::Section const&, ts::UString const&, unsigned short) tsTablesDisplay.cpp:338
    #3 0x1032def38 in ts::TablesDisplay::displaySection(ts::Section const&, ts::UString const&, unsigned short, bool) tsTablesDisplay.cpp:320
    #4 0x1032dcbfc in ts::TablesDisplay::displayTable(ts::BinaryTable const&, ts::UString const&, unsigned short) tsTablesDisplay.cpp:234
    #5 0x103300bfc in ts::TablesLogger::handleTable(ts::SectionDemux&, ts::BinaryTable const&) tsTablesLogger.cpp:599
    #6 0x1030dfba4 in ts::SectionDemux::ETIDContext::notify(ts::SectionDemux&, bool, bool) tsSectionDemux.cpp:157
    #7 0x1030e20f8 in ts::SectionDemux::processPacket(ts::TSPacket const&) tsSectionDemux.cpp:509
    #8 0x1030e061c in ts::SectionDemux::feedPacket(ts::TSPacket const&) tsSectionDemux.cpp:211
    #9 0x1032ff478 in ts::TablesLogger::feedPacket(ts::TSPacket const&) tsTablesLogger.cpp:503
    #10 0x1008c9450 in MainCode(int, char**) tstables.cpp:98
    #11 0x102e17a14 in MainWrapper(int (*)(int, char**), int, char**) tsMain.cpp:35
    #12 0x18b6290dc  (<unknown module>)

0x0001059a703b is located 3 bytes after 40-byte region [0x0001059a7010,0x0001059a7038)
allocated by thread T0 here:
    #0 0x1011890ec in wrap__Znwm+0x74 (libclang_rt.asan_osx_dynamic.dylib:arm64e+0x610ec)
    #1 0x102912264 in std::__1::vector<unsigned char, std::__1::allocator<unsigned char>>::vector(unsigned long) vector:1080
    #2 0x102912bf4 in ts::ByteBlock::ByteBlock(void const*, unsigned long) tsByteBlock.cpp:39
    #3 0x102aa74b8 in ts::DemuxedData::DemuxedData(void const*, unsigned long, unsigned short) tsDemuxedData.cpp:49
    #4 0x1030d6760 in ts::Section::Section(void const*, unsigned long, unsigned short, ts::CRC32::Validation) tsSection.cpp:32
    #5 0x1030e1d80 in ts::SectionDemux::processPacket(ts::TSPacket const&) tsSectionDemux.cpp:481
    #6 0x1030e061c in ts::SectionDemux::feedPacket(ts::TSPacket const&) tsSectionDemux.cpp:211
    #7 0x1032ff478 in ts::TablesLogger::feedPacket(ts::TSPacket const&) tsTablesLogger.cpp:503
    #8 0x1008c9450 in MainCode(int, char**) tstables.cpp:98
    #9 0x102e17a14 in MainWrapper(int (*)(int, char**), int, char**) tsMain.cpp:35
    #10 0x18b6290dc  (<unknown module>)
~~~

TS file: [test-149-fuzz-p1.zip](https://github.com/tsduck/tsduck/files/13969826/test-149-fuzz-p1.zip)

lelegard, 2024-01-18:
Fixed.

closed 2024-01-18 by svlobanov