tsduck/tsduck #1430: tsduck for t2mi stream

visplan, 2024-02-08:
hello! I am new to this topic, please help me to solve the problem...
I use a card TBS6909x
Astra 4.8E satellite, 12073 H 30000, from it I receive a stream of 8 simple channels and two T2MI streams, one stream is not encoded - DSNS Informed (PID 0x1001 4097), and the second (PID 0x1000 4096) is encoded by biss.
I accept the unencoded stream and send it to udp like this:
tsp -v -I dvb --adapter 3 --frequency 12073000000 --symbol-rate 30000000 --polarity horizontal --delivery-system DVB-S2 --modulation 8-PSK --pilots off --roll-off 0.35 -- lnb 9750,10600,11700 -P t2mi --pid 0x1001 -O ip 239.193.1.155:1234 and it works.
if I use the same command for the  pid 0х1000, then nothing happens and if I look at the traffic dump for the 239.193.1.155:1234, there are no packets
how to accept the second coded stream and where to enter the biss keys?
I did not find the necessary information in the manual. please help if you can
[image]

lelegard, 2024-02-08:
> is encoded by biss.

Please use the word "scrambled" or "encrypted". The word "encoded" is used for video and audio encoding / compression such as MPEG, AVC, HEVC, etc.

There is no BISS support in TSDuck. There is a `descrambler` plugin which supports various encryption algorithms with a fixed control word, but no direct support to the BISS system.

visplan, 2024-02-09:
> Please use the word "scrambled" or "encrypted". The word "encoded" is used for video and audio encoding / compression such as MPEG, AVC, HEVC, etc.
>
> There is no BISS support in TSDuck. There is a [...]

Thank you :)

closed 2024-02-09 by visplan