tsduck/tsduck #1431: [Q] How To set up tsp command to broadcast a service with a specific provider id and ca_system_id?

assallay, 2024-02-08:
I am playing around with tsduck for a while and i used it to rewire some services but i cant manage to make a specific service with a specific provider id .. any examples will be helpful.
I use dektec 107 dvbs qpsk for output.
tbs6906 for input streams and i use some remapping .
this is my code any help
#!/bin/bash
killall -9 tsp1
killall -9 tsp2

tsp1 -t -b 72012450 -I null \
-P inject "<?xml version='1.0' encoding='UTF-8'?>
<tsduck>
  <PAT transport_stream_id='1'>
    <service service_id='700' program_map_PID='5000'/>
  </PAT>
</tsduck>" -b 1024 -s -p 0 \
-P inject "<?xml version='1.0' encoding='UTF-8'?>
<tsduck>
  <SDT transport_stream_id='1' original_network_id='1'>
 <service service_id='700' running_status='running'>
      <service_descriptor service_type='0x19' service_name='CHANNEL 1' service_provider_name='PROVIDER 1'/>
    </service>
  </SDT>
</tsduck>" -b 1024 -s -p 17 \
-P inject "<?xml version='1.0' encoding='UTF-8'?>
<tsduck>
  <PMT service_id='700'>
    <component elementary_PID='5001' stream_type='0x1B'/>
    <component elementary_PID='5002' stream_type='0x0F'/>
  </PMT>
</tsduck>" -b 1780 -s -p 5000 \
-P merge -t --no-pcr-restamp "tsp2 -t -I dvb -a 3 --frequency 12284000000 --symbol-rate 27500000 --fec-inner 2/3 --spectral-inversion auto --polarity vertical --lnb 9750,10600,11700 --satellite-number 0 --receive-timeout 0 --signal-timeout 0 \
	-P analyze -i 30 -o ./input1.txt \
	-P remap -u 701=5001 702=5002 \
	-P filter -p 5001 -p 5002" \
-P regulate -b 72012450 \
-P pcradjust -b 72012450 \
-P analyze -i 30 -o ./output2.txt \
-O dektec --channel 0 --device 1 -r 7/8 -f 1000000000 -l -27 -s --symbol-rate 44000000 -m DVB-S-QPSK 2>>out.log &

best regards.

lelegard, 2024-02-09:
This seems complicated. Does it work? What kind of problem do you have, if any?

I would rather do this:
- Use one single `tsp` command, with `-I dvb` and `-O dektec`.
- If the input TS contains more than one service: `-P zap --stuffing service-id`
- If the input bitrate is higher than the output one: `-P reduce --target-bitrate 72012450` (based on your example).
- Remap PID's if you really want to. `-P remap` will adjust the PAT and PMT accordingly.
- Use plugin `sdt` to update service name or id.

assallay, 2024-02-09:
yes it works @lelegard
every thing works fine .. I can change the ` service_provider_name='PROVIDER 1' ` but provider id  its always 000000 and i want change it to 0407EA ... is there a plugin to change it or this is a STB level of describing channels for every provider per frequency.
thanks in advance

lelegard, 2024-02-09:
> but provider id its always 000000

Where do you this ?

There is no such thing as "provider id". The provider name in the SDT is only an informational name without technical interpretation. There is no corresponding id.

assallay, 2024-02-09:
@lelegard what i was asking is how to differentiate from this list of services and other services from another provider on the STB level >> the stb can some how no this channel SID belongs to provider id and when you change the channel it changes as well..
or am i confused .
i am making a research on stb pairing as theoretically .. it can change provider id from source and change that provider id on stb level so it matches whithin the smartcard as requested..
any conclusion will be appreciated..

thnx

lelegard, 2024-02-10:
There is some confusion between concepts here.

The "provider" is related to a service. This is only a meaningless name. There is no provider id. There is no rule to use it. Sometimes, the provider name is the MUX technical operator, sometimes it is the network operator, sometimes it is the holding or parent company of the service. In short, this is purely informational, technically useless, and never used by the STB to decide anything.

The smartcard is distributed by a commercial operator, which may create bouquets and sell them based on different levels of subscription.

The smartcard is produced by a CAS vendor (Conditional Access System). The commercial operator is a customer of the CAS vendor.

A service does not "belong" to a commercial operator or CAS vendor. A scrambled service can be commercialised by distinct commercial operators, using distinct CAS vendors, for different types of smart cards. The rules of operations are described by DVB SimulCrypt.

The CA System is described by a 16-bit CAS id. The CAS id is present in CA descriptors which describe the ECM (PMT) and EMM (CAT) PIDs. The PMT of a scrambled service contain one or more CA descriptors per CAS which can be used to access the service.

The term "STB pairing" is used to described a cryptographic mechanism through which a given smartcard can only be used in a given STB. A "paired" smartcard cannot be used in another STB, even the same model, from the same commercial operator.

In short, this can become quite complicated. There is no hierarchical tree of services and operators. There is a mesh of editorial, technical and commercial actors. I would take pages and pages to fully describe and I cannot do it here. I think that you first need to clarify your understanding of TV networks and then clearly describe what you want to do in technical terms.

assallay, 2024-02-12:
thanks  alot @lelegard ..  I am happy that you cleared for me the path , more reading and studying to do than.
my goal of research is is scrambling a service from source to STB and then descramble this service only on specific STBs from my choice ... can it be done using Tsduck..

the whole idea is .. can we use Tsduck on Linux STBs like ENIGMA2 or alike .. if so can i ask for an example for scrambling and descrambling.
another [Q]. can a scrambled service be scrambled again or not?.
thanks for the effort and such a great software. it makes me happy when things work for as it clears the way of understanding things in a technical style.
best regards

lelegard, 2024-02-12:
The problem is not scrambling (the plugins `scrambler` and `descrambler` can be used for a demo). The problem is the secure distribution of subscription rights and descrambling keys. This is where you need a CAS, Conditional Access System. TSDuck can interface with a CAS, it can perform the scrambling and inject EMM's and ECM's. But TSDuck alone cannot implement a secure PayTV system. The security is the responsibility of a CAS.

assallay, 2024-02-12:
Thank you for Explaining alot here @lelegard ... i will go through the documentation more and read more about CAS systems.
i am glad to reach a point of understanding and another journey of search .. god bless us all.
best regards.

closed 2024-02-12 by assallay