tsduck/tsduck #1432: [Q] How we can add subtitles to a ts file?

jangrayogesh, 2024-02-11:
I have go through the documentation but I didn’t find anything that can help me with the subtitles problem.
Please help me how i can add subtitles to a ts file ??
I will be thankful to you if you can help me with that.

lelegard, 2024-02-12:
I recommend using ffmpeg for this.

Subtitles are synchronized with audio and video. Just like audio is synchronized with video. Synchronized insertion is the job of a MUX. TSDuck is not a MUX. It typically inserts loosely synchronized data only (sections).

jangrayogesh, 2024-02-14:
Do you have the knowledge about the ffmpeg command because i have tried alot of commands but it didn’t work .
The command that i am using is :
ffmpeg -i multitrack-sub.mp4 -t 30 -map 0 -c:v libx264 -c:a aac -c:s copy -f mpegts mul-sub2.ts

konrad-dabek, 2024-02-14:
Command like this surely won't work, because MP4 subtitles have got an incompatible format, which cannot be directly inserted into TS, and you're trying to copy this directly to TS.

You can:

1
Burn subs into video (simple, but quite ugly):
https://video.stackexchange.com/questions/27889/closed-captions-missing-when-converting-from-mp4-to-ts

2
Or the hard (but elegant) way - extract subtitles from MP4 with ffmpeg to SRT, recode to DVD format with Subtitle Edit (DVB TS subtitles are bitmap-based, ffmpeg is able to put them only if text-based source is recoded to bitmap-based, which ffmpeg knows as dvdsub) and finally insert to TS accordingly to ffmpeg docs, keeping sync in mind.
Surely this won't be an one-liner and it rather won't be possible to do this in one step, because as far as I'm aware ffmpeg misses a way to convert subtitles from text-based to bitmap-based. Some external tool is needed.
Help/tip: https://stackoverflow.com/questions/59094335/muxing-dvb-subtitles-into-transport-stream-with-ffmpeg
https://superuser.com/questions/1702250/how-to-convert-text-based-subtitles-to-image-based-subtitles

But we're definitely not in a ffmpeg support board :)

jangrayogesh, 2024-02-14:
Thanks for the reponse
Hard subtitle are burned into the video and they are not able to disabled like soft subtitle .
I want to add soft subtitle for Example like English and Hindi.
And is there any other option like i dont want to use subtitle edit because i will be using multiple ts files in future and it will be not possible to use subtitle edit .
I want to use command in my code so that it will add the subtitle or convert them according to need.

Again thanks for your reponse

jangrayogesh, 2024-02-14:
Thanks for reponse

closed 2024-07-14 by github-actions[bot]